dotintent/react-native-ble-plx #608: Characteristic notifications arrive out of order on Android

robwalkerco, 2020-01-31:
## Expected Behavior

I expect 'onNext' of `safeMonitorCharacteristicForDevice` for each notification to be called in the same order that the notifications are received. 
This would then log (using `characteristic.logValue("Notification from", bytes);`) the notification bytes in the correct order.

This is the behaviour that is observed in iOS.

## Current Behavior

Currently, some notifications arrive out of order for Android only.

### Steps to Reproduce

We are using a BLE device that sends notifications very fast (every few milliseconds) in order to transfer large amounts of data when we request it from the device.

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

I was originally thinking that this could be similar to the issue fixed in https://github.com/Polidea/RxAndroidBle/pull/639 but I'm not sure?

* Library version: 1.1.1
* Platform: Android
* Contents of the `package.json` file:
<details>

```
{
  "name": "...",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "concurrently -r \"tsc -w\" \"REACT_DEBUGGER=\\\"open 'rndebugger://set-debugger-loc?host=localhost&port=8081' --args\\\" react-native start\"",
    "test": "jest",
    "lint": "prettier \"app/**/*.{ts,tsx}\" --write "
  },
  "dependencies": {
    "@react-native-community/async-storage": "^1.5.0",
    "base64-arraybuffer": "^0.2.0",
    "date-fns": "^2.9.0",
    "react": "16.8.6",
    "react-native": "0.60.5",
    "react-native-ble-plx": "1.1.1",
    "react-native-gesture-handler": "^1.5.3",
    "react-native-keep-awake": "^4.0.0",
    "react-native-permissions": "^2.0.2",
    "react-native-reanimated": "^1.7.0",
    "react-native-share": "^2.0.0",
    "react-native-tab-view": "^2.11.0",
    "react-redux": "^6.0.1",
    "redux": "^4.0.1",
    "redux-devtools-extension": "^2.13.5",
    "redux-persist": "^6.0.0",
    "redux-saga": "^1.1.3",
    "reselect": "^4.0.0",
    "rn-fetch-blob": "^0.12.0",
    "styled-components": "^4.4.0",
    "text-encoding": "^0.7.0",
    "ts-enum-util": "^4.0.1",
    "uuid": "^3.3.3"
  },
  "devDependencies": {
    "@babel/core": "^7.8.3",
    "@babel/runtime": "^7.8.3",
    "@react-native-community/eslint-config": "^0.0.6",
    "@types/react": "16.8.6",
    "@types/react-native": "^0.60.11",
    "@types/react-native-keep-awake": "^2.0.2",
    "@types/react-native-share": "^1.1.5",
    "@types/react-redux": "^7.0.5",
    "@types/styled-components": "^4.1.19",
    "@types/text-encoding": "^0.0.35",
    "@types/uuid": "^3.4.6",
    "babel-jest": "^24.9.0",
    "concurrently": "^4.0.1",
    "eslint": "^6.8.0",
    "jest": "^24.9.0",
    "metro-react-native-babel-preset": "^0.56.4",
    "prettier": "^1.19.1",
    "react-test-renderer": "16.8.6",
    "tslint": "^5.11.0",
    "typescript": "^3.7.4"
  },
  "jest": {
    "preset": "react-native"
  }
}
```
</details>

Cierpliwy, 2020-01-31:
Can you test library version 1.1.0?

dariuszseweryn, 2020-01-31:
Notifications from a single characteristic should arrive in order. From how many characteristics you receive data?

robwalkerco, 2020-01-31:
Hi guys. 
It's a single notification that's sending the data. I am also subscribed to a second notification that's used to signal the end of the data sending. 

I'll give 1.1.0 a try, then share the verbose log output from both iOS and Android showing the issue.

dariuszseweryn, 2020-01-31:
And which notifications come out-of-order? The `end of data` signal and the last packet of data? Or data is scrambled?

robwalkerco, 2020-01-31:
The out-of-order notifications seem to happen randomly while the BLE device is sending them. The issue is not when we receive the `end of data` signal.

These are a sample of the Logcat logs while we are fetching data from the device.

We ask for some data `18fa1407-c7a6-4f8d-af28-5c57a00775b2`
We get loads on notifications from `18fa1408-c7a6-4f8d-af28-5c57a00775b2` where the order they are sent/received matters.
We receive a notification from `18fa140a-c7a6-4f8d-af28-5c57a00775b2` to signal that all the data is sent

We expect to receive a particular pattern in the data. When this pattern is broken its a sign that the notifications were received in the wrong order.
1) 488 hex characters
2) 488 hex characters
3)4 hex characters
Note: The final value received before `18fa140a-c7a6-4f8d-af28-5c57a00775b2` could be any length.

I'll share a similar log from iOS shortly, but we are not seeing any issues on iOS.

Logcat log available at https://gist.github.com/robwalkerco/41e8fa48f53902738e9e47d5ece1f0fb

robwalkerco, 2020-01-31:
I also meant to add the line numbers to look at in that gist!

Check 
194 - 198
351 - 355
1029 - 1031

robwalkerco, 2020-01-31:
@Cierpliwy The Logcat example is using react-native-ble-plx 1.1.0, so there appears to be no difference between 1.1.0 and 1.1.1 .

Cierpliwy, 2020-01-31:
That's good to know. Ok, so it looks like the number of notifications is correct, but sometimes values are duplicated.

@dariuszseweryn Maybe we don't handle notifications fast enough on the Android and values are overridden as Characteristic is just an instance to the same object under the hood?

robwalkerco, 2020-01-31:
Yeah, I think you are correct @Cierpliwy .
Some notifications are getting duplicated (values and timestamps are exact).

I may be able to implement some deduplication logic somewhere in my app as a quick fix for now, while the cause of the issue is investigated, though I would clearly prefer to be able to avoid that ;)

Please let me know if you would like any additional data or assistance in testing any changes :)

dariuszseweryn, 2020-01-31:
[The implementation retrieves a reference to characteristic byte array synchronously in the `onCharacteristicChange` callback](https://github.com/Polidea/RxAndroidBle/blob/master-rxjava1/rxandroidble/src/main/java/com/polidea/rxandroidble/internal/connection/RxBleGattCallback.java#L144). I have not experienced any issues up to date with this implementation as Android OS seem to not reuse byte arrays used for returned characteristic values.

I have looked into [other established implementations](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/ba1968d1f5a3f54706110ec1da2052a789357047/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L2025) and they also seem to use the same approach.

robwalkerco, 2020-01-31:
@Cierpliwy 
Sorry for even more messages!

I've been looking at some log data I got yesterday (running 1.1.1), and in that data it looks like the duplicate data actually replaces other data. It actually seems to happen really frequently:

17-18
20-21
32-33
35-36
56-57
...

I've put the timestamps and values in a gist at https://gist.github.com/robwalkerco/6a1875ef54b5627c51c36e416504a6e3

dariuszseweryn, 2020-01-31:
Could you add info about what device / OS you test on?

robwalkerco, 2020-01-31:
> Could you add info about what device / OS you test on?

I'm testing on a Galaxy S7 edge running Android 8.0.0

robwalkerco, 2020-01-31:
I'm just comparing the Android data to some iOS log data - https://gist.github.com/robwalkerco/6954d369c7d9470831a01534b4aaefd0

It seems that the data is frequently duplicated on iOS as well, so that must be the BLE device sending the data like that.

However, the iOS data always follows exactly the pattern of 2 long values followed by 1 short, so I'm still convinced there is an issue somewhere with Android.

Thanks for your help looking into this issue guys

dariuszseweryn, 2020-01-31:
It is possible that you have encountered a previously unknown (at least to me) Android bug. There is also a possibility that your peripheral's firmware has a bug which manifests on Android as iOS may negotiate different connection parameters.

robwalkerco, 2020-01-31:
Thanks for your  @dariuszseweryn and @Cierpliwy  

I'll take it up with the hardware manufacturers, as there does seem to be something strange happening on iOS as well.

I'll close this issue for now until we have ruled out a hardware issue.

dariuszseweryn, 2020-01-31:
Feel free to update with whatever outcome you will have so other users who will stumble upon this issue could get the final thoughts as well.

robwalkerco, 2020-02-13:
I havent managed to solve this yet, but I have found a very similar issue what was happening in a Flutter BLE library.
The issue is discussed at https://github.com/pauldemarco/flutter_blue/issues/317
And was resolved via these two commits -
https://github.com/pauldemarco/flutter_blue/commit/fad8be1329d86ba4f7152e33899e1f9f7f6eba0f
https://github.com/pauldemarco/flutter_blue/commit/fc6637d11ee448f1b30b52c84787ecd47d1e9603

The BLE device that our app connects too does disconnect itself from our our app, after which we attempt a reconnection and wait for the device to connect again in future. This seems to be a big contributing factor in the linked issue.

@dariuszseweryn I don't know enough about RX to see if the same problem may exist in react-native-ble-plx, so I was hoping you would be able to tell?

dariuszseweryn, 2020-02-13:
The commits do not look like anything that could be potentially related to what you are experiencing AFAIK. `BluetoothGatt` is disconnected and closed for each connection — it is never reused.

I would need to get a sniffed BLE traffic from an external sniffer and probably HCI logs from the device to tell more what can be happening.

robwalkerco, 2020-03-02:
Hi @dariuszseweryn 

We are still trying to fully resolve this issue :(

I've received some advice regarding handling fast notifications from Nordic (https://github.com/NordicSemiconductor/Android-nRF-Connect/issues/117#issuecomment-593291007)

Are the libraries that underpin react-native-ble-plx implemented in a way that follows this advice? (1. Values should be read as fast as possible just after receiving. 2. A default Handler (on Android) or DispatchQueue (on iOS) should be used to read those values. Using another, dedicated handler/queue causes additional delay.)?

I'm trying to work out if I would be better off sticking with react-native-ble-plx or trying to move to a different implementation (possibly Android-BLE-Library with a custom React Native wrapper)

 Thanks so much for all your help.

dariuszseweryn, 2020-05-01:
@robwalkerco Sorry for such a delay — I was on holiday when you responded and somehow I missed it. This is a known issue and values are read as quickly as possible. There is one more thing that `RxAndroidBle` (or currently MBA) could do to prevent overwriting value issue when a characteristic may be written and notify at the same time. That would be to use a copy of characteristic object for writing.

robwalkerco, 2020-05-01:
Thanks for your reply.
We are currently dealing with this by having the BLE device send characteristics slower. It mostly works but not a great solution.

We are implementing a system where we try to receive notifications fast, then check the data received for issues and fallback to a receive where we request the data is sent slower.

We have found that some phones were receiving the fast data fine, and others really struggled.

dariuszseweryn, 2020-05-01:
When dealing with a lot of notifications it may happen that you will face crashes (`MissingBackpressureException`) on Android devices when React Native bridge will not cope with the amount of objects to push through. We were thinking about this issue but currently there is no one strategy that would fit all users.

What is the approximate throughput when you start to experience issues? How many notifications per second and how long the notifications?

robwalkerco, 2020-05-01:
In an attempt to fix these issues, I'm actually using a fork of the last 1.x.x version, where I batch up characteristic notifications to reduce traffic over the RN bridge. (https://github.com/robwalkerco/react-native-ble-plx/blob/cb12cc3baf8bfda0ac0f7bc5c9e79c9e443ee1bf/android/src/main/java/com/polidea/reactnativeble/BleModule.java#L1897) I do the same on iOS as well.

I'm note sure if that actually makes any difference though!

I'll be updating to 2.0.0 soon, so will replicate by batching approach there I think.

I'll share some example data and the number of notifications with you as soon.

robwalkerco, 2020-05-22:
Just wanted to update this issue with our final solution.

We have ended up supporting two data transfer speeds on our BLE device firmware. (normal & slow)

When we fetch a batch of data, we check the integrity of the data to see if there were any transfer issues. If there were issues we refetch in slow mode.

When testing on an iPhone 7 and Samsung S7, we sometimes need to fallback to slow mode, and on a Pixel 3 we always get the data through first time (and super fast!)

We also do the same slow mode fallback if we get a `MissingBackpressureException`.

We are using version 2.0.0 of react-native-ble-plx now, and have not bothered about batching the notification data on the native side. It does not seem to make any difference (this was originally to rule out issues caused by transfer over the RN bridge).

We are finally happy with this solution, and it seems to make for a fairly reliable data transfer :)

Thanks for your help @dariuszseweryn

closed 2020-05-22 by robwalkerco