dotintent/react-native-ble-plx #609: When showing devices from scan, it doesn't populate my paired devices?

mmrk09, 2020-02-02:
I have a device paired with my phone that has a button to click, it works at the OS level. But when listing devices found during scan i get devices with no name to display and they all say not connected so I'm not sure which one is my remote I'm already using?

So I am trying to test using this remote, clicking a button, and seeing the data come through using this library. But why would the name be blank and it say not connected for every device it finds during scan, when it has a name and it appears in the OS under paired devices?

Cierpliwy, 2020-02-03:
This library supports only Bluetooth Low Energy devices. First of all, make sure that your device uses this technology. Each application on iOS/Android has so-called "virtual connection" to the device and even when the device is connected to your phone it may not be connected to your application.

mmrk09, 2020-02-08:
I purchased a BLE device off the internet, it's bluetooth 4+, and it shows and registers as a BLE device using diagnostic tools. However, I can only see it in the operating system blue tooth connections and it has a device name and does not show up in the ble react native library code or if it does is there without a name or localName.

So this doesn't make sense to me, my samsung tv's show up with proper names. I then paired it with the OS (samsung tv's are not paired but show up) and still nothing changed. So what could the issue be and how do i inspect one of these unknown devices to see if that is in fact it?

njdullea, 2020-02-17:
@Cierpliwy How do you reconnect a device that is connected to the phone, but not the application?

Cierpliwy, 2020-02-18:
Please check if you can discover and connect to your device via `nRF Connect` application.

closed 2020-04-25 by stale[bot]

Joao360, 2020-06-17:
I'm having the same issue. The device I paired with is a Bluetooth Low Energy device and I'm able to connect to it via `nRF Connect` on the `Bonded` menu.

If I unpair it, it will appear on the scanned devices.

I've been testing this on android devices. Tested on android 6 and 10 and it seems like this issue only happens on android 10

taj567, 2022-01-03:
how can I retrieve the data from a paired Bluetooth device in react-native  using ----react-native-ble-plx