dotintent/react-native-ble-plx #610: Can't monitor device

WQvist, 2020-02-05:
According to the docs, I've done the following:
- Scanned for devices
- Connected to device
- device.discoverAllServicesAndCharacteristics
- writeCharacteristicWithResponseForDevice

My problem lies in reading the response I get from sending my command.

RX characteristic is writable, but TX is only notifiable and not readable. Therefore I need to monitor the characteristic with monitorCharacteristicForDevice, however I can't get this to function.

I never recieve neither "Monitor success" nor "monitorDevice called" in the code below. If I run the checkBattery() function after connection it returns "Not monitoring".

**How do I read the return?**

```
monitorDevice = () => {
	console.log("monitorDevice called")
	this.manager.monitorCharacteristicForDevice(this.state.connectedTo, SERVICE_UUID, TX_CHARACTERISTIC, (error, characteristic) => {
		if (error) {
			console.error("Error at receiving data from device", error);
			return     
		}
		else{
			this.setState({monitoring:true})
			console.log("Monitor success")
			console.log("Monitor success " + characteristic.value);
			this.state.messagesRecieved.push(characteristic.value)
			return
		}
	})
}

connectAndFetchServicesAndCharacteristics = (device) => {
	this.manager.connectToDevice(device.id, null)
	.then(device => {
		this.state.connectedTo = device.id
		console.log('Connection successful\nScan stopped')
		this.manager.stopDeviceScan();
		this.setState({scan: false})
		return device.discoverAllServicesAndCharacteristics()
	})
	.then(device => {
		this.setState({servicesDiscovered: true})
		console.log("Services discovered")
		this.monitorDevice
		return
	})
	.catch(err => {
		console.log(err)
	})
}
  
checkBattery = () => {
	if(this.state.monitoring){
		this.manager.writeCharacteristicWithResponseForDevice(this.state.connectedTo,
			SERVICE_UUID, 
			RX_CHARACTERISTIC,
			"YmF0dGVyeQ==")
		.then(characteristic => {
			console.log("Successfully sent: " + characteristic.value)
			return
		})
		.catch(err => {
			console.log(err)
		})
	}
	else{
		alert("Not monitoring")
	}
}
```

dariuszseweryn, 2020-02-05:
`this.monitorDevice` is a function therefore you must
`this.monitorDevice` => `this.monitorDevice()`

WQvist, 2020-02-05:
My bad, changed it in my original post. There is still no difference however, other than that I recieve ```monitorDevice called``` since I now actually call the function.

dariuszseweryn, 2020-02-05:
Good. So now you can show the native logs to see what is happening if your peripheral does indeed send any notifications

WQvist, 2020-02-05:
Alright, with verbose loglevel I get "ConnectOperation finished" followed by peripheral info. After I run checkBattery() I get:
```
D RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(88434918)
D RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(88434918)
D RxBle#BluetoothGatt$1: onCharacteristicWrite characteristic=... status=0
V RxBle#BleModule: Write to Characteristic(uuid: ..., id: 8, value: 62617474657279)
D RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(88434918) in 121 ms
```

Nothing happens after that.

dariuszseweryn, 2020-02-05:
The interesting part is if notification is properly set

WQvist, 2020-02-05:
Sorry, I don't understand. Do you think both notification and write is not properly set up in the above code?

dariuszseweryn, 2020-02-05:
I dunno. I want to check what is going on.

btw. you directly set the state `this.state.connectedTo = device.id`

WQvist, 2020-02-06:
Progress! Hooking the peripheral up to a serial monitor I can see that the received string is basically gibberish. "battery" is received as "m畫mz�". Some type of encoding problem

closed 2020-02-06 by WQvist

avcismail, 2024-05-29:
Hi WQvist, I have a similar project. When I run the same code, I get an output like the following. Were you getting a printout like this?

`Write success:  {"_manager": {"_activePromises": {}, "_activeSubscriptions": {}, "_errorCodesToMessagesMapping": {"0": "Unknown error occurred. This is probably a bug! Check reason property.", "1": "BleManager was destroyed", "100": "BluetoothLE is unsupported on this device", "101": "Device is not authorized to use BluetoothLE", "102": "BluetoothLE is powered off", "103": "BluetoothLE is in unknown state", "104": "BluetoothLE is resetting", "105": "Bluetooth state change failed", "2": "Operation was cancelled", "200": "Device {deviceID} connection failed", "201": "Device {deviceID} was disconnected", "202": "RSSI read failed for device {deviceID}", "203": "Device {deviceID} is already connected", "204": "Device {deviceID} not found", "205": "Device {deviceID} is not connected", "206": "Device {deviceID} could not change MTU size", "3": "Operation timed out", "300": "Services discovery failed for device {deviceID}", "301": "Included services discovery failed for device {deviceID} and service: {serviceUUID}", "302": "Service {serviceUUID} for device {deviceID} not found", "303": "Services not discovered for device {deviceID}", "4": "Operation was rejected", "400": "Characteristic discovery failed for device {deviceID} and service {serviceUUID}", "401": "Characteristic {characteristicUUID} write failed for device {deviceID} and service {serviceUUID}", "402": "Characteristic {characteristicUUID} read failed for device {deviceID} and service {serviceUUID}", "403": "Characteristic {characteristicUUID} notify change failed for device {deviceID} and service {serviceUUID}", "404": "Characteristic {characteristicUUID} not found", "405": "Characteristics not discovered for device {deviceID} and service {serviceUUID}", "406": "Cannot write to characteristic {characteristicUUID} with invalid data format: {internalMessage}", "5": "Invalid UUIDs or IDs were passed: {internalMessage}", "500": "Descriptor {descriptorUUID} discovery failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "501": "Descriptor {descriptorUUID} write failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "502": "Descriptor {descriptorUUID} read failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "503": "Descriptor {descriptorUUID} not found", "504": "Descriptors not discovered for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "505": "Cannot write to descriptor {descriptorUUID} with invalid data format: {internalMessage}", "506": "Cannot write to descriptor {descriptorUUID}. It's not allowed by iOS and therefore forbidden on Android as well.", "600": "Cannot start scanning operation", "601": "Location services are disabled"}, "_eventEmitter": {"_nativeModule": [Object]}, "_scanEventSubscription": null, "_uniqueId": 8}, "deviceID": "16802D6E-0F2E-8CAC-91AA-5E5463A6F61F", "id": 10791798272, "isIndicatable": false, "isNotifiable": false, "isNotifying": false, "isReadable": false, "isWritableWithResponse": true, "isWritableWithoutResponse": false, "serviceID": 10740243328, "serviceUUID": "6e400001-b5a3-f393-e0a9-e50e24dcca9e", "uuid": "6e400002-b5a3-f393-e0a9-e50e24dcca9e", "value": null}`

My Method is here:
```
const writeCharacteristic = async () => {
    if (!deviceID) {
      console.log("Device ID is not set");
      return;
    }
    await manager
      .writeCharacteristicWithResponseForDevice(
        deviceID,
        serviceUUID,
        fullUUID(characteristicUUIDRX),
        btoa(resetCounter()) //`${resetCounter()}`
      )
      .then((value) => {
        console.log("Write success: ", value);
        return;
      })
      .catch((error) => {
        console.log("Write Failed: ", error);
      });
  };
```

WQvist, 2024-05-29:
> Hi @WQvist , I have a similar project. When I run the same code, I get an output like the following. Were you getting a printout like this?

I did not, but I can see that you log "value" when you should be logging "value.value" to read the actual response. However, the response is currently null and I'm not sure if it makes sense or not. It's been a few years so I'm sorry I can't help you further. Good luck!