dotintent/react-native-ble-plx #611: Is connection to multiple devices simultaneously possible?

srossi3, 2020-02-05:
Hi everyone! 
Would it be possible to connect multiple devices to an app using this library?
In particular, I would need to read controls from one bluetooth device, process them in the application and send new controls to a different device. 
Thank you, 
Silvia

dariuszseweryn, 2020-02-05:
> Would it be possible to connect multiple devices to an app using this library?

Yes, it is possible

closed 2020-02-05 by dariuszseweryn

federicomiralles, 2020-04-27:
Out of curiosity... Is there a limit? If so what's the max number of simultaneous connections?

Thanks in advance!

Federico

dariuszseweryn, 2020-04-27:
Look for limits on individual OSes that you target

punndcoder28, 2020-06-02:
Hi @dariuszseweryn ! I was trying out the module for a project I was working on. Could you guide me on how we can connect to multiple devices?
Thanks in advance!

jithinlal, 2020-07-23:
@punndcoder28 did you achieved this functionality?

punndcoder28, 2020-07-25:
Hi @jithinlal ! Yes I was able to achieve this. Multiple connections can be opened in parallel using this library. I just followed their official docs and used data structure to store connection details.

kristinadarroch, 2020-11-15:
@punndcoder28 - how did you achieve this? where are the docs for this functionality?

punndcoder28, 2020-11-18:
Hey @kristinadarroch ! All you have to do connect to multiple devices in sequence. The previous connections are still maintained and are not disconnected. Once connected the connections are maintained and you can receive data from multiple connected devices simultaneously. But there might be platform specific constraints as to how many connections can be made. You'll have to look into iOS/Android docs for them.There isn't a doc section particularly for this, but shouldn't be too hard.