dotintent/react-native-ble-plx #613: Bug Report

gajendra-weir, 2020-02-12:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ Yes] I am running the latest version
- [ Yes] I checked the documentation and found no answer
- [ Yes] I checked to make sure that this issue has not already been filed

## Expected Behavior
Please describe the behavior you are expecting
When subscribed to Streaming channel for data notifications, One should get the notifications with complete value (hexadecimal encoded in base64)

## Current Behavior

What is the current behavior?
When subscribed to Streaming channel for data notifications, a part of hexadecimal value encode in base64 is getting chopped.
**current value:** oF+5Ak9TWU4xMjM0NTYAAAAAAAAeAAAACAAUtFekprOhAENvbmUgQ3J1c2hlciAxMjMAAAAAAAAAAAAAAAAAABBhuQJPygkbBAAEAAQAAAAAAAAATPyw/xwAAAAAAA8NEIO5Ak/KCRsEAAQABAAAAAAAAABL/K//HAAAAAAADw0QobkCT8oJGwQABAAEAAAAAAAAAEv8rv8cAAAAAAAPDRDDuQJPygkbBAAEAAMAAAAAAAAAS/w=
**expected:** 
oF+5Ak9TWU4xMjM0NTYAAAAAAAAeAAAACAAUtFekprOhAENvbmUgQ3J1c2hlciAxMjMAAAAAAAAAAAAAAAAAABBhuQJPygkbBAAEAAQAAAAAAAAATPyw/xwAAAAAAA8NEIO5Ak/KCRsEAAQABAAAAAAAAABL/K//HAAAAAAADw0QobkCT8oJGwQABAAEAAAAAAAAAEv8rv8cAAAAAAAPDRDDuQJPygkbBAAEAAMAAAAAAAAAS/yu/xwAAAAAAA8N

### Steps to Reproduce
Since device is specific to organisation, can't share details.
Otherwise, below are the steps i am following
Please provide detailed steps for reproducing the issue.

1. change operating mode to transfer
2. subscribe to streaming channel for data transfer
3. trigger transfer
4. we will start receiving notifications
5. a part of value in the notification is being chopped and no subsequent notification that has this chopped data.

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: 1.1.0
* Platform: iOS.
* Platform logs (logcat/XCode):
* JS logs: 
* Contents of the `package.json` file:
    "react": "16.8.6",
    "react-native": "0.60.3",
* Formatted code sample or link to a repository:

```js
function example(): string {
  return "Your code sample"
}
```

dariuszseweryn, 2020-02-12:
Hello,

Apart from observing a chopped notification – have you confirmed by other means that the expected behaviour is actually possible?

gajendra-weir, 2020-02-12:
Yes. Verified this on light blue app. Expected behaviour is possible

gajendra-weir, 2020-02-12:
> Hello,
> 
> Apart from observing a chopped notification – have you confirmed by other means that the expected behaviour is actually possible?

Yes. Verified this on light blue app. Expected behaviour is possible

dariuszseweryn, 2020-02-12:
Could you include native logs?

gajendra-weir, 2020-02-12:
I don't see any log in xcode.

@dariuszseweryn One another observation, on my android phone i get expected value. However on iPhone, even in light blue, i get it chopped.

Does this means it is device limitation?

dariuszseweryn, 2020-02-12:
> I don't see any log in xcode.

This part of the issue template describes how you can enable platform logs
> Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via setLogLevel function call.

> @dariuszseweryn One another observation, on my android phone i get expected value. However on iPhone, even in light blue, i get it chopped.
> 
> Does this means it is device limitation?

Yup. Your peripheral does not take into account that MTU may be lower than it expects.

closed 2020-02-12 by dariuszseweryn