dotintent/react-native-ble-plx #614: Unable to scan services/characteristics, or report on any device data

mmrk09, 2020-02-13:
code samples just have comments //do something but the device doesn't end up being populated with services or characteristics that are available to read or try to subscribe to a notification for?            

this.device.data.connect()
                .then((device) => {
                    return device.discoverAllServicesAndCharacteristics();
                })
                .then((device) => {
                      console.log(device.serviceUUIDs); <-- null   
                      console.log(device.serviceData); <-- null
                })
                .catch((error) => {
                    // Handle errors
                    console.log(error);
                    Alert.alert(
                        'Error',
                        'Failed connecting to device',
                        [
                            {text: 'OK', onPress: () => console.log('OK Pressed')},
                        ],
                        {cancelable: false},
                    );
                });

dariuszseweryn, 2020-02-13:
`serviceUUIDs` and `serviceData` is available only in scan advertisement.
Here you have wiki for service discovery. https://github.com/Polidea/react-native-ble-plx/wiki/Device-Service-Discovery

You need to first make service discovery then use `device.services().then(services => /* use services here */)`

mmrk09, 2020-03-05:
The code we have above does do a discovery first? We are trying to get a working code snippet that does the discovery then allows us to review services/characteristics and try one to evaluate BLE communication with a device. 

So we do discovery using promises then evaluate data, do you have a code snippet we can review that will work since this is generic for all ble devices?

dariuszseweryn, 2020-05-03:
I was on holidays. The rough example would be:
```
device.connect()
  .then(() => device. discoverAllServicesAndCharacteristics())
  .then(() => device.services())
  .then(services => /* do your stuff */)
```

closed 2020-05-03 by dariuszseweryn