dotintent/react-native-ble-plx #615: I want to use react-native-ble-plx with the expo android app

riegaz, 2020-02-15:
I install nodejs.

`npm install -g expo-cli`
`expo init testapp`

`npm install react-native-ble-plx --save`
`react-native link react-native-ble-plx`

Can I run 

`expo start` afterwards and use the bluetooth addon via the expo android app or is this not possible? 

Or do I need to run it as 

`react-native run-android`

dariuszseweryn, 2020-02-16:
In readme you have installation instructions https://github.com/Polidea/react-native-ble-plx#configuration--installation

riegaz, 2020-02-16:
Thanks,

I followed this guide. However, I get error messages after linking. Can someone tell me if an external plugin like bluetooth can be used with the expo start command? Or isn't this the bare workflow?

dariuszseweryn, 2020-02-17:
Currently you need to use a bare workflow if you want to use BLE with Expo. See from https://expo.canny.io/feature-requests/p/bluetooth-api:

> Evan Bacon
> 
> TL;DR: We cannot add BLE to the Expo client in the near future but you can use other libraries for this in the bare-workflow today. We are working on improving this and will let you know when we have updates.
> 
> 👋🏻 As you may have seen iOS 13 introduced a much stricter set of rules around the Bluetooth permission: https://www.theverge.com/2019/9/19/20867286/ios-13-bluetooth-permission-privacy-feature-apps
> 
> In order to keep the Expo client in the App Store we've had to slow down the addition of new (protected) features. This is pretty disappointing for me personally as I've been working on the expo-bluetooth package for a while now https://github.com/expo/expo/pull/3131 🙃
> 
> BUT because we obviously want this and all awesome native features to be accessible through the Expo SDK we've been working on making the bare-workflow have all of the great features users have come to expect (the last remaining modules to build are OTA updates and notifications). You can read more about the bare-workflow here: https://blog.expo.io/you-can-now-use-expo-apis-in-any-react-native-app-7c3a93041331
> 
> You can start using the bare-workflow today and install any third-party BLE solution like react-native-ble-plx or react-native-ble-manager. Here is an example of doing that: https://github.com/Cierpliwy/SensorTagExpo
> 
> We are actively optimizing the bare-workflow so other critical features like in-app purchases and native Firebase that require build-time configuration will be available with the Expo SDK too!
> 
> We are also working on a new service for creating custom builds in the cloud so windows users will be able to utilize all of these great tools as well! You can learn more about the existing adhoc build functionality here: https://docs.expo.io/versions/latest/guides/adhoc-builds/
> 
> Thank you for your patience.
> 
> To stay updated with these changes you can follow:
> 
> Expo Blog: https://blog.expo.io
> 
> Twitter: https://twitter.com/expo
>
> October 19, 2019

I.e. you definitely need to do `expo eject` something similar to use bare workflow. Consult Expo documentation.

closed 2020-04-25 by stale[bot]

gregfenton, 2022-02-24:
I suspect this issue needs to be revisited now that there is Expo EAS.

`expo eject` is no longer necessary for other native modules, so I don't think it should be necessary here.