dotintent/react-native-ble-plx #617: startDeviceScan no working

henriquemarioto, 2020-02-20:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

Hello, I'm working on a simple project using HC-05, making only the connection to it and a button to turn a relay on and off.

## Current Behavior

It's something very strange, basically I have a problem with startDeviceScan, where I study it usually works, as soon as I have the bluetooth on my device (Android) it works normally showing the devices found, but when I get home, it just doesn't work, the variable indicates that "PoweredON" is actually on, but the scan simply does not start, and as soon as I disable bluetooth, it returns an error saying that bluetooth is off. Sometimes it is the opposite, it works in my house but where I study not, sometimes in neither place, I have already been two unable to use my app because of that. My biggest fear is that it will happen on the day of the final presentation.

https://imgur.com/a/oPIc5Mj

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. step 1
2. step 2
3. ...

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* React Native: 0.60.5
* Platform: Android
* Contents of the `package.json` file: 
```json
{
  "name": "IceFiApp",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint ."
  },
  "dependencies": {
    "react": "16.8.6",
    "react-native": "0.60.5",
    "react-native-ble-plx": "^1.1.1"
  },
  "devDependencies": {
    "@babel/core": "7.8.4",
    "@babel/runtime": "7.8.4",
    "@react-native-community/eslint-config": "0.0.3",
    "babel-jest": "24.9.0",
    "eslint": "6.8.0",
    "jest": "24.9.0",
    "metro-react-native-babel-preset": "0.54.1",
    "react-test-renderer": "16.8.6"
  },
  "jest": {
    "preset": "react-native"
  }
}
```

```js
async componentWillMount() {
    await requestLocationPermission();
    await requestCoarseLocationPermission();
    const subscription = this.manager.onStateChange((state) => {
      if (state === 'PoweredOn') {
          this.setState({
            powered: state,
          });
          this.scanAndConnect();
      }
      else{
        this.setState({
          powered: state,
          count: 0,
          deviceName: 'No Device',
          time: 0,
          scan: 'No scanning...',
          charac: '',
        });
      }
  }, true);
  }

scanAndConnect() {
    if(this.state.powered === 'PoweredOn'){
      this.manager.startDeviceScan(null, null, (error, device) => {
        if (error) {
            // Handle error (scanning will be stopped automatically)
            {Alert.alert('Erro no scan: ' + error)}
            return
        }

        this.setState({
          deviceName: device.name,
          scan: 'Scaning...',
          time: this.state.time+1,
        });
        
        if(this.state.time >= 100)
        {
          this.manager.stopDeviceScan();
          this.setState({
            scan: 'Time out',
            time: 0,
          })
        }

        // Check if it is a device you are looking for based on advertisement data
        // or other criteria.
        if (device.name === 'HC-05' || device.name === 'HC05' || device.name === 'HC 05' || 
        device.name === 'hc-05' || device.name === 'hc05' || device.name === 'hc 05'){
            // Stop scanning as it's not necessary if you are scanning for one device.
            this.manager.stopDeviceScan();
            this.setState({
              deviceName: device.name,
              scan: 'HC-05 found',
            });
           
            device.connect()
            .then((device) => {
                this.charac("Discovering services and characteristics");
                return device.discoverAllServicesAndCharacteristics()
            })
            .then((device) => {
              this.charac(device.id);
              // Do work on device with services and characteristics
            })
            .catch((error) => {
                {Alert.alert('Erro ao conectar ou caracteristicas: ' + error)}
                // Handle errors
            });
            
        }
      });
    }
    else{
      {Alert.alert('Turn on the bluetooth')};
    }
  }
```

dariuszseweryn, 2020-02-20:
Hello. Please add native logs as the template asks you to. Also please provide info about the phone you are testing on.

henriquemarioto, 2020-02-20:
Hello, the device I use is a Samsung Galaxy S7 Neo, Android 9.
How can I do to generate a log? because it does not generate any type of error.

dariuszseweryn, 2020-02-20:
> Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via setLogLevel function call.

Platform logs on Android can be taken from `Logcat` in Android Studio or otherwise (Google will help you)

henriquemarioto, 2020-02-21:
I'm where I study and like I said, it's working here, but I noticed something else. I tested it several times and it wasn't working, but as soon as I started 'run-android' on my phone, the project worked normally. As soon as I get home I will do another test.

https://imgur.com/a/uOofEsW

```
2020-02-20 21:54:19.163 1870-1870/com.icefiapp D/ViewRootImpl@5ced781[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
2020-02-20 21:54:19.163 1870-1870/com.icefiapp D/InputMethodManager: prepareNavigationBarInfo() DecorView@d814026[MainActivity]
2020-02-20 21:54:19.164 1870-1870/com.icefiapp D/InputMethodManager: getNavigationBarColor() -855310
2020-02-20 21:54:20.387 1870-1870/com.icefiapp W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["StateChangeEvent","Resetting"])
2020-02-20 21:54:20.388 1870-1870/com.icefiapp W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["StateChangeEvent","Resetting"])
2020-02-20 21:54:20.392 1870-1870/com.icefiapp W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["StateChangeEvent","Resetting"])
2020-02-20 21:54:20.723 1870-1912/com.icefiapp D/BluetoothAdapter: onBluetoothStateChange: up=true
2020-02-20 21:54:20.723 1870-1912/com.icefiapp D/BluetoothAdapter: onBluetoothStateChange: Bluetooth is on
2020-02-20 21:54:20.823 1870-1870/com.icefiapp W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["StateChangeEvent","PoweredOn"])
2020-02-20 21:54:20.824 1870-1870/com.icefiapp I/chatty: uid=10236(com.icefiapp) identical 1 line
2020-02-20 21:54:20.826 1870-1870/com.icefiapp W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["StateChangeEvent","PoweredOn"])
2020-02-20 21:54:20.909 1870-2588/com.icefiapp D/BluetoothAdapter: STATE_ON
2020-02-20 21:54:20.911 1870-2588/com.icefiapp D/BluetoothAdapter: STATE_ON
2020-02-20 21:54:20.915 1870-2588/com.icefiapp D/BluetoothAdapter: STATE_ON
2020-02-20 21:54:20.915 1870-2588/com.icefiapp D/BluetoothLeScanner: Start Scan with callback
2020-02-20 21:54:20.924 1870-2674/com.icefiapp D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2020-02-20 21:54:21.058 1870-1870/com.icefiapp D/ViewRootImpl@5ced781[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
2020-02-20 21:54:21.058 1870-1870/com.icefiapp D/InputMethodManager: prepareNavigationBarInfo() DecorView@d814026[MainActivity]
2020-02-20 21:54:21.059 1870-1870/com.icefiapp D/InputMethodManager: getNavigationBarColor() -855310
2020-02-20 21:54:21.105 1870-1870/com.icefiapp D/SensorManager: unregisterListener ::   
2020-02-20 21:54:21.290 1870-1870/com.icefiapp D/ViewRootImpl@5ced781[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
2020-02-20 21:54:21.290 1870-1870/com.icefiapp D/InputMethodManager: prepareNavigationBarInfo() DecorView@d814026[MainActivity]
2020-02-20 21:54:21.290 1870-1870/com.icefiapp D/InputMethodManager: getNavigationBarColor() -855310
2020-02-20 21:54:24.117 1870-1870/com.icefiapp D/ViewRootImpl@5ced781[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
2020-02-20 21:54:24.117 1870-1870/com.icefiapp D/InputMethodManager: prepareNavigationBarInfo() DecorView@d814026[MainActivity]
2020-02-20 21:54:24.117 1870-1870/com.icefiapp D/InputMethodManager: getNavigationBarColor() -855310
2020-02-20 21:54:24.158 1870-1870/com.icefiapp D/ViewRootImpl@5ced781[MainActivity]: stopped(false) old=false
2020-02-20 21:54:24.219 1870-1870/com.icefiapp D/SensorManager: registerListener :: 0, K2HH Accelerometer, 66667, 0,  
2020-02-20 21:54:24.254 1870-4055/com.icefiapp D/NetworkManagementSocketTagger: tagSocket(54) with statsTag=0xffffffff, statsUid=-1
```

henriquemarioto, 2020-02-21:
Hello, as I said, my house is not working, here's the log.

```2020-02-21 01:53:23.085 18840-18840/com.icefiapp D/ViewRootImpl@d8dfbab[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
2020-02-21 01:53:23.085 18840-18840/com.icefiapp D/InputMethodManager: prepareNavigationBarInfo() DecorView@9647208[MainActivity]
2020-02-21 01:53:23.085 18840-18840/com.icefiapp D/InputMethodManager: getNavigationBarColor() -855310
2020-02-21 01:53:25.064 18840-18891/com.icefiapp D/BluetoothAdapter: onBluetoothStateChange: up=true
2020-02-21 01:53:25.064 18840-18891/com.icefiapp D/BluetoothAdapter: onBluetoothStateChange: Bluetooth is on
2020-02-21 01:53:25.157 18840-18970/com.icefiapp D/BluetoothAdapter: STATE_ON
2020-02-21 01:53:25.158 18840-18970/com.icefiapp D/BluetoothAdapter: STATE_ON
2020-02-21 01:53:25.160 18840-18970/com.icefiapp D/BluetoothAdapter: STATE_ON
2020-02-21 01:53:25.160 18840-18970/com.icefiapp D/BluetoothLeScanner: Start Scan with callback
2020-02-21 01:53:25.163 18840-18891/com.icefiapp D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2020-02-21 01:53:25.209 18840-18840/com.icefiapp D/SensorManager: unregisterListener ::   
2020-02-21 01:53:26.771 18840-18840/com.icefiapp D/ViewRootImpl@d8dfbab[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
2020-02-21 01:53:26.772 18840-18840/com.icefiapp D/InputMethodManager: prepareNavigationBarInfo() DecorView@9647208[MainActivity]
2020-02-21 01:53:26.772 18840-18840/com.icefiapp D/InputMethodManager: getNavigationBarColor() -855310
2020-02-21 01:53:26.826 18840-18840/com.icefiapp D/ViewRootImpl@d8dfbab[MainActivity]: stopped(false) old=false
2020-02-21 01:53:26.870 18840-18840/com.icefiapp D/SensorManager: registerListener :: 0, K2HH Accelerometer, 66667, 0,  
2020-02-21 01:53:26.909 18840-19761/com.icefiapp D/NetworkManagementSocketTagger: tagSocket(54) with statsTag=0xffffffff, statsUid=-1```

dariuszseweryn, 2020-02-21:
Please enable library logs as described above before copying data from logcat. In what you posted there are no library log enabled/visible.

henriquemarioto, 2020-02-21:
Hello, I tried several ways to enable logging, apparently it is right, but logcat does not seem to return something from Ble Plx.

```js

constructor(props){
    super(props);
    this.manager = new BleManager();
    this.manager.setLogLevel(LogLevel.Verbose);
  }
```

```
2020-02-21 20:35:38.276 23540-23557/? I/FA: Collection disabled with firebase_analytics_collection_enabled=0
2020-02-21 20:35:38.276 23540-23557/? I/FA: App measurement is starting up, version: 12451
2020-02-21 20:35:38.277 23540-23557/? I/FA: To enable debug logging run: adb shell setprop log.tag.FA VERBOSE
2020-02-21 20:35:38.277 23540-23557/? I/FA: To enable faster debug mode event logging run:
      adb shell setprop debug.firebase.analytics.app com.sec.android.app.sbrowser
2020-02-21 20:35:38.331 23540-23540/? D/CustomTabsConnectionService: CustomTabsConnectionService on Bind called
2020-02-21 20:35:38.491 23540-23581/? D/CustomTabsConnection: warmupInternal::doInBackground start
2020-02-21 20:35:38.499 23540-23581/? D/CustomTabsConnection: warmupInternal::doInBackground end
2020-02-21 20:35:38.778 11691-11691/? I/HeadsetConnectionManager: HeadsetConnectionManager already instantiated
2020-02-21 20:35:39.581 23540-23540/? D/CustomTabsConnection: cancelPrerender()
2020-02-21 20:35:40.010 23661-23715/? I/FA: App measurement is starting up, version: 16250
2020-02-21 20:35:40.010 23661-23715/? I/FA: To enable debug logging run: adb shell setprop log.tag.FA VERBOSE
2020-02-21 20:35:40.011 23661-23715/? I/FA: To enable faster debug mode event logging run:
      adb shell setprop debug.firebase.analytics.app com.samsung.android.email.provider
```

dariuszseweryn, 2020-03-26:
Sorry for such a delay – been on holidays and now you know how it goes.
Questions – are you sure that the actual methods are being called each time? Maybe there is some other issue that prevent the lib from being called? If you have enabled logs yet they are still unavailable — it would point to this situation.

himanshuvats1996, 2020-04-03:
Facing the same issue did anyone find the solution ??

closed 2020-06-09 by stale[bot]

zhyafe, 2022-12-09:
react-native-ble-plx does not support classic Bluetooth, The Bluetooth version of hc-05 is v2. Does it work?