dotintent/react-native-ble-plx #620: Dependency cycle with Async Storage

jherman3, 2020-02-25:
## Expected Behavior

The project compiles without a dependency cycle

## Current Behavior

The application builds fine from Xcode, but when building with `yarn ios`, the build fails with the following error:
```
note: Using new build system
note: Planning build
note: Constructing build description
Build system information
error: Cycle in dependencies between targets 'myapp' and 'RNCAsyncStorage'; building could produce unreliable results.
Cycle path: myapp → RNCAsyncStorage → react-native-ble-plx-swift → React-CoreModules → RNVectorIcons → RNScreens → RNReanimated → RNGestureHandler → RNCMaskedView → RNCAsyncStorage → myapp
Cycle details:
Target build order preserved because “Parallelize Build” is off

→ That command depends on command in Target 'myapp': script phase “[CP] Copy Pods Resources”
→ Target 'myapp' has target dependency on Target 'RNCAsyncStorage'
→ Target 'RNCAsyncStorage' has compile command with input '/Users/jake/Developer/myapp/ios/Pods/Target Support Files/RNCAsyncStorage/RNCAsyncStorage-dummy.m'
○ That command depends on command in Target 'react-native-ble-plx-swift': script phase “Copy generated compatibility header”
→ Target 'react-native-ble-plx-swift'
→ Target 'React-CoreModules'
→ Target 'RNVectorIcons'
→ Target 'RNScreens'
→ Target 'RNReanimated'
→ Target 'RNGestureHandler'
→ Target 'RNCMaskedView' has target dependency on Target 'RNCAsyncStorage' due to target order in a “Target Dependencies” build phase or the scheme
→ Target 'RNCAsyncStorage' has compile command with input '/Users/jake/Developer/myapp/node_modules/@react-native-community/async-storage/ios/RNCAsyncStorage.m'
```

### Steps to Reproduce

Compile a project with both `react-native-ble-plx` and `react-native-community/async-storage`

### Context

This is on the latest version for this library and async-storage. The problem seems to be that both this library and async storage create generated files in their build process, and Xcode thinks that they depend on each other. What's strange is that I can build and run the app without problems if I open the Xcode workspace and use Xcode to build it. The problem only occurs when using the CLI to build.

Cierpliwy, 2020-04-16:
Closing as latest version (2.0.0) doesn't depend on `react-native-ble-plx-swift`.

closed 2020-04-16 by Cierpliwy

lifekaizen, 2020-06-03:
I encountered this error today and this is what I did to fix it.

Upgrade react-native-ble-plx to 2.0.0 in package.json:
```
{  "dependencies": { "react-native-ble-plx": "^2.0.0" }}
```

Remove old reference from `./ios/Podfile` :
`pod 'react-native-ble-plx-swift', :path => '../node_modules/react-native-ble-plx'`
_Based on [this post](https://github.com/Polidea/react-native-ble-plx/issues/336#issuecomment-423382933)._

Refresh pods:
```
pod deintigrate
pod install
```