dotintent/react-native-ble-plx #621: How to get updated rssi?

SimoneCarnio, 2020-02-26:
I'm populating an array with scanned device and if already present, I  update the rssi value, but some devices is scanned every second, other after a lot of seconds.
How to get "near to realtime" rssi values for all scanned devices (not connected)?

To understand, I would like to replicate a list like the main page of LightBlue where rssi is updated every in realtime for every row

[image]

diegogurpegui, 2020-03-18:
You will need to ping that value every X seconds (or any time frame that makes sense for you).
You can create your own subscription-like functions.

SimoneCarnio, 2020-03-31:
thank, will try tomorrow

Cierpliwy, 2020-04-16:
Closing for now. Please reopen if there are any issues.

closed 2020-04-16 by Cierpliwy