dotintent/react-native-ble-plx #623: Bug Report

Bonfahh, 2020-03-02:
- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

I should receive a response from the device.

## Current Behavior

When I try to write on a characteristic in IOS I get Characteristic XXXXXXXXXX write failed for device XXXXXXX

### Steps to Reproduce

* Platform: iOS.
* JS logs: Reactotron Characteristic XXXXXXXXXX write failed for device XXXXXXX

const bleManager = new BleManager();
      bleManager.onStateChange(async state => {
        if (state === 'PoweredOn') {
          const device = await bleManager.connectToDevice(deviceId);
          await device.discoverAllServicesAndCharacteristics();
          const details = await device.characteristicsForService(
            '0000ffe0-0000-1000-8000-00805f9b34fb'
          );
            await details[0]
              .writeWithResponse(btoa('a'))
              .then(async char => console.tron.log(await char));
        }
      });

dariuszseweryn, 2020-03-12:
https://github.com/Polidea/react-native-ble-plx/wiki/=-FAQ:-I-got-a-BleError.-Now-what%3F

closed 2020-03-12 by dariuszseweryn

juanmartin, 2020-08-18:
I stumbled upon a similar problem in production build.
The problem in my case was that **btoa** isn't available without running through browser debugger, so I had to install base-64 from npm and import it as so:
`import {decode as atob, encode as btoa} from 'base-64' `
thus avoiding refactoring code.

This was crazy since I couldn't debug in production. I was able to connect but it failed sending messages saying that it couldn't discover characteristics (???). Then I [realised](https://stackoverflow.com/questions/42829838/react-native-atob-btoa-not-working-without-remote-js-debugging/51525605#51525605) that without debugging activated btoa "wasn't found".