dotintent/react-native-ble-plx #624: Missing data in characteristic notification

sandberg75, 2020-03-13:
## Expected Behavior
notification response data should contain same data on both android and iOS version.

## Current Behavior
When I receive notifications the result in the iOS app is incorrect, data is missing. Android works correct.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. connect with device, discoverAllServicesAndCharacteristics
2. monitorCharacteristicForService
3. writeCharacteristicWithResponseForService to receive notifications for temperature datapoints
4. First notification on iOS is missing data, see below for output:
The expected number of temperature points is 9700 = 151, which sums up nice in the android response. Each temperature point is 2 bytes. I have marked with BOLD the points that only exists in the android response.

iOS first notification:
2b0300ed755b5e9700 -- header, temperature points efter this
b308b308b008b208b008af08b108b008ad08ac08ac08aa08aa08ab08ab08aa08ab08aa08ab08a908a908ad08a908ab08ab08ab08ab08ab08ab08ac08ab08ac08ac08ab08ac08ab08ab08ac08ab08ac08ac08ab08ab08aa08ac08ab08ab08a908ac08ab08aa08a708a808a808a808a908a708a708a608a608a708a608a608a608a508a808a608a308a608a608a608a708a608a608a608a708a608a708a708a808a708a608a408a508a508a608a4

iOS second notification, more temperature points:
089b089b089b0899089b089808990898089a089808970897089608940894089508930894089408940895089308960895089508970896089708980897089a0898089808

Android first notification
2b0300ed755b5e9700
b308b308b008b208b008af08b108b008ad08ac08ac08aa08aa08ab08ab08aa08ab08aa08ab08a908a908ad08a908ab08ab08ab08ab08ab08ab08ac08ab08ac08ac08ab08ac08ab08ab08ac08ab08ac08ac08ab08ab08aa08ac08ab08ab08a908ac08ab08aa08a708a808a808a808a908a708a708a608a608a708a608a608a608a508a808a608a308a608a608a608a708a608a608a608a708a608a708a708a808a708a608a408a508a508a608a4**08a408a508a508a408a408a508a608a208a208a308a208a308a208a208a208a008a008a208a208a208a108a0089c08a0089e089d089e089e089d089b089e**

Android second notification
089b089b089b0899089b089808990898089a089808970897089608940894089508930894089408940895089308960895089508970896089708980897089a0898089808 

### Context
I have tested the device with nRF Connect app and it receives correct data. Both on android and iOS.
Any help would be greatly appreciated!

dariuszseweryn, 2020-03-13:
I believe you are getting appropriate data in terms of what the OS serves you. iOS supports MTU 180-sh, and your first notification is around the same size where Android may negotiate bigger MTU.

sandberg75, 2020-03-13:
Yes, I just found this issue, https://github.com/Polidea/react-native-ble-plx/issues/554 talking about almost the same issue.
So it is the client that is sending to large data packages?

dariuszseweryn, 2020-03-13:
Seems so.

sandberg75, 2020-03-13:
Thank you for your fast reply!
I guess you can close this issue, maybe I should have labeled it as a question instead of bug, sorry for that.

dariuszseweryn, 2020-03-13:
You're welcome

closed 2020-03-13 by dariuszseweryn