dotintent/react-native-ble-plx #627: save the device into state for later use

sajadghawami, 2020-03-18:
Hey There,

what i wanted to do is, i want to connect to a device, and then save it to the state to make use of it later.
It seems like whenever i save it, i need to call `.connect()` and `.discoverAllServicesAndCharacteristics()` everytime i want to interact with the device. Is this the expected behaviour?

```
  const [selectedDevice, setSelectedDevice] = useState(null);

  const handleClickConnect = async dev => {
    const connectedDevice = await dev.connect();
    const device = await connectedDevice.discoverAllServicesAndCharacteristics();
    const services = await device.services();
    setSelectedDevice(device); <------- save the device to the state
  };

/**
* does not work
*/ 
  const handleClickCharacteristic = async (serviceUuid, characteristicUuid, value) => {
    const valueBase64 = new Buffer(value).toString('base64');

    // here the selectedDevice cannot be used, unless i do .connect and .discoverAllServicesAndCharacteristics again 
    const response = await selectedDevice.writeCharacteristicWithResponseForService(
      serviceUuid,
      characteristicUuid,
      valueBase64
    );
  };

/**
* works
*/ 

  const handleClickCharacteristic = async (serviceUuid, characteristicUuid, value) => {
    const connectedDevice = await selectedDevice.connect();
    const device = await connectedDevice.discoverAllServicesAndCharacteristics();

    const valueBase64 = new Buffer(value).toString('base64');

    const response = await connectedDevice.writeCharacteristicWithResponseForService(
      serviceUuid,
      characteristicUuid,
      valueBase64
    );
  };

```

Whats the recommended best practice here? 

By the way, i use hooks as you might have seen.

Cierpliwy, 2020-04-16:
Can you tell me where `BleManager` instance is created?

closed 2020-06-23 by stale[bot]

vitalNohj, 2025-03-03:
I'm also wondering about this. My device doesn't know it's disconnected. Maybe it's because I'm using Metro and Expo but not sure. Is there a way to save the device itself and reconnect to it without it advertising again (because it thinks it's connected)?