dotintent/react-native-ble-plx #628: Build failed on RN v0.61.5

chandrasekarvogo, 2020-03-25:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Please describe the question.

getting 
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* Where:
Build file '/Users/chandrasekar/workspace/hybrid-consumer-app/node_modules/react-native-ble-plx/android/build.gradle' line: 20

* What went wrong:
A problem occurred evaluating project ':react-native-ble-plx'.
> Failed to apply plugin [id 'groovyx.android']
   > Could not create an instance of type groovyx.internal.DefaultAndroidGroovySourceSet.
      > Could not find matching constructor for: org.gradle.api.internal.file.DefaultSourceDirectorySet(org.codehaus.groovy.runtime.GStringImpl, org.gradle.api.internal.file.BaseDirFileResolver, org.gradle.api.internal.file.collections.DefaultDirectoryFileTreeFactory)

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
A problem occurred configuring project ':react-native-ble-plx'.
> compileSdkVersion is not specified.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 4s

    at checkExecSyncError (child_process.js:621:11)
    at execFileSync (child_process.js:639:15)
    at runOnAllDevices (/Users/chandrasekar/workspace/hybrid-consumer-app/node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:94:39)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)

vaqif14, 2020-03-25:
In **android/app/build.gradle** 
add 
`android {
    ....
    defaultConfig {
        ...
        multiDexEnabled true
    }`

chandrasekarvogo, 2020-03-26:
@vaqif14 still the same issue.

TheHothihi, 2020-03-27:
@chandrasekarvogo Same problem here.
According to 
https://github.com/groovy/groovy-android-gradle-plugin/commit/a336b37fed8d9dbad1087fe010c6a03deb8fdad1

The problem seemed to be coming from groovyx rather than RN / react-native-ble-plx.

ScreamZ, 2020-03-27:
Same issue here, I just spotted it after migrating to 0.62.0

ild-nucleusstudio, 2020-03-27:
I also got the same error with 0.62.0 today.
Seems like as @ScreamZ pointed out.
Could you guys take a look at it @dariuszseweryn @Cierpliwy ? I really appreciate that.

therustydev, 2020-03-28:
Same issue here, also on 0.62.0

chancmg, 2020-03-28:
seems gradle v6.0 support added in [Gradle 6 compatibility](https://github.com/groovy/groovy-android-gradle-plugin/pull/177 ) and release in [groovy-android-gradle-plugin@3.0.0](https://github.com/groovy/groovy-android-gradle-plugin/releases/tag/RELEASE_3_0_0). This library running in older version. upgrading **groovy-android-gradle-plugin** might help. @Cierpliwy @dariuszseweryn

purrsong-km, 2020-03-28:
I also got same issues with RN 0.62
`react-native link react-native-ble-plx` doesn't work for android.

therustydev, 2020-03-29:
> seems gradle v6.0 support added in [Gradle 6 compatibility](https://github.com/groovy/groovy-android-gradle-plugin/pull/177 ) and release in [...]

Could you please point me to a resource to do that?

bwindsor, 2020-03-29:
Downgrading my project to gradle 5.6.4 worked as an interim solution for me. In Android Studio, go to File >> Project Structure. On the project tab change Gradle version to 5.6.4. This modifies `gradle/wrapper/gradle-wrapper.properties` to have `distributionUrl=https\://services.gradle.org/distributions/gradle-5.6.4-all.zip`.

ezranbayantemur, 2020-03-30:
Same problem at 0.62

ild-nucleusstudio, 2020-03-31:
Here is my workaround for those who need this package run with latest react native version 0.62.0

As the problem is pointed out with Gradle version < 6, so we should upgrade the build tool and groovy plugin.

Here is a step by step:

1/ Fork this repository
2/ In android folder, open bundle.grade:
Update this code:

```
buildscript {
    repositories {
        jcenter()
        google() // This for build tool
        maven {
          url "https://plugins.gradle.org/m2/"
        }
    }
    dependencies {
         classpath 'com.android.tools.build:gradle:3.6.0' // Update build tool gradle
        classpath 'gradle.plugin.org.codehaus.groovy:groovy-android-gradle-plugin:3.0.0' // Update groovy for gradle 6+
    }
}

apply plugin: 'com.android.library'
apply plugin: "org.codehaus.groovy.android" // Update this
```

2/ Open gradle/wrapper/gradle-wrapper.properties, update this code:

```
distributionUrl=https\://services.gradle.org/distributions/gradle-6.2-all.zip  // You can choose your gradle version 6+
```

3/ Commit and push the code to Github
4/ Install the package with GitHub packager or you can use npm-link (or yalc) to test it locally.

Note: This is a workaround for this issue while we wait for the author updates his awesome library.
Note: For IOS, we can keep the same or update MultiplatformBleAdapter to the latest version if you want.

Upvote for this comment if it works for your project.

strawberry-code, 2020-04-01:
In a more simple way I solved using a solution very similar to [the the previous one](https://github.com/Polidea/react-native-ble-plx/issues/628#issuecomment-606524308):

1.  Edit **./root/android/build.gradle**

```
buildscript {
    repositories {
        jcenter()
        google() // This for build tool
        maven {
          url "https://plugins.gradle.org/m2/"
        }
    }
    dependencies {
         classpath 'com.android.tools.build:gradle:3.6.0' // Update build tool gradle
        classpath 'gradle.plugin.org.codehaus.groovy:groovy-android-gradle-plugin:3.0.0' // Update groovy for gradle 6+
    }
}

apply plugin: 'com.android.library'
apply plugin: "org.codehaus.groovy.android" // Update this
```

2.  Edit **./roo/android/gradle/wrapper/gradle-wrapper.properties**

```
distributionUrl=https\://services.gradle.org/distributions/gradle-6.3-all.zip // Because I have latest gradle version
```

I confirm after that I can finally build on Android and now I can use bluetooth plugin APIs.

NRamirezCR, 2020-04-02:
> In a more simple way I solved using a solution very similar to [the the previous one](https://github.com/Polidea/react-native-ble-plx/issues/628#issuecomment-606524308) (no forks required):

I tried this but it didn't work. Can you provide a repo? Meanwhile, I edited the Gradle files as per @ild-nucleusstudio suggestion and it worked.

marza91, 2020-04-08:
Now that there's a pull request available, I just did this to make it work in my project:

```
cd ..

# Make sure it's ok do write the following repo to the project's parent folder
git clone git@github.com:chancmg/react-native-ble-plx.git

# Go back into the project dir
cd mybleproject

# Install the project from parent dir
npm install $(npm pack ../react-native-ble-plx | tail -1) --save
```

AlexIII, 2020-04-10:
Method suggested by @strawberry-code works, but to be more clear you should edit `build.gradle` in `node_modules\react-native-ble-plx\android\` and `android/gradle/wrapper/gradle-wrapper.properties` in your project directory.
Bear in mind, this is VERY temporary solution, all the changes in `node_modules` will be erased after the update and whatnot.

azuxx, 2020-04-14:
> Downgrading my project to gradle 5.6.4 worked as an interim solution for me. In Android Studio, go to File >> Project Structure. On the project tab change Gradle version to 5.6.4. This modifies [...]

Downgrading solution worked for me. Anyhow, waiting to get the plugin being supported with Gradle 6.
Thanks🙏🏻

Cierpliwy, 2020-04-16:
Please check version `2.0.0`. The issue should be solved.

petr166, 2020-04-16:
I can confirm that it works on  

```
"react-native": "0.62.2",
"react-native-ble-plx": "^2.0.0",
```

Cierpliwy, 2020-04-16:
Glad to hear that.

closed 2020-04-16 by Cierpliwy