dotintent/react-native-ble-plx #630: Cannot build Android project

juanjo-ramos, 2020-03-28:
## Question
I have created a new project where I want to try the library. **I can build & run with no problems on iOS**. When trying to build on Android I see the following error:
```
FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':react-native-ble-plx'.
> No such property: variantManager for class: com.android.build.gradle.LibraryPlugin
```

### Environment
**React Native** : 0.61.6

**Android SDK**
[image]

**build.gradle**
```
buildscript {
    ext {
        buildToolsVersion = "28.0.3"
        minSdkVersion = 21
        compileSdkVersion = 28
        targetSdkVersion = 27
        supportLibVersion = "28.0.0"
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.6.1'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenLocal()
        google()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }
        maven { url 'https://jitpack.io' }
    }
}

wrapper {
    gradleVersion = '4.7'
    distributionUrl = distributionUrl.replace("bin", "all")
}
```

pakoito, 2020-03-28:
Someone else has reported it on StackOverflow too: https://stackoverflow.com/questions/60811083/no-such-property-variantmanager

purrsong-km, 2020-03-29:
I also have a same problem.

juanjo-ramos, 2020-03-29:
It turned out to be a versioning problem. Using [React Native 0.60.5](https://github.com/Cierpliwy/SensorTag/blob/master/package.json#L16) and [gradle 3.4.1](https://github.com/Cierpliwy/SensorTag/blob/master/android/build.gradle#L16) as in the example setup fixed the problem for me.

closed 2020-03-29 by juanjo-ramos