dotintent/react-native-ble-plx #631: Type for ScanOptions.callbackType is incorrect

bwindsor, 2020-03-29:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x ] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior
The type for ScanOptions.callbackType should be `callbackType?: ScanCallbackType`

## Current Behavior
The type is `callbackType?: boolean`

### Steps to Reproduce
Create a typescript file with this content:
```ts
import {ScanOptions, ScanMode, ScanCallbackType} from 'react-native-ble-plx';

const scanOptions: ScanOptions = {
  allowDuplicates: false,
  scanMode: ScanMode.LowLatency,
  callbackType: ScanCallbackType.FirstMatch,
};
```

Typescript will fail with an error TS2322: Type 'ScanCallbackType.FirstMatch' is not assignable to type 'boolean | undefined'

### Context
* Typescript version: 3.8.3
* Library version: 1.1.1

Cierpliwy, 2020-04-16:
I applied your fix in `2.0.0`. I'm sorry I didn't merge it directly.

closed 2020-04-16 by Cierpliwy

bwindsor, 2020-04-18:
Thank you, this works after upgrading :)