dotintent/react-native-ble-plx #634: Cannot see Android devices (not interested in connections)

strawberry-code, 2020-04-02:
Hi, I would like clarification on device scanning. I state one thing: I am not interested in establishing connections with the devices, it is enough for me to know that they are present. I am using your library and I have tried it both on Android devices (Samsung J5 and Huawei P8) and on iOS devices (two iPhone 8 and one iPad) but also on laptops (Lenovo Thinkpad and two MacBook Pro) plus various non-devices personal (Smart TV, home speakers, smart watch and so on.

I noticed that not all devices are capable of scanning each other. I am using the startScan function, for example with the Samsung J5 I can see all the IoT devices and all the iOS devices but I cannot see the Huawei. Worse still, with iPhones I can see all IoT devices, but I can't see any of the Android smartphones.

What are the causes? What can I do to solve this problem (if it were possible)? Are there other libraries that would allow me to see the signal from other devices?

bemaverick, 2020-04-02:
looks like this library is not support classic Bluetooth devices (such as smartphones or iphones)
Only BLE

strawberry-code, 2020-04-02:
@bemaverick thanks for reply. Do you know any alternative library that is able to scan classic bluetooth also?

Voxoff, 2020-04-06:
@strawberry-code I'm new to this but AFAIK bemaverick isn't quite right. While this library only supports BLE, smartphones (and iphones) also support BLE. 'All iOS devices since the iPhone 4S support BLE' was a google away.

Hope that helps - I'm attempting to use the library to identify nearby smartphones as well

Cierpliwy, 2020-04-16:
Most of devices advertise as Bluetooth Low Energy peripheral, but it's not guaranteed.

closed 2020-06-23 by stale[bot]