dotintent/react-native-ble-plx #635: Error to get All services and cheracteristcs from ble device using Polidea/react-native-ble-plx

renatoramossilva, 2020-04-02:
Hi I am trying to connect a bluetooth device using the Polidea/react-native-ble-plx library.

First I do the scan and it works fine:

```
scanAndConnect() {
console.log('scanAndConnect');
this.manager.startDeviceScan(null, null, (error, device) => {
//console.log('Scanning...' + error);
if (error) {
// Handle error (scanning will be stopped automatically)
console.log('Scanning ERROR');
return;
}
if ((device.id=== '54:6C:0E:A0:7B:6D') 
{
console.log('device found: ' + device.name + '(' + device.id + ')');
this.manager.stopDeviceScan();
}
}
```
The second step I try to connect. And it s works too!

```
device.connect().then((dev) => {
          console.log("try to connect")          
        }).then((dev) => {
          console.log("success to connect")
        }).catch((error) => {
          console.log("connect error:" + error)
          throw error
        });
```
So, now I want to get all the services and charactrists to read some informations from device, but this feature doesnt work for me.

I am using this method device.discoverAllServicesAndCharacteristics();

And I always received an error.

Some time I receive the follow error message: 1) WARN Possible Unhandled Promise Rejection (id: 0): BleError: Device 54:6C:0E:A0:7B:6D is not connected

I didnt understand the first one, because I have the device connected.

`var status = this.manager.isDeviceConnected(device.id).then((status)=>console.log('status1: '+status))`
the line above returns false before I call the connect method and it returns true after it.

But when I try to get the services and characteristics I have the message that the device isnt connected

Cierpliwy, 2020-04-16:
Do you call `discoverAllServicesAndCharacteristics` after `connect` using `await`? I can't see it in your example.

renatoramossilva, 2020-04-16:
My connection function:

```
 this.manager.connectToDevice(device.id).then((device) => {
        this.manager.discoverAllServicesAndCharacteristicsForDevice(device.id)
        .then((results) => { 
            console.log(results);
            ...
```
I am not using await in my code. When need I use await?

Cierpliwy, 2020-04-17:
No, await just make it easier to write asynchronous code in synchronous way. You example looks good. Do you get an error in this case? Can you show me more logs?

closed 2020-06-23 by stale[bot]