dotintent/react-native-ble-plx #636: Project Setup

mayooresan, 2020-04-03:
## Setup problem

I'm trying to setup the library with React Native .62. I followed the steps and updated the podfile. Later when I do `pod install` or `pod update`, I get the following error. I have worked with previous versions of react-native but never come across this issue. 

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. create new react-native app with version .62
2. follow the steps on readme
3. do `pod intall`

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: 1.1.1
* Platform: iOS.
* Platform logs (logcat/XCode): n/a
* JS logs:
```
[!] Unable to find a specification for `react-native-ble-plx-swift` depended upon by `react-native-ble-plx`

You have either:
 * out-of-date source repos which you can update with `pod repo update` or with `pod install --repo-update`.
 * mistyped the name or version.
 * not added the source repo that hosts the Podspec to your Podfile.
```
* Contents of the `package.json` file:
```
{
  "name": "Covid19",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint ."
  },
  "dependencies": {
    "react": "16.11.0",
    "react-native": "0.62.0",
    "react-native-ble-plx": "^1.1.1"
  },
  "devDependencies": {
    "@babel/core": "^7.9.0",
    "@babel/runtime": "^7.9.2",
    "@react-native-community/eslint-config": "^1.0.0",
    "babel-jest": "^25.2.6",
    "eslint": "^6.8.0",
    "jest": "^25.2.6",
    "metro-react-native-babel-preset": "^0.59.0",
    "react-test-renderer": "16.11.0"
  },
  "jest": {
    "preset": "react-native"
  }
}
```

stephenhouser, 2020-04-16:
The README here is incorrect. The following [article](https://blog.expo.io/so-you-want-to-build-a-bluetooth-app-with-react-native-and-expo-6ea6a31a151d) shows you have to add two lines to the `Podfile` on iOS

```
pod 'react-native-ble-plx', :path => '../node_modules/react-native-ble-plx'
pod 'react-native-ble-plx-swift', :path => '../node_modules/react-native-ble-plx'
```

Cierpliwy, 2020-04-16:
We released version `2.0.0` which is tested to be compatible with 0.62.2. No API changes introduced, just build changes.

closed 2020-06-23 by stale[bot]