dotintent/react-native-ble-plx #637: Feature Request

ParshantArora, 2020-04-04:
Hi my need is to get static Bluetooth id during scanning. So that  i can differentiate a particular persons iphone/ipad  based on that id . But as you know apple provide us UUID which is not constant or static  which changes during every session.

On android we are getting static MAC address
In short i need a constant id that is throughout the app or phone in iOS

Harneetk, 2020-04-05:
Hi
I also have same requirement in my application. Is there any method in library which gives advertising packet from where unique ID can be fetched? Sample code will be highly appreciated here. Thanks

Cierpliwy, 2020-04-16:
It's not possible on iOS. What you can do is to insert your own unique value inside `Manufacturer Data` record, put unique service UUID or service data (on peripheral side).

ParshantArora, 2020-04-18:
Yes Solved the problem by setting service UUID data in advertising data

closed 2020-04-18 by ParshantArora