dotintent/react-native-ble-plx #638: Bug Report

ParshantArora, 2020-04-04:
Hi 
I just want to calculate the distance between the targeted BLE devices and the source scanning device .  
Some people gave solution over internet  to use rssi and txvalue for calculating the same 
But i am not getting Tx power value in case of every device. 

Can Any one help me out on this ? Need a solution on thew immediate basis

Cierpliwy, 2020-04-16:
> But i am not getting Tx power value in case of every device.

You can't do anything about it. It's not required to put TX power value in an advertisement data.

ParshantArora, 2020-04-18:
Yes @Cierpliwy Do you another way to calculate the distance between  targeted BLE devices and the source scanning device?
 .

closed 2020-06-24 by stale[bot]