dotintent/react-native-ble-plx #640: Help request in background mode on iOS

strawberry-code, 2020-04-05:
Hi everyone, I am writing to ask you for help. I have been trying to deal with a problem running in the background on iOS for two days now and I have not yet been able to solve it. I hope some of you can help me.

I am using:

- Xcode Version 11.3.1
- iOS 13.3.1
- React-Native 0.62.0
- react-native-ble-plx ^ 1.1.1

I have read the `react-native-ble-plx` documentation page in depth ([Background mode (iOS)](https://github.com/Polidea/react-native-ble-plx/wiki/Background-mode-(iOS))), but I'm not sure I understand exactly all the steps, in particular regarding the background task identifier registration (task identifier **restoreStateIdentifier** and **restoreStateFunction**, when building a new **BleManager**).

I don't want to perform complex tasks in the background: I simply need to perform the `this.manager.startDeviceScan (...)` function regularly. In fact, I am finding myself in the exact same situation as the example of the automatic door lock, an example cited in the Apple Developer documentation ([Performing Long-Term Actions in the Background](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html) or read the quote at the bottom: _Performing Long-Term Actions in the Background_).

In particular, I import the library, initialize **BleManager** in the constructor and use as a task identifier '**BleInTheBackground**' and then use as a **restorStateFunction** a function that reproduces a sound in the background (for sounds I use **react-native-track-player** and I have seen that it works in background), an alert and a console.log. I noticed that this function (**resotreStateFunction**) is called after a long random time, even when the App is in the foreground.

The mechanism of registering the task identifier in **info.plist** is not entirely clear to me: is it necessary to run the `this.manager.startDeviceScan` function in the background? Also, as far as I've said so far, I don't have a clear idea of how to scan in the background at regular intervals (for example, 5 or 10 seconds may be fine), by using for example `setInterval` in JS.

If someone has already dealt with this situation or has an idea on how to deal with it, can kindly help me? Possibly providing an example code?

> **Performing Long-Term Actions in the Background**
 Some apps may need to use the Core Bluetooth framework to perform long-term actions in the background. As an example, imagine you are developing a home security app for an iOS device that communicates with a door lock (equipped with Bluetooth low energy technology). The app and the lock interact to automatically lock the door when the user leaves home and unlock the door when the user returns—all while the app is in the background. When the user leaves home, the iOS device may eventually become out of range of the lock, causing the connection to the lock to be lost. At this point, the app can simply call the connectPeripheral:options: method of the CBCentralManager class, and because connection requests do not time out, the iOS device will reconnect when the user returns home.
Now imagine that the user is away from home for a few days. If the app is terminated by the system while the user is away, the app will not be able to reconnect to the lock when the user returns home, and the user may not be able to unlock the door. For apps like these, it is critical to be able to continue using Core Bluetooth to perform long-term actions, such as monitoring active and pending connections.

RashVenkat6795, 2020-04-12:
Did you find a solution to this ? @strawberry-code

Cierpliwy, 2020-04-16:
> I have read the react-native-ble-plx documentation page in depth (Background mode (iOS)), but I'm not sure I understand exactly all the steps, in particular regarding the background task identifier registration (task identifier restoreStateIdentifier and restoreStateFunction, when building a new BleManager).

If you find something understandable after a while, help us to make docs clearer!

> I don't want to perform complex tasks in the background: I simply need to perform the this.manager.startDeviceScan (...) function regularly.

In `Signing & Capabilities`/`Background Modes` enable Uses Bluetooth LE accessories and call 
`startDeviceScan` with your service UUID (you have to specify it). After a while your application should wake up or will be relaunched.

> In particular, I import the library, initialize BleManager in the constructor and use as a task identifier 'BleInTheBackground' and then use as a restorStateFunction 

When your application is killed by the OS and launched again or your device was previously paired and the connection is handled by the OS, you have the opportunity to restore your state when the app launches. That's the purpose of it.

> The mechanism of registering the task identifier in info.plist is not entirely clear to me: is it necessary to run the this.manager.startDeviceScan function in the background?

Background tasks are separate things. They only allow you to work for a longer time while in background mode.

>  I don't have a clear idea of how to scan in the background at regular intervals

You can't control that in the background mode on iOS.

closed 2020-06-23 by stale[bot]