dotintent/react-native-ble-plx #643: Unable to build empty template project

AlexIII, 2020-04-08:
Hi! I'm trying to build empty project with `react-native-ble-plx` installed.
What I do:
1. `npx react-native init myapp  --template react-native-template-typescript`
2. `cd myapp`
3. `npm install --save react-native-ble-plx`
4. `npm run android`

Build fails with the next error message:

```
FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* Where:
Build file '---\myapp\node_modules\react-native-ble-plx\android\build.gradle' li
ne: 20

* What went wrong:
A problem occurred evaluating project ':react-native-ble-plx'.
> Failed to apply plugin [id 'groovyx.android']
   > Could not create an instance of type groovyx.internal.DefaultAndroidGroovySourceSet.
      > Could not find matching constructor for: org.gradle.api.internal.file.DefaultSourceDirectorySet(org.codehaus.gro
ovy.runtime.GStringImpl, org.gradle.api.internal.file.BaseDirFileResolver, org.gradle.api.internal.file.collections.Defa
ultDirectoryFileTreeFactory)

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with
--scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
A problem occurred configuring project ':react-native-ble-plx'.
> compileSdkVersion is not specified.
```

Without installing `react-native-ble-plx` the template project builds and runs just fine.

Hugogsalas, 2020-04-10:
the same probleme here 🤨

AlexIII, 2020-04-10:
I just realized this is a double of #628. Look there for solutions.

closed 2020-04-10 by AlexIII

simolabsoft, 2020-04-13:
any solution ?

AlexIII, 2020-04-13:
@simolabsoft, look at issue #628. Several approaches has been suggested there.