dotintent/react-native-ble-plx #644: Question

BlackCod3, 2020-04-08:
Hi, 
First of all, thanks for this great lib and clear doc.
I just read in the readme that the lib does not allow `communicating between phones using BLE (Peripheral support)`.  I'm a little confuse about the term "communicating" . Can scanning phone devices be considered as a communication ... ? 

I'm very new to BLE APIs and capabilities. I want to achieve a "simple" thing. Let's say I'm developing the APP-X. I want to be able to scan other phone around me, with APP-X installed (exactly like this app here https://www.tracetogether.gov.sg/ ) . So I don't want to establish a connection. I just want to scan and filter the wanted devices via a "simple" metadata (advertising data ? specific for my app) within the signal emitted by the concerned phones. I'm stuck because Idk how to "emit" that custom signal... 

Can you please help me to find a way to do this in RN ?

Many thanks 🙏

Cierpliwy, 2020-04-16:
This library doesn't support Peripheral mode. Please look at existing [tracing implementations](https://github.com/opentrace-community) and try to reuse their knowledge / create RN native modules as needed.

francorobles, 2020-05-03:
@BlackCod3 its not implemented in this library. @Cierpliwy is correct you need to implement your own. Use the concept of https://github.com/opentrace-community.

closed 2020-07-10 by stale[bot]

whatnottodo, 2022-03-05:
@BlackCod3 Did you ever implement or find a library that fit your need?

BlackCod3, 2022-03-05:
@whatnottodo unfortunately no :/