dotintent/react-native-ble-plx #646: Question

Deboracgs, 2020-04-12:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [x ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

I don't know I appears this message on android

`BluetoothLE is unsupported on this device`

and on ios

`unsupported`

I'm using this code

```
componentWillMount() {
    if (Platform.OS === 'ios') {
      this.manager.onStateChange(state => {
        if (state === 'PoweredOn') {
          this.scanAndConnect();
        }
      });
    } else {
      this.scanAndConnect();
    }
  }

  scanAndConnect() {
    this.manager.startDeviceScan(null, null, (error, device) => {
      console.log('Scanning...');
      console.log(device);

      if (error) {
        console.log("error", error)
        return;
      }
}
```

Cierpliwy, 2020-04-16:
Did you check that application has permissions enabled for Bluetooth?

Deboracgs, 2020-04-16:
Yes, I did @Cierpliwy but was only simulator, maybe I need to do a test on device?

Cierpliwy, 2020-04-16:
Simulator on iOS doesn't support BLE, that's why you got unsupported state :)

Deboracgs, 2020-04-16:
@Cierpliwy android too? I test in both

Cierpliwy, 2020-04-16:
I'm not sure about Android. I'm always working with real devices.

dariuszseweryn, 2020-04-16:
[Official documentation of Android Emulator states that Bluetooth is not supported.](https://developer.android.com/studio/run/emulator#limitations)

closed 2020-04-16 by dariuszseweryn