dotintent/react-native-ble-plx #647: Bug Report

kevinswartz, 2020-04-13:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

Awesome library guys!  Thanks for the help.  

I'm connecting to a Secugen Unity 20 bluetooth device (https://secugen.com/products/unity-20-bluetooth/), and trying to read a fingerprint image from it.  I believe I've connected to the device, discovered all services and characteristics, and basic commands seem to work.  I'm trying to communicate by sending my write command with response to one characteristic, then read the data back at a different characteristic.  I've also increased the MTU to 301, which I've seen in some sample code from the manufacturer.

Please describe the behavior you are expecting

## Current Behavior

I'm only receiving a small amount of data within the read characteristic.value, but I think it should be a larger data packet.  In this instance, I get back:
```
deviceID: "............................"
id: 2
isIndicatable: false
isNotifiable: true
isNotifying: false
isReadable: true
isWritableWithResponse: false
isWritableWithoutResponse: false
serviceID: 1
serviceUUID: "0000fda0-0000-1000-8000-00805f9b34fb"
uuid: "00002bb1-0000-1000-8000-00805f9b34fb"
value: "AEMBAAAAwNQBAADZ"
```

Using logcat, I can view the raw data being logged out.  I see this:
```
04-13 12:33:08.720 20247 20355 V RxBle#BleModule: Write to Characteristic(uuid: 00002bb2-0000-1000-8000-00805f9b34fb, id: 4, value: 004301000000000000000044)
04-13 12:33:08.729 20247 20354 D RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(66766085) in 135 ms
04-13 12:33:08.748 20247 20310 D RxBle#ConnectionOperationQueue: QUEUED   CharacteristicReadOperation(181834880)
04-13 12:33:08.752 20247 20354 D RxBle#ConnectionOperationQueue: STARTED  CharacteristicReadOperation(181834880)
04-13 12:33:08.863  4570  4570 D SecStatusBarWifiView: updateState: WifiIconState(resId=2131232670, visible=true, activityId=2131232695)
04-13 12:33:08.864  4570  4570 D SecStatusBarWifiView: updateState: WifiIconState(resId=2131232670, visible=true, activityId=2131232695)
04-13 12:33:09.208 20247 20261 D RxBle#BluetoothGatt$1: onCharacteristicRead characteristic=00002bb1-0000-1000-8000-00805f9b34fb status=0
04-13 12:33:09.212 20247 20355 V RxBle#BleModule: Read from Characteristic(uuid: 00002bb1-0000-1000-8000-00805f9b34fb, id: 2, value: 004301000000C0D4010000D9)
04-13 12:33:09.222 20247 20354 D RxBle#ConnectionOperationQueue: FINISHED CharacteristicReadOperation(181834880) in 470 ms
04-13 12:33:09.243 20247 20310 D RxBle#ConnectionOperationQueue: QUEUED   CharacteristicReadOperation(104263638)
04-13 12:33:09.246 20247 20354 D RxBle#ConnectionOperationQueue: STARTED  CharacteristicReadOperation(104263638)
04-13 12:33:09.377  4246  4419 D WifiStateMachine: enter getWifiLinkLayerStats
04-13 12:33:09.378  4246  4419 I WifiVendorHal: getWifiLinkLayerStats(l.2973) before calling iface.getLinkLayerStats
04-13 12:33:09.378  3796  3796 I android.hardware.wifi@1.0-service: enter getLinkLayerStats
04-13 12:33:09.378  3796  3796 I android.hardware.wifi@1.0-service: enter getLinkLayerStatsInternal
04-13 12:33:09.380  3796  3796 I android.hardware.wifi@1.0-service: Successfully getLinkLayerStats.
04-13 12:33:09.381  4246  4419 I WifiVendorHal: getWifiLinkLayerStats(l.2973) after calling iface.getLinkLayerStats
04-13 12:33:09.390  4246  4419 I WifiStateMachine: checkScoreBasedQuality -  mPreviousScore[0]:84 mPreviousScore[1]:83 mPreviousScore[2]:84 s2Score:84mPrevoiusScoreAverage:83
04-13 12:33:09.486 20247 20261 D RxBle#BluetoothGatt$1: onCharacteristicRead characteristic=00002bb1-0000-1000-8000-00805f9b34fb status=0
04-13 12:33:09.491 20247 20355 V RxBle#BleModule: Read from Characteristic(uuid: 00002bb1-0000-1000-8000-00805f9b34fb, id: 2, value: E2FFF0F1F0F0F0EFEFEEEFEEEEF0EDEBEDE9EAEEEFEFEAE9D5EEF2CDE8DEC6C5E9ECEFECEAEAEBE8E9EEEAD8CFEEE1B5E4E1E4C6E4CDEDEEE2E2F5E0D9E1DCDCC8C8D6D0BCC6D6B8C7CBE0CECCD4C90EADA995000069DCCAC9BC281E59B3C5A20000000000B5E0C1BC300000BFB9990000000000CF9E610E4EC4C7D8C35E00000000BC9D00008EB9B221CC3500001BA5ABACD0B2B86D67B2AB8E92B8A18DB1A3317D7E534A585016B8CBA9DBC8C3BD4A55A24FCFB7B2CF003BABB9A9D33FC3A7B5BBA9BCC7CBCEB76CE7C3ABCAB174ACC5B8D4C8D5E1DCA1D2B6DCE1DEBFE5EACFDEE7E1E2E1DEE6E9E4E7E4E6E9E7EDE4E2E3ECE8E9EAEAD1EFE9F0EBEDEDDFEDF4CEECE9F1F0EEEEE8C9F6F0F2EFF1EEF1F1F1F1F0F0F0F2F0F2F2ECEEF0F2FAFBF0DCF4F1F1ECF0B8E6E4DFFDF1F1EFEFF0F0EAEBECECF0ECEAEDEBE4E4E5ECEDEDECD5D3EDECE8E5D7E6DEEAECEAEAE9EBEBEBEEE5CBAFB3DD91E0E0E3EEDFDFE1E1E4E09AE2D7E2DEE0D7ECDABCD5C7C8B9B87EB4D6CCBBD2ADBDA4A400000093CDDDBB00000004DBCC000000000013EDB5AE0000005DC7880000000000D79C01000069D6C3CF03000000008DB200001242AD7EBA2100000081B3877782C8BAAB8E918A98A7A77FB1A6A179B26167000000AFB335B7CAC3B9AF7EB9037096DEBF000049C2B3C1CAC189B5BE95C0C6C4C8D2D1D663A1CEDBBEAEBCB1CAC0D7DDE1D4C0C6CBD4E4C6DEE0E0D6E2CEDDE1E5E6E7E9E3E0E6E9E5E7E7E3E3E4E4E7EBECE0E5E6F0ECECEDE0E7E9E5EBEFF0F0F0F0F2F2F4F1F0F1EFEDEDF1F1F1EFF2F0F0F3EFE8F0F0F0F0FE9EE1DCF3F0F0F0F7D8EAE1)
```

It looks to me like not all of the data is being received inside the handler.

### Context

I'm on version 1.1.1 of the read-native-ble-plx library.  I'm currently testing against android, but I've not had any luck with ios either.  

Sorry for the newbie question, but that's me right now.  This is my first foray into bluetooth connected devices.

Cierpliwy, 2020-04-16:
Hmm... something is wrong here. I can see that the last value you received is `004301000000C0D4010000D9`. However next one looks suspicious. It's 600 bytes were maximum limit for attribute value length is 512 bytes. Can you check how other apps behave? For example, nRF Connect app?

kevinswartz, 2020-04-17:
Thanks for the response @Cierpliwy . I agree that response looks suspicious.  

With nRF connect, I'm able to get (I think) the interaction I'm looking for.  I send a write command, which is received, and seems to work.  I scan my fingerprint on the device, and I see a response packet available on the read characteristic.  The first response just tells me how large the incoming image will be, so I trigger a 2nd read on that characteristic and I see the image data available in the characteristic value.

With rn-ble-plx, I monitor the read characteristic, and I can see the initial response packet become available.  When it appears, I then trigger a readCharacteristicForService on that characteristic.  It doesn't seem to return any image data, it returns me the same initial response packet.  I'm attaching some sample code below for how I'm trying to do this.

```
//...earlier in code
this.manager = new BleManager()
this.manager.setLogLevel(LogLevel.Verbose);
this.manager.connectToDevice(device.id)
  .then((connectRes) => {
    return this.manager.discoverAllServicesAndCharacteristicsForDevice(device.id)
  })
```
```
let subscription = this.manager.monitorCharacteristicForDevice(this.state.connectedDevice.id, this.UUID_SG_SERVICE.toLowerCase(), this.UUID_SG_CHARACTERISTICS_READ.toLowerCase(), (err, characteristic) => {
      
      if(err){
        logger.error('got error monitoring', err, characteristic);
        return;
      }
      
      this.prettyPrintResult('monitor characteristic res', characteristic);
      let parsedPacket = FMSPacket.parseResponsePacket(characteristic.value);
      this.prettyPrintPacket('monitor response packet', parsedPacket);
      
      subscription.remove();
      
      if(parsedPacket.getChannel() !== 0){
        logger.log('making extra read');
        this.manager.readCharacteristicForDevice(this.state.connectedDevice.id, this.UUID_SG_SERVICE.toLowerCase(), this.UUID_SG_CHARACTERISTICS_READ.toLowerCase())
          .then((res) => {
            this.prettyPrintResult('read characteristic res', res);
            let parsedReadPacket = FMSPacket.parseResponsePacket(res.value);
            this.prettyPrintPacket('read response packet', parsedReadPacket);
            resolve(parsedReadPacket);
          })
          .catch((err) => {
            logger.error('got error reading', err);
            reject(err);
          })
      }
    })
    
    return this.state.connectedDevice.requestMTU(301)
      .then((res) => {
        logger.log('set mtu res', res);
        return this.manager.writeCharacteristicWithResponseForDevice(this.state.connectedDevice.id, this.UUID_SG_SERVICE.toLowerCase(), this.UUID_SG_CHARACTERISTICS_WRITE.toLowerCase(), data)
      })
      .then((res) => {
        return new Promise((resolve, reject) => {
          this.prettyPrintResult('write characteristic result', res);
  
          let response = FMSPacket.parseResponsePacket(res.value);
          this.prettyPrintPacket('write characteristic packet', response);
        })
      })
```

jianglove7, 2020-04-21:
maby look this page https://github.com/Polidea/react-native-ble-plx/wiki/MTU-Negotiation

dariuszseweryn, 2020-04-21:
@kevinswartz — I believe you need to read documentation of your peripheral and how the supposed interaction with it should look like. `react-native-ble-plx` supports all BLE communication ways (writes, long writes, reads, long reads) the same way as the underlying platforms. We do not know how every individual peripheral out there is exposing its data.

closed 2020-06-27 by stale[bot]