dotintent/react-native-ble-plx #648: Interacting with nrf51

Talkwondo, 2020-04-17:
Hello,
So I want to change a color in my bluetooth device for a project and I used this library.
I think I am getting to the end, I'm sending the value and nothing happens... I am receiving an object with the charecharestic at the end which mean there is a response from the device.
I think I am missing something here or the Api I got is wrong.

This is the code:
`const manager = new BleManager();
  const result = Buffer.from('0x021').toString('base64');
  function scanAndConnect() {
    manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        return;
      }
      if (device.name === '$react-2') {
        console.log('Found');
        manager.stopDeviceScan();
        device
          .connect()
          .then(rdevice => {
            return rdevice.discoverAllServicesAndCharacteristics();
          })
          .then(rdevice => {
            console.log(rdevice);
            let services = rdevice.services(device.id);
            return services;
          })
          .then(services => {
            console.log('found services: ', services);
            console.log(result);
            manager
              .characteristicsForDevice(services[0].deviceID, services[0].uuid)
              .then(characteristics => {
                characteristics[1]
                  .writeWithResponse(result)

                  .then(resolved => {
                    console.log(resolved);
                  });
              });
          })
          .catch(err => {
            console.log(err);
          });
      }
    });`

If anyone has an experiment to connect and change values in a device like that it will be awsome to give me a hand.

Thanks,

Cierpliwy, 2020-04-17:
Are you sure you want to send '0x021' string instead of binary 0x21 value? Maybe you wanted that: `Buffer.from([0x21]).toString('base64')`?

Talkwondo, 2020-04-17:
You are correct. I should pass an array.

Thanks

closed 2020-04-17 by Talkwondo