dotintent/react-native-ble-plx #649: how to create custom services with uuid?

soicon, 2020-04-17:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

I'm develop app using ble to scan other device which install same app. My idea is registe an uuid for bluetooth services so i can detect it when scan. But i had seaching and read docs but didn't find any clues.Anyone help me?

dariuszseweryn, 2020-04-26:
Currently this library supports only central role i.e. scanning other devices.

closed 2020-04-26 by dariuszseweryn

francorobles, 2020-05-03:
> ## Prerequisites
> Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**
>
> * [ ]  I am running the latest version
> * [ ]  I checked [...]

Same here. Can't find anything related to creating custom service.