dotintent/react-native-ble-plx #652: How to enable RACP to read all measurements on Glucose device?

renatoramossilva, 2020-04-21:
Hi, 

I am Trying to read all measurements from glucose device.

I can scanning and found the the devices that I wanna connect to.

```
this.manager.startDeviceScan(null, null, (error, device) => {
(...)
}
```
I already discovered the services and characteristics from this device.
To read the last measure I am using the follow code:

 ```
 readLastMeasurement(device){
    //##CONNECT OK
    this.manager.connectToDevice(device.id).then((device) => {
      this.manager.discoverAllServicesAndCharacteristicsForDevice(device.id)
      .then((results) => { 
        //console.log(results);
        this.manager.servicesForDevice(device.id)
          .then((services) => { 
            console.log('services');
            console.log(services);
            //console.log(this.getValueFromBase64("AwYA5AcEDRAmCgAAeLDx"));
            //console.log(this.getValueFromBase64("AwUA5AcEBxYRFwAAW7Dx"));
            this.manager.monitorCharacteristicForDevice(
              device.id, 
              '00001808-0000-1000-8000-00805f9b34fb',
              '00002A18-0000-1000-8000-00805f9b34fb', 
              (err, characteristic) => {  
                //console.log(characteristic.value);
                console.log("Glucose: " + this.getValueFromBase64(characteristic.value));
              })
          });
      });
    });
  }
```
1808 - Glucose (Service)
2A18 - Glucose Measurement (Characteristic)

Using rhe code above I can read the last glucose measured. 
Now I wanna read all the measurements from this device. I found in the ble documentation that to read all the measurements I need write something using 1808 (Service) and 2A52 (Characteristics) as you can see in: [https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk51.v9.0.0%2Fble_sdk_app_gls.html]

I am using this function to read all measurements:

```
 readAll(device){
    this.manager.connectToDevice(device.id).then((device) => {
    console.log("Connected");

    this.manager.discoverAllServicesAndCharacteristicsForDevice(device.id)
    .then((results) => { 
      console.log('results');
      console.log(results);
      this.manager.characteristicsForDevice(device.id, '00001808-0000-1000-8000-00805f9b34fb')
      .then((services) => { 
        //console.log('services');
        //console.log(services);
        this.manager.writeCharacteristicWithResponseForDevice(
          device.id,
          '00001808-0000-1000-8000-00805f9b34fb',
          '00002A52-0000-1000-8000-00805f9b34fb', 
          "0401",
          (err, charac) => {
            console.log('charact');
            console.log(charac)
            ////////////
            charac.monitorCharacteristicForDevice(
            //this.manager.monitorCharacteristicForDevice(
              device.id, 
              '00001808-0000-1000-8000-00805f9b34fb',
              '00002A18-0000-1000-8000-00805f9b34fb', 
              (err, characteristic) => {  
                console.log('monitor');
                console.log(characteristic.value);
                console.log("Glucose values: " + this.getValueFromBase64(characteristic.value));
              })
            ////////////////
          });
      })
    })
  })
}
```

And when I run this code I receive the message error:

```
WARN  Possible Unhandled Promise Rejection (id: 1):
Error: Exception in HostFunction: Malformed calls from JS: field sizes are different.

[[5,30],[36,0],[[102,2000,1587419366866,false]],178]
[native code]
enqueueNativeCall@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:2563:43
fn@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:2161:42
createTimer@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:30856:38
setTimeout@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:30664:18
_87@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:27290:30
reject@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:27167:18
doResolve@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:27208:13
Promise@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:27055:14
fn@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:2134:27
writeCharacteristicWithResponseForDevice$@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:97836:118
tryCatch@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:25668:23
invoke@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:25843:32
tryCatch@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:25668:23
invoke@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:25744:30
http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:25774:19
tryCallTwo@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:27032:9
doResolve@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:27196:25
Promise@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:27055:14
callInvokeWithMethodAndArg@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:25773:29
enqueue@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:25778:157
async@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:25794:69

tryCallOne@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:27023:16
http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:27124:27
_callTimer@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:30578:17
_callImmediatesPass@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:30614:19
callImmediates@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:30833:33
callImmediates@[native code]
__callImmediates@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:2621:35
http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:2398:34
__guard@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:2604:15
flushedQueue@http://localhost:8081/index.bundle?platform=android&dev=true&minify=false:2397:21
flushedQueue@[native code]
invokeCallbackAndReturnFlushedQueue@[native code]
```

Can anyone help me to solve this issue using the Polidea BLE PLX library?

Thanks a lot

renatoramossilva, 2020-05-25:
Hi, Can anyone help me with this issue?
I am trying send the command to read all measurement from a glucose device, but this doesnt work for me.

i already read the documentation and I followed the steps but I dont know if i am doing something wrong or there is a bug in the library

Cierpliwy, 2020-06-19:
That `charac.monitorCharacteristicForDevice(` function call doesn't exist? Did you check documentation?

renatoramossilva, 2020-06-19:
Yes, I read the documentation, and the monitorcharacteristicfordevice enables the device to receive data. 

This issue its ok for me. The problem is that I need to send a command to ble device by RACP, and its doenst work for me.

Ex. I wanna send the command to read all measurements from glucose meter.  I know that i need send (0x) 0101 to device, but I dont know why it doenst work.

I am sure that the monitorCharacteristicForDevice works because I do it in my app, and I am using other app (Nordic ToolBox) to send the RACP. After it I can see all measurement in my app, but I want to do it without use other app.

ram95krishh, 2020-07-01:
@renatoramossilva Hi, am just trying to do what you're trying. Am not able to read anything so far. Any luck so far? Also if you could expand your getValueFromBase64 method, it would be helpful. Thanks!

dariuszseweryn, 2020-07-01:
@renatoramossilva Are you encoding your [value to Base64](https://github.com/Polidea/react-native-ble-plx/wiki/=--FAQ:-Passing-And-Retrieving-Of-Characteristic-Value)?
```
        this.manager.writeCharacteristicWithResponseForDevice(
          device.id,
          '00001808-0000-1000-8000-00805f9b34fb',
          '00002A52-0000-1000-8000-00805f9b34fb', 
          "0401", << looks suspicious
```

renatoramossilva, 2020-07-01:
Hi @dariuszseweryn, I was using other way to convert to base64. Something like this: `base64.decode('0401').`

Now, I used the code that you shared to me and its works.

Thanks a lot, I was blocked in this issue.

renatoramossilva, 2020-07-01:
> @renatoramossilva Hi, am just trying to do what you're trying. Am not able to read anything so far. Any luck for you? Also if you could expand your getValueFromBase64 method, it would be helpful. Thanks!

Hi, @ram95krishh, 
Are you able to identify all services and characteristics from BLE device?
If yes, you can first to test the readable characteristics.

here is the code that I use to convert the data received.

```
async getValueFromBase64(dataURI) {
        //var dataURI = "AwYA5AcEDRAmCgAAeLDx"; //120 You can use this string to test this function
        //var dataURI = "AwUA5AcEBxYRFwAAW7Dx";  //91
        var str = base64.decode(dataURI);
        var utf8 = [];
        for (var i = 0; i < str.length; i++) {
            var charcode = str.charCodeAt(i);
            if (charcode < 0x80) utf8.push(charcode);
            else if (charcode < 0x800) {
                utf8.push(0xc0 | (charcode >> 6),
                    0x80 | (charcode & 0x3f));
            }
            else if (charcode < 0xd800 || charcode >= 0xe000) {
                utf8.push(0xe0 | (charcode >> 12),
                    0x80 | ((charcode >> 6) & 0x3f),
                    0x80 | (charcode & 0x3f));
            }
            // surrogate pair
            else {
                i++;
                // UTF-16 encodes 0x10000-0x10FFFF by
                // subtracting 0x10000 and splitting the
                // 20 bits of 0x0-0xFFFFF into two halves
                charcode = 0x10000 + (((charcode & 0x3ff) << 10)
                    | (str.charCodeAt(i) & 0x3ff));
                utf8.push(0xf0 | (charcode >> 18),
                    0x80 | ((charcode >> 12) & 0x3f),
                    0x80 | ((charcode >> 6) & 0x3f),
                    0x80 | (charcode & 0x3f));
            }
        }

return uf8
} 
```

ram95krishh, 2020-07-01:
Hi @renatoramossilva .. Thanks for that function.. Sorry if this sounds basic, but your function returns the test value "AwYA5AcEDRAmCgAAeLDx" as a utf8 array [3, 6, 0, 195, 164, 7, 4, 13, 16, 38, 10, 0, 0, 120, 194, 176, 195, 177]. How do I infer this is equivalent the value 120 (mentioned in bracket).

Also am trying to retrieve all the values from my glucometer as well.. But when I do this..
```
this.manager.writeCharacteristicWithResponseForDevice(
          device.id,
          '00001808-0000-1000-8000-00805f9b34fb',
          '00002A52-0000-1000-8000-00805f9b34fb', 
          btoa("0401"),     // hoping this is the encoding mentioned by Dariusz
```
It succeeds and returns a characteristic object with the same value "0401" in "characteristic.value" field. Am I missing something? Will the "0401" value vary from device to device or it's just the RACP convention to retrieve all values? Am very new to working with Bluetooth devices. Any help would be appreciated. Thanks!

renatoramossilva, 2020-07-01:
> Hi @renatoramossilva .. Thanks for that function.. Sorry if this sounds basic, but your function returns the test value "AwYA5AcEDRAmCgAAeLDx" as a utf8 array [3, 6, 0, 195, 164, 7, 4, 13, 16, 38, 10, 0, 0, 120, 194, 176, 195, 177]. How do I infer this is equivalent the value 120 (mentioned in bracket).
> 
> And yes, I was able to discover all services and characteristics. Also am trying to retrieve all the values from my glucometer as well.. But when I do this..
> 
> ```
> this.manager.writeCharacteristicWithResponseForDevice(
>           device.id,
>           '00001808-0000-1000-8000-00805f9b34fb',
>           '00002A52-0000-1000-8000-00805f9b34fb', 
>           btoa("0401"),     // hoping this is the encoding mentioned by Dariusz
> ```
> 
> It succeeds and returns a characteristic object with the same value "0401" in "characteristic.value" field. Am I missing something? Will the "0401" value vary from device to device or it's just the RACP convention to retrieve all values? Am very new to working with Bluetooth devices. Any help would be appreciated. Thanks!

Hi,

So, for this example I wanted read a measurement for glucose meter and I knew that this value was 120. In this case, for catch the correct value I returned utf8[13].

I dont know what service/characteristic you are using, but in my case, to read all measurements from glucose meter you should enable the notification for these service/characteristic.

```
this.manager.monitorCharacteristicForDevice(
                            this.state.dv.id,
                            '00001808-0000-1000-8000-00805f9b34fb',
                            '00002A18-0000-1000-8000-00805f9b34fb',
                            async (err, characteristic) => {
```

Now, you are able to receive all values from device. The next step you should write some data (in my case 0101 to read all measurements ) and I used this.

```
             var myData = '0x0101';
    
                                const dataMostSignificantByte = (myData >> 8) & 0xFF;
                                const dataLeastSignificantByte = myData & 0xFF;

                                const dataByteArrayInLittleEndian = [dataLeastSignificantByte, dataMostSignificantByte];
                                const dataInBase64 = Buffer.from(dataByteArrayInLittleEndian).toString('base64');

                                this.manager.writeCharacteristicWithResponseForDevice(
                                    this.state.dv.id,
                                    '00001808-0000-1000-8000-00805f9b34fb',
                                    '00002A52-0000-1000-8000-00805f9b34fb',
                                    dataInBase64
                                )
```
Credits: @dariuszseweryn 

Pls, try using these function, or let me know about what info you wanna read from glucose meter.

dariuszseweryn, 2020-07-01:
Glad that I could help

closed 2020-07-01 by dariuszseweryn

ram95krishh, 2020-07-01:
@renatoramossilva - I can't thank you enough man! This saved my day! I was also looking for the glucose measurements and this got me all of them in a jiffy! However, I found some anomalies with your `getValueFromBase64` implementation. The glucose reading that am looking for is found at 15th index one time, and 14th in the other, sometimes at 13 too.. If this would be of any help to you, please check it out... Install this library called 'buffer'

npm i buffer

```
const Buffer = require('buffer/').Buffer
...
  async(err, char) => {
    const values = Buffer.from(char.value, 'base64')
    const glucoseValue = values[12]   // 12th position in the case of my device
    console.log(glucoseValue)
}
...

```
And thanks to you too @dariuszseweryn !!

fitbuff, 2020-08-11:
Hi @ram95krishh ,
iv been following the instructions on this thread to get  `Glucose Measurement` value from `Accu-Chek guide` device. 
iv added a `writeCharacteristicWithResponseForDevice` for `ACP` and `Glucose Measurement` characteristics.  iv gotten back a response from `Glucose Measurement` but iv been stumped by its response data. i cant make out its value.
we're you able to solve your problem?

this is what i see after ` const values = Buffer.from(char.value, 'base64')` on `Glucose Measurement` characteristic.
`[27, 4, 0, 195, 164, 7, 7, 14, 11, 6, 5, 90, 2, 119, 194, 176, 195, 184, 0, 0]`

fitbuff, 2020-08-11:
i was able to decode the returned value `Glucose Measurement characteristic`.
this article helped me heaps http://www.cocoasteam.com/?p=566 . 
especially this breakdown of the response from characteristic 
http://www.cocoasteam.com/wp-content/uploads/2018/11/GlucoseDecoder.png

thanks @renatoramossilva for your code example helped me heaps as well.

renatoramossilva, 2020-08-12:
Hi @fitbuff,

The code works for you? If yes, great! If dont, please let me know what is your problem and I try help you.

I wrote this code to Contour Next One glucometer and now I´m working in the accu check integration.

Do you have this issue solved? I ask it because to connect the app to device is easier using contour and I dont know how I should use (in the app) the PIN that the accu check generates.

Can you explain me the flow and how do you implemented this issue?

Thanks

> i was able to decode the returned value `Glucose Measurement characteristic`.
> this article helped me heaps http://www.cocoasteam.com/?p=566 .
> especially this breakdown of the response from [...]

fitbuff, 2020-08-12:
hi @renatoramossilva ,
yes the code works for me with `Accu-Chek Guide`. 

> ... to connect the app to device is easier using contour and I dont know how I should use (in the app) the PIN that the accu check generates.

yeah, once you hit `Pairing` (Settings > Wireless > Pairing) and the your app is monitoring `2a18` (Characteristic).

```
this.device.monitorCharacteristicForDevice(
  device.id,
  '00001808-0000-1000-8000-00805f9b34fb',
   '00002a18-0000-1000-8000-00805f9b34fb',
   () => {}
)
```

it should prompt you for a pin to pair on your app (pin is shown on the Accu Chek device)
at least thats whats happening to me

renatoramossilva, 2020-08-13:
Hi @fitbuff,

I am trying to connect accu check to my app and when the glucometer ask me about the pin the android opens a popup to insert this pin.

Are you doing this way?

Because I wanna do equals to MySugar App that the user insert the pin inside of the app.

Please. let me know how works your solution.

ram95krishh, 2020-08-13:
Hey @fitbuff  and @renatoramossilva , I guess all of us just want to know a way to check if the pin was successfully entered. The device.isConnected() handler returns true even before the pin is entered and is leading to many bugs for me!

fitbuff, 2020-08-14:
> I am trying to connect accu check to my app and when the glucometer ask me about the pin the android opens a popup to insert this pin. Are you doing this way?

yes, showing the pop-up asking for a pin is normal. you can check the `Accu-Chek Connect` app to see how their doing it.

> The BLEDeviceInstance.isConnected() handler returns true even before the pin is entered and is leading to many bugs for me!
i havnt thought about checking the connectivity of a device for the app im working on. i cant think of a use case where il need one. this feature on app is simple it'll only retrieve the Glucose reading from peripheral and display it to the user. :]

ram95krishh, 2020-08-14:
@fitbuff You will need it later.. You can't install listeners before pairing is completed! First pair scenario will end up becoming problematic. Coming to your question.. You can't receive all values at once at your listener.. You send the RACP command [1,1] you will receive all values from the accuchek device one by one at the listener you installed for the Glucose Measurement Characteristic Id '00002A18-0000-1000-8000-00805f9b34fb'. Here is a small snippet with regards to the RACP Command I write for all readings.. (written for redux-saga implementation in my app, convert it to your equivalent approach)... 

```
function* syncAllReadings() {
  try {
    const meter = yield select(selectors.getConnectedMeter)  // this is the BLEDevice Instance
    // RACP Command [1,1] - for all values
    var allValuesRetriever = allValuesRACPCommand;    // "0x0101"
    const dataMostSignificantByte = (allValuesRetriever >> 8) & 0xFF;
    const dataLeastSignificantByte = allValuesRetriever & 0xFF;

    const dataByteArrayInLittleEndian = [dataLeastSignificantByte, dataMostSignificantByte];
    const allValuesRetrieverInBase64 = Buffer.from(dataByteArrayInLittleEndian).toString('base64');

    yield call([manager, manager.writeCharacteristicWithResponseForDevice],
      meter.id,
      BLE_GlucoseServiceID,          // "00001808-0000-1000-8000-00805f9b34fb"
      BLE_CharacteristicIds.RACP, // '00002a52-0000-1000-8000-00805f9b34fb'
      allValuesRetrieverInBase64
    )
  } catch (err) {
    handleError(err)
  }
}

```

fitbuff, 2020-08-17:
thanks @ram95krishh .

i cant continually monitor Glucose characteristic. any ideas?
so theres a `Data Transfer` option where i can transfer data manually from Glucose peripheral device via bluetooth.
selecting it the first time works but selecting it again for the 2nd time does not execute 
```
this.manager.monitorCharacteristicForDevice(
                  device.id,
                  '00001808-0000-1000-8000-00805f9b34fb',
                  '00002a18-0000-1000-8000-00805f9b34fb'
)
```

renatoramossilva, 2020-08-17:
Hi @fitbuff,

I did the test how you told me and it works. But I want that the ble notification to insert the PIN appears as a pop up.

Do you know how I can do it?

fitbuff, 2020-08-27:
hi @renatoramossilva ,
insert PIN pop up should appear once you have `monitorCharacteristicForDevice()`.

mvmmarcus, 2021-08-01:
@renatoramossilva can you share your code to get all values and reat latest glucose ? I am getting an error when send command 0x0101 with write method and i don't have Idea what is wrong :(

mvmmarcus, 2021-08-03:
@fitbuff How do you do to **Data Transfer** option works? I can get all glucose values from the accu chek and write commands to the RACP characteristic, but when im trying to transfer data to my device its is disconnecting the accu chek from my app.

GiuseppeEpifani, 2021-08-24:
Hello, excuse my bad English, I am using a blood glucose monitoring system, the problem I have is that it does not monitor the changes in my sensor and tells me that it is connected, and it does not throw an error either, it just does nothing, this is my code:

            if (device.name === 'name_device') {
                manager.stopDeviceScan();
                device.connect()
                .then((device) => {
                    return device.discoverAllServicesAndCharacteristics()
                }) 
                .then((device) => {
                    device.monitorCharacteristicForService('UDDISERVICE', 'UDDI', (error, characteristic) => {
                        if (error) {
                          console.log(error)
                          return
                        }

                        const arrayBytes = base64js.toByteArray(characteristic.value);
                        console.log(arrayBytes)
                    });
                }, (error) => {
                    console.log(error.message)
                    manager.destroy();
                })
            }

renatoramossilva, 2021-08-24:
> Hello, excuse my bad English, I am using a blood glucose monitoring system, the problem I have is that it does not monitor the changes in my sensor and tells me that it is connected, and it does not [...]

@GiuseppeEpifani, Can you connect to device? I follow these steps
1 - Scan (you can use this function using the GLUCOSE service to get only glucometers)
2 - The scan return is a device object, so you can call the function devicetoDevice, passing the device.id as argument.
3 - After that, you should discover all services and charac and so you can read the properties
    3.1 Use readCharacteristicForDevice to read the characteristics that have the READ property
    3.2 Use monitorCharacteristicForDevice to read the characteristics that has NOTIFY or INDICATE as propety

GiuseppeEpifani, 2021-08-24:
The device apparently does connect and discover the services and characteristics, but it is not monitoring me and it does not throw me an error, it simply does not do anything, it should be noted that I use this code to connect to pulse oximeter sensors, Blood Pressure meter and thermometer , and all can be monitored correctly, I leave an example of my code:

            if (device.name === 'Samico GL') {
                manager.stopDeviceScan();
                device.connect()
                .then((device) => {
                    return device.discoverAllServicesAndCharacteristics()
                }) 
                .then((device) => {
                        device.monitorCharacteristicForService('FFF0', 'FFF4', (error, characteristic) => {
                        if (error) {
                          console.log(error)
                          return
                        }
                        console.log(characteristic)
                    });
                }, (error) => {
                    console.log(error.message)
                    manager.destroy();
                })
            }

I don't understand what could be wrong, as I am using the glucose sensor documentation, I would appreciate any help.
I leave a screenshot of the api steps
[image]
[image]

JamesOwen7, 2022-05-04:
@GiuseppeEpifani did you figure it out? Are you using the Sam_Andesfit monitor?

silldepair, 2024-02-25:
> Yes, I read the documentation, and the monitorcharacteristicfordevice enables the device to receive data.
>
> This issue its ok for me. The problem is that I need to send a command to ble device by [...]

hi, i have problem in activate sensor to monitor, how do you know its (0x) 0101 to activate the sensor ? and if it in documentation paper, how do i find the value to activate sensor if i dont have documentation paper ? cz i have different device than you