dotintent/react-native-ble-plx #653: Large file sending by using react-native-ble-plx

MedRaid, 2020-04-23:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [yes ] I am running the latest version
- [ yes] I checked the documentation and found no answer
- [ yes] I checked to make sure that this issue has not already been filed
- [ yes] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. 

## Question

I'm trying to write in a characteristic that sends a binary file as the value, this value is large, it's 19215 bytes which is not a problem because  i negociated the mtu using : 

       device.connect({ requestMTU: 260 })

i've divided the file into 240 bytes for each element and each time encode the element to base64 and i use the function writeCharacteristicWithResponseForDevice() in order to write that element,

the issue is i successfully wrote the whole file 19215 bytes using a loop to write each time an element, but while i tried to read the characteristic i can only read the last written element

**Example**:

           
          this.manager.writeCharacteristicWithResponseForDevice(device.id,"1111", "40E1ED56-EC4A-4DC6-A6BD-30377F186B77", base64.encode(element1)

          this.manager.writeCharacteristicWithResponseForDevice(device.id,"1111", "40E1ED56-EC4A-4DC6-A6BD-30377F186B77", base64.encode(element2)) 
          
          this.manager.writeCharacteristicWithResponseForDevice(device.id,"1111", "40E1ED56-EC4A-4DC6-A6BD-30377F186B77", base64.encode(element3))

**when i read the Characteristic using**

          device.readCharacteristicForService("1111", "40E1ED56-EC4A-4DC6-A6BD-30377F186B77")

**i get as a value :**

        console.log(base64.decode(characteristic.value) => element3

**it should be element1+ element2 + element3**

-

**here is my code for writing :**

  ```
WriteIncaracteristic() {

   // this.state.fileSize is calculated in another function
      const nbPackages = Math.floor(this.state.fileSize/240) + 1

      var fileArray = []

     for (let i = 0; i <= nbPackages; i++) {

      // this.state.fileContent is created in another function
      fileArray.push(this.state.fileContent.slice(0, 240))
      this.state.fileContent = this.Remove(this.state.fileContent, 0, 240)

     }

     for (let j = 0; j <= fileArray.length; j++) {

       if (fileArray[j]) {

       const device = this.state.myDevice

           this.manager.writeCharacteristicWithResponseForDevice(device.id,"1111","40E1ED56-EC4A-4DC6-A6BD-30377F186B77", base64.encode(fileArray[j]))

            .then((characteristic) => {
              console.log(base64.decode(characteristic.value));

              return 
            }) .catch((error) => {
              this.createTAlert("writing", error.message  )
            });
     }}

  }
```

**here is the function to read characteristic:**

     
    ReadCaracteristic() {

          const device = this.state.myDevice

          device.readCharacteristicForService("1111", "40E1ED56-EC4A-4DC6-A6BD-30377F186B77")
          .then((characteristic) => {

               console.log(base64.decode(characteristic.value));
               console.log(characteristic );

            return 
           }) .catch((error) => {
               // Handle errors
               this.createTAlert("Reading", error.message  )
           });
   }
```

dariuszseweryn, 2020-04-26:
Why do you think the read should return element1 + element2 + element3? It’s up to the peripheral what it will respond with

MedRaid, 2020-04-27:
thank you for responding,  so it's from the peripheral's side that i should treat the information received and i can then respond with elem1 + elem2 + elem3, thank you

closed 2020-04-27 by MedRaid