dotintent/react-native-ble-plx #654: How to use restoreStateFunction/restoreStateIdentifier

nucleusdeveloper, 2020-04-26:
Hi, thanks for your awesome library, it's very useful in my project.
I am working on reconnecting device after the application has relaunched. Following the document, I see that we have 2 properties: restoreStateFunction/restoreStateIdentifier.

The snippet code:

```
  bleManager = new BleManager({
    restoreStateIdentifier: 'bleManagerRestoredState',
    restoreStateFunction: (bleRestoredState: BleRestoredState) => {
      if (bleRestoredState == null) {
        // BleManager was constructed for the first time.
      } else {
        // BleManager was restored. Check `bleRestoredState.connectedPeripherals` property.
      }
    },
  });
```
In the document, I didn't see how to use this value. I log the value and see that it is an array and contains device information from the last connected. However, how do we can reconnect with this information?

Do we have another way to reconnect the device instead of rescan and connect as usual?

dariuszseweryn, 2020-04-26:
When you get the information in restored state that an interesting peripheral is connected by the OS the only thing you need is to connect it via the library. This call should resolve very quickly. No scanning is needed in this scenario.

Does it answer your question?

nucleusdeveloper, 2020-04-26:
@dariuszseweryn 
Thank you for your quick response.
Could you help me clarify a bit about "connect it via the library"?
I see that the library provides 2 ways to connect a device:
- 1/ using method from BleManager: connectToDevice (https://polidea.github.io/react-native-ble-plx/#blemanagerconnecttodevice)
- 2/ using method from Device: connect (https://polidea.github.io/react-native-ble-plx/#deviceconnect)
When using these methods, do we need to store the instance of BleManager?

dariuszseweryn, 2020-04-26:
To use the library one needs a BleManager instance. In the restore state you should get information which identifiers connected peripherals have. Then you call the adequate function on BleManager to connect the peripheral device.

I am not sure what is the problem you face

nucleusdeveloper, 2020-04-26:
My goals: automatically reconnecting the device in case the application has already connected the device before
Current behavior: sometimes bleRestoredState is null or contains connectedPeripherals. It is unstable.
Also, in the document, I didn't see how to use bleRestoredState.connectedPeripherals value to reconnect the device.
Using bleRestoredState.connectedPeripherals[0].connect() or bleManagerInstance.connectToDevice(bleRestoredState.connectedPeripherals[0].id)?
That's why I asked this question.
Current my solution: stored the last connected device information, when app is relaunched, we have to scan and connect again based on that info.

@dariuszseweryn

dariuszseweryn, 2020-04-27:
I think that both calls should work.

> When using these methods, do we need to reuse the old instance of BleManager which is created before the app is killed? (It means we have to store BleManager instance somewhere)
Or we simply use the new BleManager instance which returns the bleRestoredState.connectedPeripherals?

If this method is called your application has to have a `BleManager` instance created already (because otherwise you would not register the listener). Use the instance you have. An instance of `BleManager` should be created only once for each JS application start.

nucleusdeveloper, 2020-04-27:
Thanks for your explanation @dariuszseweryn 
Because after app is killed, relaunching the app, I think BleManager instance is gone and has to create again for each JS app. 
So call these functions will not work as I think.

I will close this question here.

dariuszseweryn, 2020-04-27:
I think you see a chicken and an egg problem where there is none. Since `restoreStateFunction` is a JS function and the app was killed then `BleManager` has to be created _again_ so the function can be called from the native part of the library. There is no magic there.

If the `BleManager` would be gone then the function would also be not attached (or non-existent) and thus- not callable.

nucleusdeveloper, 2020-04-27:
@dariuszseweryn Great. I think I understand better now.

closed 2020-04-27 by nucleusdeveloper

LucasAndrad, 2020-09-23:
@nucleusdeveloper did you manager to get your devices connected with BleManager again? If yes, could you share you solution please?