dotintent/react-native-ble-plx #655: bleManager.isDeviceConnected() always return false, irrespective of it's actually connected or not

federicomiralles, 2020-04-27:
## Expected Behavior

According to the docs **isDeviceConnected** should return either true or false depending on whether the BLE device is connected to the mobile/app

## Current Behavior

Despite of running isDeviceConnected on a device that it's actually connected it always returns **false**.

### Context
I need to check by using the built-in RN AppState component whether a device the user was using before switching of apps (or after coming back from background) it's still connected...

* Library version: 2.0.0

Every time I call isDeviceConnected with a connected device Id, I always get Device AA:BB:CC:DD:EE:FF is NOT connected!
```js
const isBleDeviceConnected = async (deviceId = null) => {
    if (!deviceId) return false;
    const isConnected = await bleManager.isDeviceConnected(deviceId);
    if (isConnected) {
      console.log(`Device ${deviceId} is connected!`);
    } else {
      console.log(`Device ${deviceId} is NOT connected!`);
    }
    return isConnected;
  };
```

Is there any fix in progress or a workaround available?

Thanks in advance!

Federico

andrewcharnley, 2020-05-21:
I can confirm this is a bug and it's major. If a user clicks the back button and the App goes to the background the connection is 'muted' but kept open. I've verified this on my bluetooth board.

The only thing I can think about doing is hitting a characteristic as a workaround and handling the error as a disconnect that way.

andrewcharnley, 2020-05-21:
Further investigation

Hitting a characteristic didn't work either. The plx 'device' thinks it's disconnected but BLE_GAP is still connected.

At first attempt I get 'BleError: Device ? is already connected'. After this it hangs on scanning, which I'd expect it to as my BLE is still connected and not advertising.

dariuszseweryn, 2020-05-22:
@andrewcharnley could you prepare a minimal example repo?

andrewcharnley, 2020-05-22:
I've semi solved it (only tested on Android).

Rather than let the hardware back button push the app to the background you have to use       BackHandler.exitApp(). Oddly this React Native feature is also broken - the app doesn't exit but retains it's state in the background in a suspended state, however the BLE device is still connected and crucially  isConnected() works to indicate it as such. For some reason this is different than without using exitApp.

Without using BackHandler.exitApp there is no way to get back to the BLE device without killing the app completely (not just using back button).

I haven't tested the Home button yet.

dariuszseweryn, 2020-05-22:
How do you store the `BLEManager` instance?

andrewcharnley, 2020-05-22:
It's at file level outside of React's component function, so there is only the one, and it isn't affected by re-renders.

dariuszseweryn, 2020-05-22:
I see nothing in the code that should justify such behaviour. Could you create a minimal example project?

I assume the issue is visible only on Android?

federicomiralles, 2020-05-22:
@dariuszseweryn unfortunately I can confirm this also happens on iOS as well...

dariuszseweryn, 2020-05-23:
Could you create a minimal example project which I could compile myself and dig into it?

andrewcharnley, 2020-05-24:
Although I have my workaround I've got a possibly related bug where if I leave the app on searching after a amount of time (which I hope to quantify) it hangs without finding the device when I set it to advertise. If I then disable bluetooth in android and re-enable the search then finds the device.

I'm not knowledgeable on how this library interacts with Android native but it looks like it's missing an event handler or similar for some type of sleep state. Android does rate limit how much hunting an app can do if it keeps at it which may be part of the problem. I'm just theorising though.

andrewcharnley, 2020-05-24:
`import { PermissionsAndroid } from "react-native";
import { Buffer } from "buffer";
import { BleManager } from "react-native-ble-plx";
import EventEmitter from "react-native/Libraries/vendor/emitter/EventEmitter";
import props from "../config/props.json";

const manager = new BleManager();

const serviceUUID = props.service.uuid;

const { characteristics } = props;

const charLoadOrder = Object.keys(characteristics)
  .sort((a, b) => characteristics[a].main - characteristics[b].main);

const ret = {
  characteristics: charLoadOrder
    .reduce((a, b) => {
      a[b] = {
        emitter: new EventEmitter()
      }
      return a;
    }, {}),
  emitter: new EventEmitter(),
  commands: props.commands,
  status: {
    statusCode: -1,
    statusText: "Initilizing"
  }
};

const findTargetDevice = manager => new Promise((resolve, reject) => {

  manager.startDeviceScan([serviceUUID], null, (error, device) => {
    if (error) {
      manager.stopDeviceScan();
      reject(error);
      return;
    }
    if (device.name === props.device.name) {
      manager.stopDeviceScan();
      resolve(device);
    }
  });
});

const scanAndConnect = async (manager, setState) => {

  let version;

  try {
    await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION);
  } catch (e) {
    throw 403;
  }

  if (ret.device) {
    setState(904);
    try {
      //await ret.device.readCharacteristicForService(serviceUUID, characteristics['device'].uuid);
      const connected = await ret.device.isConnected();
      if (! connected) {{
        delete ret.device;
      }}
    } catch (e) {
      delete ret.device;
    }
  }

  if (! ret.device) { 
    setState(900);
    ret.device = await findTargetDevice(manager);
    setState(901);
    await ret.device.connect();
    setState(902);
  }

  const { device } = ret;

  await device.discoverAllServicesAndCharacteristics();

  // retrieve read values - starting with core uuid
  let disconnectHandle;
  await new Promise((resolve, reject) => {

    // notification and writes go here, andy error rejects the promise and puts us back into the bluetooth search

    disconnectHandle = manager.onDeviceDisconnected(error => {

      delete ret.device;
      reject(903);
    });

    setState(200);
  }).catch(e => {

    disconnectHandle.remove();
    throw e;
  });
};

const setState = code => {

  if (!props.statusCodes[String(code)]) {
    code = 500;
  }
  const status = {
    statusCode: code,
    statusText: props.statusCodes[String(code)]
  };
  ret.emitter.emit('status', status);
};

let timeout;
let inProgress;

const looper = async () => {

  try {
    inProgress = true;
    await scanAndConnect(manager, setState);
  } catch (e) {
    const statusCode = typeof e === 'number'? e: e.errorCode;
    console.log(e);
    setState(statusCode);
  }
  try {
    const state = await manager.state();
    if (state === 'PoweredOff') {
      setState(102);
      throw 102;
    }
    throw -1;
  } catch (e) {
    if (e !== 102) {
      timeout = setTimeout(() => looper(), props.connectPolling);
    }
  }
  inProgress = false;
};

manager.onStateChange(state => {

  if (state === 'PoweredOn' && !inProgress) {
    clearTimeout(timeout);
    looper();
  }
});

ret.start = looper;

ret.stop = async () => {

  try {
    if (ret.device) {
      await ret.device.cancelConnection();
    }
  } catch(e) {
    console.log(e);
  }
  delete ret.device;
};

export default ret;
`

dariuszseweryn, 2020-06-30:
Finally I had some time to double check what is going on. Using release `2.0.1` and below code:
```javascript
manager.startDeviceScan(
  null,
  null,
  (error, device) => {
    if (error) {
      console.log('error', JSON.stringify(error));
    } else if (device) {
      if (device.localName === 'MyTestDevice') {
        manager.stopDeviceScan();
        localScanChange(false);
        device
          .isConnected()
          .then(bool => console.log('isConnected0', bool));
        manager
          .isDeviceConnected(device.id)
          .then(bool => console.log('isConnectedM0', bool));
        device
          .connect()
          .then(() => {
            device
              .isConnected()
              .then(bool => console.log('isConnected1', bool));
            manager
              .isDeviceConnected(device.id)
              .then(bool => console.log('isConnectedM1', bool));
            return device.discoverAllServicesAndCharacteristics();
          })
          .then(() => {
            setConnectedDevice(device);
            console.log('connected');
          });
        const sub = device.onDisconnected((error, device) => {
          console.log('disconnected');
          setConnectedDevice(null);
          sub.remove();
        });
      }
    }
  },
);
```
I got a seemingly valid logs on Android:
```
[Tue Jun 30 2020 16:15:14.274]  LOG      isConnected0 false
[Tue Jun 30 2020 16:15:14.277]  LOG      isConnectedM0 false
[Tue Jun 30 2020 16:15:14.839]  LOG      isConnected1 true
[Tue Jun 30 2020 16:15:14.850]  LOG      isConnectedM1 true
[Tue Jun 30 2020 16:15:15.430]  LOG      connected
```
And iOS:
```
2020-06-30 16:23:28.148 [info][tid:com.facebook.react.JavaScript] 'isConnected0', false
2020-06-30 16:23:28.148 [info][tid:com.facebook.react.JavaScript] 'isConnectedM0', false
2020-06-30 16:23:28.694 [info][tid:com.facebook.react.JavaScript] 'isConnected1', true
2020-06-30 16:23:28.695 [info][tid:com.facebook.react.JavaScript] 'isConnectedM0', true
2020-06-30 16:23:29.290 [info][tid:com.facebook.react.JavaScript] connected
```
Could you retest with `2.0.1`?

andrewcharnley, 2020-07-10:
It doesn't occur on a Android 9 device (I was using a 6 device). I'll investigate more but it might be down to poor hardware/software implementation on the device side.

ad-boschung, 2020-09-10:
@dariuszseweryn Any news about this issue ?

closed 2020-11-19 by stale[bot]

pooja-wal, 2020-11-27:
I am facing the same issue. Are there any updates on this?

dariuszseweryn, 2020-11-28:
My guess is that you are misusing React Hooks. Could you edit to contain information according to the [issue template](https://github.com/Polidea/react-native-ble-plx/blob/master/.github/ISSUE_TEMPLATE/bug_report.md)? I am especially interested in native logs.

rubend151, 2021-01-18:
I'm having the same issue...

dan-lee, 2021-01-20:
Our workaround is to try to read an arbitrary characteristic from a device with try/catch and if it throws it's not connected.

dariuszseweryn, 2021-01-21:
Could anyone prepare and post a full (but minimal) example of code that exhibits this behaviour? Complete example including creation of the manager.

mauromazzei, 2021-09-05:
Seems like that the bug is still there.
Here is a repo: https://github.com/Spoutnik97/react-native-arduino-ble-example.git

gabrielnovaisgs, 2022-04-04:
I have the same issue, any news about it?

AkashWaghule, 2022-07-04:
I have the Same issue, App is connected to device. isDeviceConnected() always show the device is disconnected but I can see the ble device is connected. After I connect it and perform some operation. In the next write characterstic cycle it again show the ble is disconnected and I have to connect back again.

I am also facing issue with disconnect.  Neither device.cancelConnection() work or bleManager.cancelDeviceConnection() work.

paurosellfarre, 2022-07-13:
As [AkashWaghule](https://github.com/AkashWaghule) said, for me the problem was that I had mutiple BleManager instances.
Just make sure you only have `const manager = new BleManager()` 1 time on your code.

If you need to use it into more files make the variable exportable `export const manager = new BleManager()`

celciusjj, 2023-02-15:
> I have the Same issue, App is connected to `device. isDeviceConnected()` always show the device is disconnected but I can see the ble device is connected. After I connect it and perform some operation. In the next write characterstic cycle it again show the ble is disconnected and I have to connect back again.
> 
> I am also facing issue with disconnect. Neither `device.cancelConnection()` work or `bleManager.cancelDeviceConnection()` work.
> 
> Update:
> 
> After lot of Efforts, I was able to work it. How?
> 
> Make sure new `BleManager()` is call only once in entire life cycle of your code or the instance is destroy before you leave it. Basically no Multiple Instance.
> 
> --I was using
> 
> `const [manager, setManager] = useState(new BleManager());`
> 
> Instead of this you can directly declare it as a
> 
> `const manager = new BleManager()`
> 
> --Make sure the constant is out of your UI component Instead of doing this
> 
> ```
> export default function HomeScreen() {
>     const [manager, setManager] = useState(new BleManager());
>     // etc.
> }
> ```
> 
> go for this
> 
> ```
> import { BleManager } from 'react-native-ble-plx'
> 
> // Where to instantiate the BleManager
> const manager = new BleManager()
> 
> //Ui Component
> const BluetoothTest = () => {
> }
> export default BluetoothTest 
> ```
> 
> I think this will help both the issue. `disconnect `and is `deviceConnected`.

Thanks bro, you save my life <3

andrewcharnley, 2023-02-15:
I just checked and yes this is also how my code is.