dotintent/react-native-ble-plx #656: bleManager.connectToDevice is NOT reporting any error on fail

federicomiralles, 2020-04-29:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

Trying to run **bleManager.connectToDevice(deviceId)** on a recently _scanned_ but now _shutdown_ device, should fail throwing an error.

## Current Behavior

No error is thrown, however obviously the connection is not achieved.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Switch on sensor tag
2. Init scan
3. Switch off sensor tag
4. Try to connect to reported sensor tag

### Context
Let's say that for some reason a device just successfully scanned gets shutdown, there should a way of failing the connection intent (or at least to have a configurable timeout).

* Library version: 2.0.0
* Platform: Android/iOS.
* Formatted code sample or link to a repository:

```js
bleManager
        .connectToDevice(deviceId)
        .then(device =>
          console.log(`Device ${deviceId} connected successfully!`),
        )
        .catch(err => console.log('ERROR ON CONNECTION =', err));
```

dariuszseweryn, 2020-04-30:
For Android using this setup the promise should fail after ~30 seconds. I do not remember if iOS is doing the same. What platform you test on?

federicomiralles, 2020-05-01:
@dariuszseweryn thanks for your answer.
I'm working with both platforms.
I can confirm a fail message after 30'' on Android:
`BleError: Device AA:BB:CC:DD:EE:FF was disconnected`
Unfortunately that's not the case on iOS. Nothing is being returned...

dariuszseweryn, 2020-05-01:
That is exactly as the underlying platforms behave by default. iOS does not timeout a connect request on its own. Android may not timeout if `autoConnect` is set to `true` but it will most probably take more time than in default (`autoConnect: false`) mode. Exactly as it is described in [the wiki](https://github.com/Polidea/react-native-ble-plx/wiki/Device-Connecting). I will add a note on default behaviour of iOS.

closed 2020-05-06 by dariuszseweryn