dotintent/react-native-ble-plx #657: [iOS] monitorCharacteristicForDevice fails with error Characteristic 0000ffe1... notify change failed for device 4221AED1... and service 0000ffe0... on sensor tag

federicomiralles, 2020-04-30:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

To be able to subscribe using monitorCharacteristicForDevice() as expected on iOS.

## Current Behavior

On iOS trying to monitor the _button pressed_ characteristic on an Sensor Tag/iTAG fails returning the following BleError: **Characteristic 0000ffe1... notify change failed for device 4221AED1... and service 0000ffe0..**.

### Context

The strangest thing is that the very same code DOES WORKS perfectly well on Android, being able to observe Sensor Tag/iTag button presses without getting any issues.

* Library version: 2.0.0
* Platform: iOS
* Formatted code sample or link to a repository:

```js
const subscription = bleManager.monitorCharacteristicForDevice(
  deviceId,
  serviceUuid,
  characteristicUuid,
  (error, result) => {
    if (error) {
      console.log("ERROR ON MONITOR =", error);
      return;
    }
    console.log("MONITORED =", result.value);
  }
);
```
Am I missing some extra step on iOS that is not required on Android?

Thanks in advance!

Cheers,

Federico

dariuszseweryn, 2020-05-03:
[I got a BleError. Now what?](https://github.com/Polidea/react-native-ble-plx/wiki/=-FAQ:-I-got-a-BleError.-Now-what%3F)

federicomiralles, 2020-05-04:
@dariuszseweryn thanks for your answer!

I checked the link you shared, specifically the info regarding the [iOS notification emit error](https://github.com/Polidea/react-native-ble-plx/wiki/=-FAQ:-I-got-a-BleError.-Now-what%3F#ios-notifications-emit-error-but-it-works-using-different-tools) but if there is an error with the CCCD wouldn't that also affect the Android build?

On the other hand I tried to subscribe to the _onDisconnected_ as suggested too but I couldn't find any clue since the device is not actually being disconnected while the error is thrown.

So is there anything else I can do to overcome this issue? I mean it shouldn't be impossible to monitor a sensor tag button from iOS isn't it? Specially taking into account that the very same code works perfectly well on Android...

Cheers,

Federico

dariuszseweryn, 2020-05-05:
> I checked the link you shared, specifically the info regarding the iOS notifications emit error but if there is a problem with the CCCD, wouldn't that also affect the Android build?

Not necessarily — there are some platform nuances. Without logs as per issue template I cannot do much.

federicomiralles, 2020-05-05:
@dariuszseweryn how can I get more logs?

Btw I tried an iOS app for sensor tags and using the same device I was able to detect the button press on it: https://apps.apple.com/us/app/itag-one-bluetooth-key-finder/id1476723114

So I guess there must some missing some setup/option/something on my end while using your lib in order to monitor the very same sensor tag.
If you can pinpoint at me how to retrieve more log about what could be going wrong would be very helpful.

Thanks,

Federico

dariuszseweryn, 2020-05-06:
The issue template which, you ignored, describes how to get logs. `setLogLevel(Verbose)` and get the native logs.

or at least you could paste the printed error as per what I gave you in my first answer to your post

federicomiralles, 2020-05-06:
@dariuszseweryn enabling native logs as per your advise reported this:
```
error: Optional(Error Domain=CBATTErrorDomain Code=10 "The attribute could not be found." UserInfo={NSLocalizedDescription=The attribute could not be found.}))
Unhandled error happened: Characteristic notify change error has occured: The attribute could not be found.
 subscription called from:

2020-05-06 18:23:09.072 [info][tid:com.facebook.react.JavaScript] 'ERROR ON MONITOR =', [BleError: Characteristic 0000ffe1-0000-1000-8000-00805f9b34fb notify change failed for device 4221AED1-.... and service 0000ffe0-0000-1000-8000-00805f9b34fb]
```
Any idea?

Thanks!

gilador, 2020-05-19:
im seeing it on 1.1.0

federicomiralles, 2020-05-19:
@dariuszseweryn @corymsmith @bitcrumb would you please provide some guidance on this issue?

dariuszseweryn, 2020-05-21:
Could you please provide logs? Print _all_ attributes your peripheral has? It looks like your peripheral does not have characteristic you expect

federicomiralles, 2020-05-22:
@dariuszseweryn after looping through all the services and characteristics of the iTag by using **_discoverAllServicesAndCharacteristics()_** on iOS I get the following attributes:
```
SERVICE 1
{
  "_manager": {
    "_activePromises": {},
    "_activeSubscriptions": {
      "1": [Object],
      "6": [Object]
    },
    "_errorCodesToMessagesMapping": {
     ...
    },
    "_eventEmitter": {
      "_nativeModule": [Object],
      "_subscriber": [EventSubscriptionVendor]
    },
    "_uniqueId": 9
  },
  "deviceID": "4221AED1-WWWW-XXXX-YYYY-ZZZZZZZZZZZZ",
  "id": 10739853184,
  "isPrimary": true,
  "uuid": "0000180f-0000-1000-8000-00805f9b34fb"
}
|
--> CHARACTERISTIC 1
    {
        "_manager": {
        "_activePromises": {},
        "_activeSubscriptions": {
            "1": [Object],
            "6": [Object]
        },
        "_errorCodesToMessagesMapping": {
        ...
        },
        "_eventEmitter": {
            "_nativeModule": [Object],
            "_subscriber": [EventSubscriptionVendor]
        },
        "_uniqueId": 10
        },
        "deviceID": "4221AED1-WWWW-XXXX-YYYY-ZZZZZZZZZZZZ",
        "id": 10791031584,
        "isIndicatable": false,
        "isNotifiable": true,
        "isNotifying": false,
        "isReadable": true,
        "isWritableWithResponse": false,
        "isWritableWithoutResponse": false,
        "serviceID": 10739853184,
        "serviceUUID": "0000180f-0000-1000-8000-00805f9b34fb",
        "uuid": "00002a19-0000-1000-8000-00805f9b34fb",
        "value": null
    }

SERVICE 2
    {
    "_manager": {
        "_activePromises": {
        "11": [Function anonymous]
        },
        "_activeSubscriptions": {
        "1": [Object],
        "6": [Object]
        },
        "_errorCodesToMessagesMapping": {
        ...
        },
        "_eventEmitter": {
        "_nativeModule": [Object],
        "_subscriber": [EventSubscriptionVendor]
        },
        "_uniqueId": 11
    },
    "deviceID": "4221AED1-WWWW-XXXX-YYYY-ZZZZZZZZZZZZ",
    "id": 10739851648,
    "isPrimary": true,
    "uuid": "00001802-0000-1000-8000-00805f9b34fb"
    }
    |
    --> CHARACTERISTIC 1
        {
            "_manager": {
            "_activePromises": {},
            "_activeSubscriptions": {
                "1": [Object],
                "6": [Object]
            },
            "_errorCodesToMessagesMapping": {
                ...
            },
            "_eventEmitter": {
                "_nativeModule": [Object],
                "_subscriber": [EventSubscriptionVendor]
            },
            "_uniqueId": 12
            },
            "deviceID": "4221AED1-WWWW-XXXX-YYYY-ZZZZZZZZZZZZ",
            "id": 10791048544,
            "isIndicatable": false,
            "isNotifiable": true,
            "isNotifying": false,
            "isReadable": false,
            "isWritableWithResponse": true,
            "isWritableWithoutResponse": true,
            "serviceID": 10739851648,
            "serviceUUID": "00001802-0000-1000-8000-00805f9b34fb",
            "uuid": "00002a06-0000-1000-8000-00805f9b34fb",
            "value": null
        }

SERVICE 3
    {
      "_manager": {
        "_activePromises": {
          "13": [Function anonymous]
        },
        "_activeSubscriptions": {
          "1": [Object],
          "6": [Object]
        },
        "_errorCodesToMessagesMapping": {
         ...
        },
        "_eventEmitter": {
          "_nativeModule": [Object],
          "_subscriber": [EventSubscriptionVendor]
        },
        "_uniqueId": 13
      },
      "deviceID": "4221AED1-WWWW-XXXX-YYYY-ZZZZZZZZZZZZ",
      "id": 10739856704,
      "isPrimary": true,
      "uuid": "0000ffe0-0000-1000-8000-00805f9b34fb"
    }
    |
    --> CHARACTERISTIC 1
        {
            "_manager": {
            "_activePromises": {},
            "_activeSubscriptions": {
                "1": [Object],
                "6": [Object]
            },
            "_errorCodesToMessagesMapping": {
                ...
            },
            "_eventEmitter": {
                "_nativeModule": [Object],
                "_subscriber": [EventSubscriptionVendor]
            },
            "_uniqueId": 14
            },
            "deviceID": "4221AED1-WWWW-XXXX-YYYY-ZZZZZZZZZZZZ",
            "id": 10790972192,
            "isIndicatable": false,
            "isNotifiable": true,
            "isNotifying": false,
            "isReadable": true,
            "isWritableWithResponse": false,
            "isWritableWithoutResponse": false,
            "serviceID": 10739856704,
            "serviceUUID": "0000ffe0-0000-1000-8000-00805f9b34fb",
            "uuid": "0000ffe1-0000-1000-8000-00805f9b34fb",
            "value": null
        }
```
Out of which the latest characteristic (UUID = 0000ffe1-0000-1000-8000-00805f9b34fb) is the one I expect to use and it's failing with the reported error:
```
[BleError: Characteristic 0000ffe1-0000-1000-8000-00805f9b34fb notify change failed for device 4221AED1-WWWW-XXXX-YYYY-ZZZZZZZZZZZZ and service 0000ffe0-0000-1000-8000-00805f9b34fb]
```
Thanks,

Federico

dariuszseweryn, 2020-05-23:
Could you also add info about descriptors?

federicomiralles, 2020-05-23:
@dariuszseweryn there aren't...

federicomiralles, 2020-06-18:
@dariuszseweryn do you have any update on this?

dariuszseweryn, 2020-06-29:
Having no Client Characteristic Configuration Descriptor is violation of Bluetooth Spec — not sure if iOS in recent versions did show all descriptors. I recall that iOS has been working correctly even for characteristics without CCCD. What phone/OS version you test against? Could you also print the error using `JSON.stringify(error)`?

closed 2020-09-04 by stale[bot]

federicomiralles, 2020-09-14:
> Having no Client Characteristic Configuration Descriptor is violation of Bluetooth Spec — not sure if iOS in recent versions did show all descriptors. I recall that iOS has been working correctly [...]

@dariuszseweryn but if that's the case it shouldn't work on the other iOS app I mentioned here:
https://github.com/Polidea/react-native-ble-plx/issues/657#issuecomment-624005624
given I'm using the same iOS device and BLE sensor tag... isn't it?

BTW I'll provide the other details you requested me later on during this week...

nolife08021, 2021-02-22:
@federicomiralles  did you solve it? i am facing the same issue.

`{"message":"Characteristic 000036f6-0000-1000-8000-00805f9b34fb notify change failed for device A and service 0000fee7-0000-1000-8000-00805f9b34fb","errorCode":403,"attErrorCode":10,"iosErrorCode":null,"androidErrorCode":null,"reason":"The attribute could not be found.","name":"BleError"}`

Salva222, 2022-01-01:
Same issue [BleError: Characteristic 00002a00-0000-1000-8000-00805f9b34fb notify change failed for device ? and service 00001800-0000-1000-8000-00805f9b34fb]

JamesOwen7, 2022-04-25:
Has anyone solved this?

zain148, 2022-05-20:
anyone fixed this?

ariccio, 2022-11-29:
@Salva222 very bizzare, I'm hitting the same on ios.

`00002a00-0000-1000-8000-00805f9b34fb` is not showing up, and I don't know why.

My formatter prints:

```
formatted BLE error string for human consumption: Error name: "BleError", error message: "Characteristic 00002a00-0000-1000-8000-00805f9b34fb not found", error code: "404", error reason: "null", error code (IOS specific): "null", error code (android specific): "null", error code (ATT specific, whatever that is): "null"
```

savety6, 2022-11-29:
@ariccio
for me, the problem was that I tried to "write with a response" to a characteristic that is "without response" and for some reason android let me do it, but ios did catch my error. 
I suggest you use some Bluetooth test app like "nFR Connect" to see what your services and characteristics.

ariccio, 2022-11-29:
And now I can't read it through other apps?!

ariccio, 2022-11-29:
Ah, it appears, very strangely, that [apple *does not* expose the GENERIC_ACCESS bluetooth service](https://lists.apple.com/archives/bluetooth-dev/2016/Feb/msg00018.html) for some reason.

What's odd, is I thought I saw a bluetooth diagnostic app reading from it yesterday. Not sure why, other than staring at my damn code too long. I'm wondering if it's worth the MultiPlatformBleAdapter to note this in the docs somewhere?