dotintent/react-native-ble-plx #658: discoverAllServicesAndCharacteristics hangs

gaultierq, 2020-05-01:
Trying to discover the services for a device hangs, and result in the device disconnected.
The following snippet :

```
this.manager.startDeviceScan(
      [SERVICE_UUID],
      null,
      async (error, device) => {
        if (error) return

        if (device) {
            console.log("scanned device:", device);

            device = await device.connect()
            console.log("connected device:", device);

            device = await device.discoverAllServicesAndCharacteristics()
            console.log("device with discovered services:", device);
        }
      })
```
will result with:
```
scanned device: {serviceUUIDs: [...], ...}
connected device: {serviceUUIDs: null, ...}
{... wait a few seconds}
[BleError: Device xxx was disconnected]
and device is actually disconnected
```
I am using [this](https://github.com/himelbrand/react-native-ble-peripheral) to advertise the service. 
What am I doing wrong ?

dariuszseweryn, 2020-05-01:
Why do you think this question belongs to this library and not the other one?

gaultierq, 2020-05-01:
I have no certainty, but the behaviour is strange. The scan find out about the serviceUuids, and then `discover` disconnect without explanation.

dariuszseweryn, 2020-05-01:
There was an error. There is a wiki article about getting data from the errors. You have ignored the template which is supposed to help in fixing bugs (or getting enough info quickly) but you decided to post an issue anyway. Could you take a step back and try to provide at least the needed info?

gaultierq, 2020-05-01:
Sorry I didn't provided the details 

* Library version: 2.0.0:
* Platform: Android
* Platform logs (logcat/XCode):
* JS logs:
no extra logs with logcat, just the one I gave in in my original message (and with `setLogLevel(LogLevel.Verbose);`)
* Contents of the `package.json` file:
{
  "main": "index.js",
  "dependencies": {
    "@react-native-community/masked-view": "^0.1.10",
    "@react-native-community/netinfo": "^5.8.0",
    "@react-navigation/bottom-tabs": "^5.2.8",
    "@react-navigation/material-top-tabs": "^5.1.10",
    "@react-navigation/native": "^5.1.7",
    "@react-navigation/stack": "^5.2.14",
    "amazon-cognito-identity-js": "^4.2.2",
    "aws-amplify": "^3.0.9",
    "aws-amplify-react-native": "^4.1.0",
    "dot-prop-immutable": "^2.1.0",
    "expo": "~37.0.3",
    "expo-constants": "^9.0.0",
    "expo-updates": "~0.1.0",
    "lodash": "^4.17.15",
    "react": "~16.9.0",
    "react-dom": "~16.9.0",
    "react-native": "~0.61.5",
    "react-native-ble-peripheral": "^2.0.1",
    "react-native-ble-plx": "^2.0.0",
    "react-native-device-info": "^5.5.6",
    "react-native-gesture-handler": "^1.6.1",
    "react-native-paper": "^3.8.0",
    "react-native-peripheral": "^0.0.3",
    "react-native-reanimated": "^1.8.0",
    "react-native-safe-area-context": "^0.7.3",
    "react-native-screens": "^2.7.0",
    "react-native-tab-view": "^2.14.0",
    "react-native-unimodules": "~0.9.0",
    "react-native-web": "~0.11.7"
  },
  "devDependencies": {
    "@babel/core": "~7.9.0",
    "@types/react": "~16.9.23",
    "@types/react-native": "~0.61.23",
    "babel-preset-expo": "~8.1.0",
    "jest-expo": "~37.0.0",
    "typescript": "~3.8.3"
  },
  "jest": {
    "preset": "react-native"
  },
  "private": true
}
* Formatted code sample or link to a repository:
in the original message

if you need extra information, I am happy to help. I have spend a lot of times in the issues, and I followed the wiki before posting the issue, but did not find an explanation. 
I wouldn't be surprised the peripheral lib is faulty, but I was surprised that react-native-ble-plx could detect the device advertising the service, and failed to discover the service.

dariuszseweryn, 2020-05-02:
BLE integration is a native one. Native logs are must to debug what exactly is happening.

Given the amount of false positives and the adoption of this library I would expect that the other lib is at fault or you simply did not used it well.

What a peripheral advertises may be totally unrelated to what the central device is able to discover. These are two separate things.

gaultierq, 2020-05-04:
agreed, thanks

closed 2020-05-04 by gaultierq