dotintent/react-native-ble-plx #659: Bug Report

ikaiser, 2020-05-02:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

I followed the guide and write this code:

`import React from 'react';
import { StyleSheet, Text, View } from 'react-native';
import { BleManager } from 'react-native-ble-plx';

export default class BluetoothComponent extends React.Component {

    constructor() {
        super();        
        this.manager = new BleManager();
    }

    scanAndConnect() {
        this.manager.startDeviceScan(null, null, (error, device) => {
            if (error) {
                // Handle error (scanning will be stopped automatically)
                return
            }

            // Check if it is a device you are looking for based on advertisement data
            // or other criteria.
            if (device.name === 'TI BLE Sensor Tag' ||
                device.name === 'SensorTag') {

                // Stop scanning as it's not necessary if you are scanning for one device.
                this.manager.stopDeviceScan();

                // Proceed with connection.
            }
        });
    }
    

    componentDidMount() {
        const subscription = this.manager.onStateChange((state) => {
            if (state === 'PoweredOn') {
                this.scanAndConnect();
                subscription.remove();
            }
        }, true);
    }

    render() {
        return ( null );
    }
}`

# config files

android/app/src/main/AndroidManifest.xml:

`<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.termoscanner">

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>

    
    <uses-permission android:name="android.permission.MANAGE_DOCUMENTS" />
[... 103 lines cut ...]
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:10092:42 in invokeGuardedCallback
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:23663:36 in commitRootImpl
* [native code]:null in commitRootImpl
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:43668:30 in unstable_runWithPriority
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:23525:24 in commitRoot
* [native code]:null in commitRoot
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:22858:34 in runRootCallback
* [native code]:null in runRootCallback
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:13588:38 in <unknown>
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:43668:30 in unstable_runWithPriority
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:13583:28 in flushSyncCallbackQueueImpl
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:13572:35 in flushSyncCallbackQueue
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:22740:37 in scheduleUpdateOnFiber
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:25165:21 in scheduleRootUpdate
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:25910:26 in render
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:73996:78 in renderApplication
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:73582:28 in run
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:73632:28 in runApplication
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:3724:49 in __callFunction
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:3437:31 in <unknown>
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:3678:15 in __guard
* http://packager.qa-e99.anonymous.termoscanner.exp.direct/index.bundle?platform=android&dev=true&minify=false&hot=false:3436:21 in callFunctionReturnFlushedQueue
* [native code]:null in callFunctionReturnFlushedQueue

thanks for help me ....

dariuszseweryn, 2020-05-02:
You did not integrate the library properly. Try going through the steps again and check them once again.

ikaiser, 2020-05-03:
@dariuszseweryn thanks for helpme _ 

My conf is node v 14
and start the app with expo

I followed this step:

1 npm install --save react-native-ble-plx
2 npx react-native link react-native-ble-plx
3 In top level build.gradle make sure that min SDK version is at least 18
4 In build.gradle make sure to add jitpack repository to known repositories
5 In AndroidManifest.xml, add Bluetooth permissions and update <uses-sdk/>

-----

I try to download and start https://github.com/Cierpliwy/SensorTag 
but I had same error.

dariuszseweryn, 2020-05-03:
Is expo project ejected/detached?

ikaiser, 2020-05-04:
no, i'm sorry i didn't run the commands expo eject, because I used this command to start the project: create-react-native-app and inside i found the IOS and ANDROID folder _
-----
now I restart project  and a followed this step:

1) expo init termoscanner
2) expo eject
3) npm install --save react-native-ble-plx
4) npx react-native link react-native-ble-plx
5) In top level build.gradle make sure that min SDK version is at least 18
6) In build.gradle make sure to add jitpack repository to known repositories
7) In AndroidManifest.xml, add Bluetooth permissions and update

but I received same error: null is not an object (evaluating '_BleModule.BleModule.createClient') _

Why ? thanks for the valuable help, I'm a beginner and ask for patience :) _

ikaiser, 2020-05-06:
OK I restart all configuration and work fine now. thanks.

closed 2020-05-06 by ikaiser

dariuszseweryn, 2020-05-06:
I am happy to hear that :)

hadpro24, 2020-05-20:
> OK I restart all configuration and work fine now. thanks.

I got this error. it's not fixed by the above instruction

ikaiser, 2020-05-21:
@hadpro24  I solved not using expo but only native react directly

merimp-gambuzza, 2020-09-17:
Hi all,
I'm using:
`
    "expo": "~38.0.9",
    "expo-updates": "~0.2.10",
    "react": "~16.11.0",
    "react-dom": "~16.11.0",
    "react-native": "~0.62.2",
    "react-native-ble-plx": "^2.0.1"
`
i've ejected project, followed the guide but  I've the same problem.
don't yet exists a solution?

thanks in advance

acenturion, 2020-09-22:
its possible to use this, with expo?

merimp-gambuzza, 2020-09-24:
> its possible to use this, with expo?

as i understood the answer is no, neither expo managed project nor expo ejected projects.

dariuszseweryn, 2020-09-24:
I do not know the current state though I was happily using this library with Expo in the past so I do not see why it shouldn't be possible now. There was even [an integration blogpost on the official expo site](https://blog.expo.io/so-you-want-to-build-a-bluetooth-app-with-react-native-and-expo-6ea6a31a151d).

Most probably you have not integrated the library properly.

RehanShakir, 2022-01-29:
@ikaiser Can you please tell, how you fixed that error?