dotintent/react-native-ble-plx #660: startDeviceScan() responding

ShashwatMDas, 2020-05-02:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I checked all example applications mentioned in README file and they don't work for me.
- [ ] I followed every installation step mentioned in README file.

## Setup problem

So I have everything setup properly( I guess). So what happens is, whenever I am trying to call startDeviceScan(), it is not being called or is stuck at something, no clue! 

### Context

Here's my code:

`constructor() {
    super()
    this.manager = new BleManager()
    this.state = {info: "", values: {}}

    this.scanAndConnect = this.scanAndConnect.bind(this);
  }

   info(message) {
    this.setState({info: message})
  }

  error(message) {
    this.setState({info: "ERROR: " + message})
  }

  updateValue(key, value) {
    this.setState({values: {...this.state.values, [key]: value}})
  }

   componentDidMount() {
    if (Platform.OS === 'ios') {
      this.manager.onStateChange((state) => {
        if (state === 'PoweredOn') this.scanAndConnect()
      })
    } else {
      this.scanAndConnect()
    }
async function requestLocationPermission() 
{
  try {
    const granted = await PermissionsAndroid.request(
      PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
      {
        'title': 'Example App',
        'message': 'Example App access to your location '
      }
    )
    if (granted === PermissionsAndroid.RESULTS.GRANTED) {
      console.log("You can use the location")
      alert("You can use the location");
    } else {
      console.log("location permission denied")
      alert("Location permission denied");
    }
  } catch (err) {
    console.warn(err)
  }
}
requestLocationPermission();
  }

    scanAndConnect() {
      this.info("Start Scan");
      console.log("Scanning")
    this.manager.startDeviceScan(null,
                                 null, (error, device) => {
      this.info("Scanning...")
      console.log(device)

      if (error) {
        this.error(error.message, )
        return
      }

      if(device) console.log(device);
      if (device.id === 'F4:BC:DA:57:D7:CC' || device.name === 'JBL Flip 3 SE') {
        this.info("Connecting to JBL FLip")
        this.manager.stopDeviceScan()
        device.connect()
          .then((device) => {
            console.log("Yohooo");
            this.info("Discovering services and characteristics")
            return device.discoverAllServicesAndCharacteristics()
          }).catch(err => console.log(err));
          // .then((device) => {
          //   this.info("Setting notifications")
          //   return this.setupNotifications(device)
          // // })
          // .then(() => {
          //   this.info("Listening...")
          // }, (error) => {
          //   this.error(error.message)
          // })
      }
    });
  }
`

Here's my console log:
[Sun May 03 2020 01:36:31.321]  LOG      Running "nativeapp" with {"rootTag":131}
[Sun May 03 2020 01:36:31.323]  LOG      Scanning
[Sun May 03 2020 01:36:31.324]  LOG      You can use the location
[Sun May 03 2020 01:37:19.260]  LOG      Scanning
[Sun May 03 2020 01:37:19.640]  LOG      You can use the location

Here's the package.json :
```

{
  "name": "nativeapp",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint ."
  },
  "dependencies": {
    "@remobile/react-native-toast": "^1.0.7",
    "jshint": "^2.11.0",
    "react": "16.11.0",
    "react-native": "0.62.2",
    "react-native-ble-plx": "^2.0.0"
  },
  "devDependencies": {
    "@babel/core": "^7.9.6",
    "@babel/runtime": "^7.9.6",
    "@react-native-community/eslint-config": "^1.1.0",
    "babel-jest": "^25.5.1",
    "eslint": "^6.8.0",
    "jest": "^25.5.3",
    "metro-react-native-babel-preset": "^0.59.0",
    "react-test-renderer": "16.11.0"
  },
  "jest": {
    "preset": "react-native"
  }
}
```

dariuszseweryn, 2020-05-02:
What platform do you test on? Where are native logs as per the issue template request?

ShashwatMDas, 2020-05-02:
> What platform do you test on? Where are native logs as per the issue template request?

Hey,
I'm using OnePlus 6, Android 10 (OxygenOS version 10.3.1).
The logcat log is as follows:
`2020-05-03 03:54:15.619 10605-10793/com.nativeapp I/ReactNativeJS: Scanning
2020-05-03 03:54:15.638 10605-10809/com.nativeapp D/BluetoothAdapter: isLeEnabled(): ON
2020-05-03 03:54:15.643 10605-10809/com.nativeapp D/BluetoothAdapter: isLeEnabled(): ON
2020-05-03 03:54:15.648 10605-10627/com.nativeapp D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=9 mScannerId=0
2020-05-03 03:54:58.157 10605-10624/com.nativeapp D/DecorView: onWindowFocusChangedFromViewRoot hasFocus: true, DecorView@9c7b867[MainActivity]
2020-05-03 03:54:58.159 10605-10605/com.nativeapp E/SensorManager: unregisterListenerImpl callingApp: com.nativeapp,callingPid:10605,callingUid:10426`

dariuszseweryn, 2020-05-02:
Do you have permission for fine location granted? Are Location Services on?

ShashwatMDas, 2020-05-02:
> Do you have permission for fine location granted? Are Location Services on?

Yes, fine location permissions are granted and are checked everytime the app refreshes, so that won't be an issue.

dariuszseweryn, 2020-05-02:
And what about Location Services (GPS)?

ShashwatMDas, 2020-05-03:
> And what about Location Services (GPS)?

As per the docs, we were required to include this:
`<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />`

Also react native docs also state the same thing for geolocation services. Is anything else required?

dariuszseweryn, 2020-05-03:
Location Services. Top bar (usually), location switch — must be turned on. This is something different than Location Permission. Both are needed. 

Afair the library should raise an error if it is off when starting the scan but I cannot be sure if it does on this new Android.

ShashwatMDas, 2020-05-03:
Yes, the location switch is on and the app is able to receive location services from the phone.
Even bluetooth services are properly received. startDeviceScan method return this error if I turn of bluetooth: BluetoothLE is powered off
Which means the library should be integrated fine. It's just that whenever bluetooth is on and it should be scanning for devices, the method doesn't show any logs.

dariuszseweryn, 2020-05-03:
Could you show native logs after you have set the log level to Verbose in the library?

ShashwatMDas, 2020-05-03:
```
2020-05-03 14:01:09.547 14709-14741/com.nativeapp D/DecorView: onWindowFocusChangedFromViewRoot hasFocus: true, DecorView@2f8f426[MainActivity]
2020-05-03 14:01:10.341 14709-14801/com.nativeapp I/ReactNativeJS: Scanning
2020-05-03 14:01:10.372 14709-16269/com.nativeapp D/BluetoothAdapter: isLeEnabled(): ON
2020-05-03 14:01:10.376 14709-14741/com.nativeapp D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=9 mScannerId=0
2020-05-03 14:01:21.136 14709-14709/com.nativeapp E/SensorManager: unregisterListenerImpl callingApp: com.nativeapp,callingPid:14709,callingUid:10426
2020-05-03 14:03:07.116 14709-14709/com.nativeapp E/SensorManager: registerListenerImpl sensorName:BMI160_ACCELEROMETER Accelerometer Non-wakeup,isWakeUpSensor:false,callingApp: com.nativeapp,callingPid:14709,callingUid:10426
2020-05-03 14:03:07.131 14709-17014/com.nativeapp W/unknown:ReconnectingWebSocket: Couldn't connect to "ws://localhost:8081/message?device=ONEPLUS%20A6000%20-%2010%20-%20API%2029&app=com.nativeapp&clientid=DevSupportManagerImpl", will silently retry
2020-05-03 14:03:07.133 14709-14771/com.nativeapp D/DecorView: onWindowFocusChangedFromViewRoot hasFocus: true, DecorView@2f8f426[MainActivity]
2020-05-03 14:03:12.283 14709-14801/com.nativeapp I/ReactNativeJS: Scanning
2020-05-03 14:03:12.299 14709-16269/com.nativeapp D/BluetoothAdapter: isLeEnabled(): ON
2020-05-03 14:03:12.303 14709-16269/com.nativeapp D/BluetoothAdapter: isLeEnabled(): ON
2020-05-03 14:03:12.309 14709-14771/com.nativeapp D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=9 mScannerId=0
```

This is the complete log with log level verbose

dariuszseweryn, 2020-05-03:
Nope. I mean, again, per issue template: [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel). I see no native logs from the library in what you just pasted. Please set the library log level to verbose as well and then paste the logs.

ShashwatMDas, 2020-05-03:
Yeah so i did this, 
    this.manager.setLogLevel(LogLevel.Verbose);

I have pasted the complete log over here (since I have no idea what all information you are looking for). Thanks

```
2020-05-03 14:25:28.437 25031-25031/? I/com.nativeapp: Late-enabling -Xcheck:jni
2020-05-03 14:25:28.453 25031-25031/? E/com.nativeapp: Unknown bits set in runtime_flags: 0x8000
2020-05-03 14:25:28.628 25031-25031/com.nativeapp I/Perf: Connecting to perf service.
2020-05-03 14:25:28.633 25031-25031/com.nativeapp V/Font: Change font:2
2020-05-03 14:25:28.634 25031-25031/com.nativeapp V/Font: Default family:android.graphics.Typeface@1cdaafdc
2020-05-03 14:25:28.637 25031-25031/com.nativeapp D/SoLoader: init start
2020-05-03 14:25:28.637 25031-25031/com.nativeapp D/SoLoader: adding system library source: /vendor/lib
2020-05-03 14:25:28.637 25031-25031/com.nativeapp D/SoLoader: adding system library source: /system/lib
2020-05-03 14:25:28.637 25031-25031/com.nativeapp D/SoLoader: adding application source: com.facebook.soloader.DirectorySoSource[root = /data/app/com.nativeapp-O_7G3vjEaJGtU-Udc76n0Q==/lib/arm64 flags = 0]
2020-05-03 14:25:28.639 25031-25031/com.nativeapp D/SoLoader: adding backup source from : com.facebook.soloader.ApkSoSource[root = /data/data/com.nativeapp/lib-main flags = 1]
2020-05-03 14:25:28.639 25031-25031/com.nativeapp D/SoLoader: Preparing SO source: com.facebook.soloader.DirectorySoSource[root = /system/lib flags = 2]
2020-05-03 14:25:28.639 25031-25031/com.nativeapp D/SoLoader: Preparing SO source: com.facebook.soloader.DirectorySoSource[root = /vendor/lib flags = 2]
2020-05-03 14:25:28.639 25031-25031/com.nativeapp D/SoLoader: Preparing SO source: com.facebook.soloader.DirectorySoSource[root = /data/app/com.nativeapp-O_7G3vjEaJGtU-Udc76n0Q==/lib/arm64 flags = 0]
2020-05-03 14:25:28.639 25031-25031/com.nativeapp D/SoLoader: Preparing SO source: com.facebook.soloader.ApkSoSource[root = /data/data/com.nativeapp/lib-main flags = 1]
2020-05-03 14:25:28.640 25031-25031/com.nativeapp V/fb-UnpackingSoSource: locked dso store /data/user/0/com.nativeapp/lib-main
2020-05-03 14:25:28.641 25031-25031/com.nativeapp I/fb-UnpackingSoSource: dso store is up-to-date: /data/user/0/com.nativeapp/lib-main
2020-05-03 14:25:28.641 25031-25031/com.nativeapp V/fb-UnpackingSoSource: releasing dso store lock for /data/user/0/com.nativeapp/lib-main
2020-05-03 14:25:28.641 25031-25031/com.nativeapp D/SoLoader: init finish: 4 SO sources prepared
2020-05-03 14:25:28.641 25031-25031/com.nativeapp D/SoLoader: init exiting
2020-05-03 14:25:28.650 25031-25031/com.nativeapp D/SoLoader: init exiting
2020-05-03 14:25:28.651 25031-25031/com.nativeapp D/SoLoader: About to load: libjscexecutor.so
2020-05-03 14:25:28.651 25031-25031/com.nativeapp D/SoLoader: libjscexecutor.so not found on /data/data/com.nativeapp/lib-main
2020-05-03 14:25:28.651 25031-25031/com.nativeapp D/SoLoader: libjscexecutor.so found on /data/app/com.nativeapp-O_7G3vjEaJGtU-Udc76n0Q==/lib/arm64
2020-05-03 14:25:28.651 25031-25031/com.nativeapp D/SoLoader: Not resolving dependencies for libjscexecutor.so
2020-05-03 14:25:28.659 25031-25031/com.nativeapp D/JavaScriptCore.Version: 245459.0.0
2020-05-03 14:25:28.660 25031-25031/com.nativeapp D/SoLoader: Loaded: libjscexecutor.so
2020-05-03 14:25:28.660 25031-25031/com.nativeapp D/ReactNative: ReactInstanceManager.ctor()
2020-05-03 14:25:28.660 25031-25031/com.nativeapp D/SoLoader: init exiting
2020-05-03 14:25:28.679 25031-25031/com.nativeapp D/NetworkSecurityConfig: No Network Security Config specified, using platform default
2020-05-03 14:25:28.682 25031-25031/com.nativeapp W/com.nativeapp: Accessing hidden method Lcom/android/org/conscrypt/OpenSSLSocketImpl;->setUseSessionTickets(Z)V (greylist,core-platform-api, reflection, allowed)
2020-05-03 14:25:28.682 25031-25031/com.nativeapp W/com.nativeapp: Accessing hidden method Lcom/android/org/conscrypt/OpenSSLSocketImpl;->setHostname(Ljava/lang/String;)V (greylist,core-platform-api, reflection, allowed)
2020-05-03 14:25:28.682 25031-25031/com.nativeapp W/com.nativeapp: Accessing hidden method Lcom/android/org/conscrypt/OpenSSLSocketImpl;->getAlpnSelectedProtocol()[B (greylist,core-platform-api, reflection, allowed)
2020-05-03 14:25:28.682 25031-25031/com.nativeapp W/com.nativeapp: Accessing hidden method Lcom/android/org/conscrypt/OpenSSLSocketImpl;->setAlpnProtocols([B)V (greylist,core-platform-api, reflection, allowed)
2020-05-03 14:25:28.682 25031-25031/com.nativeapp W/com.nativeapp: Accessing hidden method Ldalvik/system/CloseGuard;->get()Ldalvik/system/CloseGuard; (greylist,core-platform-api, reflection, allowed)
2020-05-03 14:25:28.682 25031-25031/com.nativeapp W/com.nativeapp: Accessing hidden method Ldalvik/system/CloseGuard;->open(Ljava/lang/String;)V (greylist,core-platform-api, reflection, allowed)
2020-05-03 14:25:28.682 25031-25031/com.nativeapp W/com.nativeapp: Accessing hidden method Ldalvik/system/CloseGuard;->warnIfOpen()V (greylist,core-platform-api, reflection, allowed)
2020-05-03 14:25:28.690 25031-25031/com.nativeapp E/SensorManager: registerListenerImpl sensorName:BMI160_ACCELEROMETER Accelerometer Non-wakeup,isWakeUpSensor:false,callingApp: com.nativeapp,callingPid:25031,callingUid:10426
2020-05-03 14:25:28.706 25031-25084/com.nativeapp D/SoLoader: About to load: libfbjni.so
2020-05-03 14:25:28.706 25031-25084/com.nativeapp D/SoLoader: libfbjni.so not found on /data/data/com.nativeapp/lib-main
[... cut ...]
2020-05-03 14:25:29.623 25031-25031/com.nativeapp V/FlingOptimizerScroller: FlingOptimizerOverScroller Init
2020-05-03 14:25:29.637 25031-25031/com.nativeapp V/ViewRootImpl: The specified message queue synchronization  barrier token has not been posted or has already been removed
2020-05-03 14:25:29.657 25031-25052/com.nativeapp D/DecorView: onWindowFocusChangedFromViewRoot hasFocus: true, DecorView@ed515c[MainActivity]
2020-05-03 14:25:29.666 25031-25031/com.nativeapp W/Choreographer: Already have a pending vsync event.  There should only be one at a time.
```

dariuszseweryn, 2020-05-03:
These are the interesting bits on Android:
```
2020-05-03 14:25:29.568 25031-25111/com.nativeapp D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(189904923)
2020-05-03 14:25:29.570 25031-25130/com.nativeapp D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(189904923)
2020-05-03 14:25:29.573 25031-25133/com.nativeapp I/RxBle#QueueOperation: Scan operation is requested to start.
2020-05-03 14:25:29.574 25031-25133/com.nativeapp D/BluetoothAdapter: isLeEnabled(): ON
2020-05-03 14:25:29.576 25031-25052/com.nativeapp D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=6 mScannerId=0
2020-05-03 14:25:29.580 25031-25130/com.nativeapp D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(189904923) in 9 ms
```
Scanning seems to start normally.
Could you check if `nRF Connect` application will show any devices?

I have several ideas:
1. Bluetooth stack on your phone has gone off. Try restarting the phone
2. You try to connect to a device that is not a BLE peripheral but a classic Bluetooth one

ShashwatMDas, 2020-05-05:
> These are the interesting bits on Android:
> 
> ```
> 2020-05-03 14:25:29.568 25031-25111/com.nativeapp D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(189904923)
> 2020-05-03 14:25:29.570 25031-25130/com.nativeapp D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(189904923)
> 2020-05-03 14:25:29.573 25031-25133/com.nativeapp I/RxBle#QueueOperation: Scan operation is requested to start.
> 2020-05-03 14:25:29.574 25031-25133/com.nativeapp D/BluetoothAdapter: isLeEnabled(): ON
> 2020-05-03 14:25:29.576 25031-25052/com.nativeapp D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=6 mScannerId=0
> 2020-05-03 14:25:29.580 25031-25130/com.nativeapp D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(189904923) in 9 ms
> ```
> 
> Scanning seems to start normally.
> Could you check if `nRF Connect` application will show any devices?
> 
> I have several ideas:
> 
> 1. Bluetooth stack on your phone has gone off. Try restarting the phone
> 2. You try to connect to a device that is not a BLE peripheral but a classic Bluetooth one
> 3. Your device is already connected to the phone so it is not advertising itself

Sorry for late reply, got caught up in work. Thanks a lot it worked. Turns out restarting the phone was all I needed to do. Closing the issue with this.

closed 2020-05-05 by ShashwatMDas