dotintent/react-native-ble-plx #662: only iOS - errorCode:205 (Device is not connected) when performing readCharacteristicForService

azuxx, 2020-05-06:
Hello, here's the following:

## Expected Behavior

I would like to get BLE connection flow to work also in iOS. Android everything works.
Precisely, I would get my device to be connected even though it seems discoverAllServicesAndCharacteristics performs successfully.

## Current Behavior

I got this error (only on iOS):

{"message":"Device 18645E73-80DD-5909-9663-XXXXXXX is not connected","errorCode":205,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":null,"name":"BleError","line":29823,"column":28,"sourceURL":"http://XXXX:8081/index.bundle?platform=ios&dev=true&minify=false"}

### Context

AppBLENavigator.tsx: controls the orchestration of BLE flow. Here's the following code

```javascript
//other code...
//the 'core' part
const scanAndConnect = () => {
    console.log('do scanAndConnect');
    setAppState((prev) => ({...prev, deviceStatus: ''}));
    bleManager.startDeviceScan(uuids, options, (error, device) => {
      console.log('scanning...');
      if (error) {
        // Handle error (scanning will be stopped automatically)
        setAppState((prev) => ({...prev, deviceStatus: 'scan_error'}));
        console.log('Error while scanning', error);
        return;
      }
      if (device) {
        setAppState((prev) => ({
          ...prev,
          deviceStatus: 'connecting',
          foundDevice: device
        }));
        console.log('device found', device);
        bleManager.stopDeviceScan();
        console.log('scan stopped, connecting');
        device
          .connect()
          .then((connectedDevice) => {
            console.log('device connected', connectedDevice);
            //Discovering services and characteristics
            return connectedDevice.discoverAllServicesAndCharacteristics();
          })
          .then(
            (discoveredDevice) => {
              //device.discoverAllServicesAndCharacteristics is successful so this callback is emitted
              console.log('device discovery completed', discoveredDevice);
              setAppState((prev) => ({
                ...prev,
                connectedDevice: discoveredDevice,
                deviceStatus: 'connected'
              }));
              return;
             //when this ends
            },
            (reason) => {
              throw new Error(reason);
            }
          );
      } else if (!_hasScanTimout.current) {
        setTimeout(() => {
          console.log('timeout bluetooth scanning');
          bleManager.stopDeviceScan();
          setAppState((prev) => ({...prev, deviceStatus: 'timeout'}));
          return;
        }, 10000);
[... 167 lines cut ...]
  id: '18645E73-80DD-5909-9663-XXXXXXX',
  manufacturerData: null,
  name: 'Demo',
  //...omitted
}
2020-05-06 18:29:51.676097+0200 myReactNativeApp[539:101708] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x280f1a100> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2020-05-06 18:29:52.134 [info][tid:com.facebook.react.JavaScript] reading device MAC address
2020-05-06 18:29:52.352 [error][tid:com.facebook.react.JavaScript] 'failed reading MAC address', [BleError: Device 18645E73-80DD-5909-9663-XXXXXXX is not connected]
```

package.json (helpful part)

```
"react": "16.11.0",
"react-native": "0.62.2",
"react-native-ble-plx": "^2.0.0"
```

Xcode 11.3.1 configuration (helpful part)

[image]
[image]

Let me know if more information is needed and I will try to add detail the best that I can.
Thank you very much

dariuszseweryn, 2020-05-06:
This line makes me wonder if you do not create a new `BleManager()` many times
```
2020-05-06 18:29:51.676097+0200 myReactNativeApp[539:101708] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x280f1a100> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
```

azuxx, 2020-05-07:
> This line makes me wonder if you do not create a new `BleManager()` many times
>
> ```
> 2020-05-06 18:29:51.676097+0200 myReactNativeApp[539:101708] [CoreBluetooth] API MISUSE: <CBCentralManager: [...]

Thanks for the hint @dariuszseweryn , now finally it worked on the pairing way (run the app, and do the connection for the first time).  But when I am in the 'dashboard'/'settings' and I do a reset/refresh of the app it breaks with this error:

```
2020-05-07 11:26:55.078 [info][tid:com.facebook.react.JavaScript] 'Language set is', 'en'
2020-05-07 11:26:55.087 [info][tid:main][RNGestureHandlerManager.m:136] [GESTURE HANDLER] Initialize gesture handler for root view <RCTRootContentView: 0x103d185d0; reactTag: 11; frame = (0 0; 375 667); gestureRecognizers = <NSArray: 0x2829051a0>; layer = <CALayer: 0x2827602a0>>
2020-05-07 11:26:55.102 [info][tid:com.facebook.react.JavaScript] Fetch user data
2020-05-07 11:26:55.749 [info][tid:com.facebook.react.JavaScript] 'Got user data', { me: 
   { __typename: 'User',
     id: '9',
     confirmedAt: '2020-05-05T07:55:06Z',
     email: 'zux.andrea@hotmail.com',
     name: 'Zaza',
     bridgeDevices: [] } }
2020-05-07 11:26:55.989 [info][tid:com.facebook.react.JavaScript] checking for existing device data...
2020-05-07 11:26:55.990 [info][tid:com.facebook.react.JavaScript] no device data found. Start device pairing procedure...
2020-05-07 11:26:55.998521+0200 FalmecReactNative[668:129019] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x283c28300> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2020-05-07 11:26:56.020 [info][tid:com.facebook.react.JavaScript] 'Device Bluetooth state changed to ', 'Unknown'
2020-05-07 11:26:56.127 [error][tid:com.facebook.react.JavaScript] unhandled status [Unknown]. Stopped scan
```

This is the fix I made based on your suggestion:

-I changed this:
```javascript
const AppBLENavigator = () => {
  const bleManager = new BleManager();
```
-to this:
```javascript
const AppBLENavigator = () => {
   const _bleManager = useRef((null as unknown) as BleManager);
```

-then I updated all the usage parts (changing bleManager with _bleManager.current) and adding this if on useEffect:

```javascript
useEffect(() => {
    if (!_bleManager.current) {
      _bleManager.current = new BleManager();
    }
    //same code of before
    return () => {
      _isMounted.current = false;
      delete _bleManager.current; //adding or commenting this does not change the issue situation 
    };
  }, []);
```

-just for reference this is the reset feature

```javascript
const onResetHubPress = () => {
    const connectedDevice: Device = getConnectedDevice();
    if (connectedDevice) {
      connectedDevice.cancelConnection();
    }
    const bridgeDevice: BridgeDevice = getUserDeviceData();
    reloadUserDeviceData();
    bridgeDeviceDelete({
      variables: {
        bridgeDeviceId: bridgeDevice.id
      }
    })
      .then((_) => {
        RNRestart.Restart();
      })
      .catch((reason) => {
        throw new Error(reason);
      });
  };
```

[image]

 
what am I still missing to handle? :( That's weird that on Android everything works instead.

dariuszseweryn, 2020-06-09:
Try adding restore function

closed 2020-06-29 by dariuszseweryn