dotintent/react-native-ble-plx #663: [Android] v2.0.0 seems to ignore onDeviceDisconnected callback

Barbiero, 2020-05-06:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [X] I am running the latest version
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed

## Behavior

I decided to update my app to RN 0.62, and with that came the react-native-ble-plx update to 2.0.

However, after updating I noticed that the `onDeviceDisconnected` callback is getting ignored, no changes made to the actual code; with BLE-PLX v1.1.0 it was working fine. Changelog makes no mention of this behavior change, but it does mention changing the multiplatform ble adapter. Maybe there's something missing there?

I'm not sure how to help diagnose this behavior. To reproduce, try to disconnect an external BLE device after connecting (via `BleManager#connectToDevice`) and register said device with a onDisconnect callback (either via `BleManager#onDeviceDisconnected` or `Device#onDisconnected`).

I have not tested this on iOS.

Barbiero, 2020-05-06:
for the record, I tried updating the `com.github.Polidea:MultiPlatformBleAdapter` dependency within ble-plx's build.gradle to the lastest commit (`92233992ec1c1c7af43674b626c747dd7c9abfd9`), however the problem remains.

dariuszseweryn, 2020-05-06:
@mikolak Do you have any thoughts about this?

mikolak, 2020-05-06:
There's a chance that name of the event might have changed somehow, since the callback itself is not passed down to MBA.

Barbiero, 2020-05-07:
I've tried to monkey-patch the `BleClientManager` to see if I could fix it, like so:
```java
                new OnSuccessCallback<Device>() {
                    @Override
                    public void onSuccess(Device data) {
                        safePromise.resolve(deviceConverter.toJSObject(data));
                    }
                },
                new OnEventCallback<ConnectionState>() {
                    @Override
                    public void onEvent(ConnectionState connectionState) {
                        Log.d("BleClientManager", "onEvent: " + connectionState.name());
                        if(connectionState == ConnectionState.DISCONNECTED) {
                            WritableArray event = Arguments.createArray();
                            event.pushNull();
                            event.pushString(deviceId); // This should actually be a "Device" object but this is just a test anyway

                            sendEvent(Event.DisconnectionEvent, event);
                        }
                    }
                },
                new OnErrorCallback() {
                    @Override
                    public void onError(BleError error) {
                        safePromise.reject(null, errorConverter.toJs(error));
                    }
                });
```

However this doesn't seem to be enough to emit the onDeviceDisconnected event to JS. 
Also I noticed that `BleAdapter` doesn't seem to have a way to get a `Device` by its ID in order to properly emit the event with a Device object like it was prior to 2.0.

Barbiero, 2020-05-07:
By the way, as a workaround I succesfully managed to compile the `1.1.1` version by going to `react-native-ble-plx/android/build.gradle` and removing the following line:
```diff
  
 apply plugin: 'com.android.library'
-apply plugin: 'groovyx.android'
  
 android {
```

If fixing 2.0 will take too long I'll probably write a pre-build script to do that across my projects. 
If there's anything I can do, I'll gladly help.

Thank you for the quick response!

Barbiero, 2020-05-14:
For future reference, I wrote a postinstall script so that version 1.1.1 can build with most recent react-native. It's a super simple script and I'm not even sure if it'll work with a wide variety of projects.
```node
#!/usr/bin/env node

/*
  Fixes react-native-ble-plx@1.1.1 build on android for RN 0.62
  See issue https://github.com/Polidea/react-native-ble-plx/issues/663
 */

const fs = require('fs')
const path = require('path')

const RN_BLEPLX_BUILDGRADLE = path.resolve(
  __dirname,
  'node_modules',
  'react-native-ble-plx',
  'android',
  'build.gradle',
)

if (!fs.existsSync(RN_BLEPLX_BUILDGRADLE)) {
  console.error('react-natibe-ble-plx not installed.')
  process.exit(1)
}

console.debug(
  'Updating file ' +
    path.relative(__dirname, RN_BLEPLX_BUILDGRADLE) +
    ' so it builds with RN 0.62',
)
const buildGradleText = fs.readFileSync(RN_BLEPLX_BUILDGRADLE, {
  encoding: 'utf8',
})
const fixedGradleText = buildGradleText.replace(
  "apply plugin: 'groovyx.android'",
  '',
)
fs.writeFileSync(RN_BLEPLX_BUILDGRADLE, fixedGradleText, { encoding: 'utf8' })
console.debug('react-native-ble-plx@1.1.1 updated!')
```

Later, add the script to your package.json postinstall hook.

josef256, 2020-05-15:
what changes do i expect if i **downgrade** to 1.1.1 ?

Barbiero, 2020-05-15:
This is the 2.0.0 changelog (from https://github.com/Polidea/react-native-ble-plx/releases):

>* Update direct dependency to Multiplatform Ble Adapter
>* Remove support for RN version < 0.60
>* Fixed Typescript callbackType type definition.

So, as far as I can tell, a typescript issue with `callbackType` type definition. From my tests on my apps, everything is working fine.

josef256, 2020-05-15:
thanks i got it to work with your script. 

after connecting to a device i disabled the bluetooth it fire the onDisconnect event so it worked , but whene i tried closing  my device it outputs nothing .. is this intended  or should it also fire whenever my device is off or too far ?

dariuszseweryn, 2020-05-20:
@Barbiero @josef256 What platform have you tested on? iOS / Android?

Barbiero, 2020-05-20:
I've since tested on iOS and it was working fine there. The issue affects Android only.

I'll test #679 out.

Barbiero, 2020-05-20:
@dariuszseweryn #679 fixes the issue!

PR #679 (2020-06-05): Send event when device is disconnected
This should fix the 3 linked issues, but I have no means to test it right now, hence the draft.

Fixes #663 
Fixes #673 
Fixes #675
closed 2020-06-05 by dariuszseweryn