dotintent/react-native-ble-plx #664: Bug Report

ikaiser, 2020-05-06:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x ] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

AndroidManifest: 

`<manifest xmlns:android="http://schemas.android.com/apk/res/android"
  package="com.awesomeproject">

    <uses-permission android:name="android.permission.INTERNET" />

    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission-sdk-23 android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION"/>

    
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

    <uses-sdk android:minSdkVersion="18" android:targetSdkVersion="28"/>

    <application
      android:name=".MainApplication"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:roundIcon="@mipmap/ic_launcher_round"
      android:allowBackup="false"
      android:theme="@style/AppTheme">
      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenSize|uiMode"
        android:launchMode="singleTask"
        android:windowSoftInputMode="adjustResize">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
      </activity>
      <activity android:name="com.facebook.react.devsupport.DevSettingsActivity" />
    </application>

</manifest>`

build.grandle:

`// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    ext {
        buildToolsVersion = "28.0.3"
        minSdkVersion = 18
        compileSdkVersion = 28
        targetSdkVersion = 28
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:3.5.2")

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
[... 176 lines cut ...]
         ]
      },
      "_uniqueId":3
   },
   "id":"61:94:4E:55:19:26",
   "isConnectable":null,
   "localName":null,
   "manufacturerData":"4AABAspTazY=",
   "mtu":23,
   "name":null,
   "overflowServiceUUIDs":null,
   "rssi":-74,
   "serviceData":{
      "0000fe9f-0000-1000-8000-00805f9b34fb":"AAAAAAAAAAAAAAAAAAAAAAAAAAA="
   },
   "serviceUUIDs":[
      "0000fe9f-0000-1000-8000-00805f9b34fb"
   ],
   "solicitedServiceUUIDs":null,
   "txPowerLevel":null
}`

**what's wrong with my process?**

* thanks to help me !!

dariuszseweryn, 2020-05-06:
And what is wrong?

dariuszseweryn, 2020-06-29:
Closing due to lack of information and inactivity of the original poster.

When more data will be available—feel free to add that information and reopen this issue.

Best Regards

closed 2020-06-29 by dariuszseweryn

vagnerwentz, 2020-09-11:
When I log the `device` appears the same, something "100": "BluetoothLE is unsupported on this device" and I would like to know if this can throw something error. But I can connect to the bluetooth module.

shonie2xx, 2023-01-09:
have you fixed the issue?

MaisimZaman, 2023-12-02:
has this been fixed