dotintent/react-native-ble-plx #665: Project Setup

Haugf, 2020-05-06:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [ ] I checked all example applications mentioned in README file and they don't work for me.
- [x] I followed every installation step mentioned in README file.

## Setup problem

Please describe the problem you are experiencing.
Hi, I believe I have the project set up correctly for ios. I am trying to run the manger.scanDevice example but I am running into this. error when trying to call the library. Native Module Cannot Be Null. I have not done the Android setup yet by the way.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. step 1: Open projectand run expo-cli start --tunnel
2. step 2: Open app with QR code on iPhone 10, wait for the app to load

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: ^2.0.0
* Platform: Android/**iOS.**
[image]

* Platform logs (logcat/XCode):
* JS logs: Native module cannot be null.
- node_modules/react-native/Libraries/EventEmitter/NativeEventEmitter.js:36:16 in constructor
- node_modules/react-native-ble-plx/src/BleManager.js:64:47 in constructor
* http://packager.9f-kqn.haugf.loose.exp.direct/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&minify=false&hot=false:124629:52 in App
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:10696:27 in renderWithHooks
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:13481:6 in mountIndeterminateComponent
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:307:15 in invokeGuardedCallbackImpl
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:531:36 in invokeGuardedCallback
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:20488:8 in beginWork$$1
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:19370:24 in performUnitOfWork
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:19347:39 in workLoopSync
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:18997:22 in renderRoot
* [native code]:null in renderRoot
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:18709:28 in runRootCallback
* [native code]:null in runRootCallback
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:5642:32 in runWithPriority$argument_1
- node_modules/scheduler/cjs/scheduler.development.js:643:23 in unstable_runWithPriority
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:5638:22 in flushSyncCallbackQueueImpl
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:5627:28 in flushSyncCallbackQueue
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:18556:30 in scheduleUpdateOnFiber
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:21822:15 in scheduleRootUpdate
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:23042:20 in ReactNativeRenderer.render
- node_modules/react-native/Libraries/ReactNative/renderApplication.js:52:52 in renderApplication
- node_modules/react-native/Libraries/ReactNative/AppRegistry.js:116:10 in runnables.appKey.run
- node_modules/react-native/Libraries/ReactNative/AppRegistry.js:197:26 in runApplication
- node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:436:47 in __callFunction
- node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:111:26 in __guard$argument_0
- node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:384:10 in __guard
- node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:110:17 in __guard$argument_0
* [native code]:null in callFunctionReturnFlushedQueue

Warning: %s: Error boundaries should implement getDerivedStateFromError(). In that method, return a state update to display an error message or fallback UI., RootErrorBoundary
- node_modules/expo/build/environment/muteWarnings.fx.js:27:24 in error
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:645:36 in warningWithoutStack
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:18056:16 in callback
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:7595:16 in callCallback
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:7639:19 in commitUpdateEffects
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:7630:22 in commitUpdateQueue
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:17116:10 in commitLifeCycles
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:20002:23 in commitLayoutEffects
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:307:15 in invokeGuardedCallbackImpl
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:531:36 in invokeGuardedCallback
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:19768:10 in commitRootImpl
* [native code]:null in commitRootImpl
- node_modules/scheduler/cjs/scheduler.development.js:643:23 in unstable_runWithPriority
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:19590:4 in commitRoot
* [native code]:null in commitRoot
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:18709:28 in runRootCallback
* [native code]:null in runRootCallback
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:5642:32 in runWithPriority$argument_1
- node_modules/scheduler/cjs/scheduler.development.js:643:23 in unstable_runWithPriority
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:5638:22 in flushSyncCallbackQueueImpl
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:5627:28 in flushSyncCallbackQueue
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:18556:30 in scheduleUpdateOnFiber
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:21822:15 in scheduleRootUpdate
- node_modules/react-native/Libraries/Renderer/implementations/ReactNativeRenderer-dev.js:23042:20 in ReactNativeRenderer.render
- node_modules/react-native/Libraries/ReactNative/renderApplication.js:52:52 in renderApplication
- node_modules/react-native/Libraries/ReactNative/AppRegistry.js:116:10 in runnables.appKey.run
- node_modules/react-native/Libraries/ReactNative/AppRegistry.js:197:26 in runApplication
- node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:436:47 in __callFunction
- node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:111:26 in __guard$argument_0
- node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:384:10 in __guard
- node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:110:17 in __guard$argument_0
* [native code]:null in callFunctionReturnFlushedQueue

* Contents of the `package.json` file:

{
  "main": "node_modules/expo/AppEntry.js",
  "scripts": {
    "start": "expo start",
    "android": "expo start --android",
    "ios": "expo start --ios",
    "web": "expo start --web",
    "eject": "expo eject"
  },
  "dependencies": {
    "expo": "~37.0.0",
    "expokit": "37.0.0",
    "react": "~16.9.0",
    "react-dom": "~16.9.0",
    "react-native": "https://github.com/expo/react-native/archive/sdk-37.0.0.tar.gz",
    "react-native-ble-plx": "^2.0.0",
    "react-native-unimodules": "~0.8.0",
    "react-native-web": "~0.11.7"
  },
  "devDependencies": {
    "@babel/core": "^7.8.6",
    "babel-preset-expo": "~8.1.0"
  },
  "private": true
}

dariuszseweryn, 2020-05-06:
Is your project ejected?

Haugf, 2020-05-06:
Yeah,  I went with the ExpoKit eject. 

I noticed that ExpoKit was being depreciated sometime soon, are you guys ready for that change?

dariuszseweryn, 2020-05-06:
So do you experience this after ejecting the expo and performing integration steps from readme?

We're aiming mostly vanilla react-native and from what I know Expo is going to follow this way. We just provide steps to integrate it with Expo, but it does not concern us afaik.

Haugf, 2020-05-06:
Oh okay, do you think I should go back an eject through pure react native instead?

And yeah. I ejected to ExpoKit then I followed all the steps before the Android setup steps. After I had this setup, I implemented my code onto the App.jsx file for BLE and started getting this error from there.

dariuszseweryn, 2020-05-06:
I expect that if you use a QR code to run the app it opens the default Expo app, not your standalone, ejected app that you integrated

Haugf, 2020-05-06:
Ooh great point. So I can only run the application on the iOS Simulator? I
think I see what you’re saying now.

Do you know how I can get it to run on phone? I was using Expo-cli commands
to run the app. I’m not sure what the right alternative is now.

dariuszseweryn, 2020-05-06:
To use `react-native-ble-plx` you must use it on a physical device. Simulators/emulators do not support BLE yet.

I am not familiar with expo toolset at the moment but I am pretty sure you can google it out.

Haugf, 2020-05-06:
Okay,  I'll try to run the app on my phone without using expo-cli.  

Thanks for your help. I'll post whatever I find in this thread some time soon.

HerbertLim, 2020-05-15:
Hi @dariuszseweryn , I eject from expo and followed your guide. Now, I only added following code:
`const manager = new BleManager()`
If I run my project with Xcode at my own physical iPhone, it runs without any error.
However if I run this project via `expo start` and run within Expo client, I get `Native module cannot be null`. As far as I understand your answers in this post, that is very natural. Am I right? :)

dariuszseweryn, 2020-05-15:
Yes

closed 2020-06-29 by dariuszseweryn

MaeseppTarvo, 2022-03-26:
I know this issue is closed but I also do happen to have the warning. Can someone explain a bit more? Why we are seeing this error on a real real device on test environment?