dotintent/react-native-ble-plx #666: Can't monitoring characteristic

ad-boschung, 2020-05-07:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. 

- [X] I am running the latest version
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed

## Expected Behavior

I expect, as the documentation says, that if the characteristic is notifiable, I can monitor it to be informed whenever the value of the characteristic has changed.

## Current Behavior

My characteristic can't be monitored although the notifiable value is true. 
I'm having exactly the same problem on iOS and Android.

When I try to monitor the characteristic, I directly receive the following BleError : 
`Characteristic 2456e1b9-26e2-8f83-e744-f34f01e9d703 notify change failed for device X and service 2456e1b9-26e2-8f83-e744-f34f01e9d701`

I checked the following links, try the propositions, but any of them helps me to resolve it : 

- [I got a BleError. Now what?](https://github.com/Polidea/react-native-ble-plx/wiki/=-FAQ:-I-got-a-BleError.-Now-what%3F)

- [Characteristic Notifying](https://github.com/Polidea/react-native-ble-plx/wiki/Characteristic-Notifying)

But, while reading this wiki pages, I found two conflicting informations.
In the [Characteristic Notifying](https://github.com/Polidea/react-native-ble-plx/wiki/Characteristic-Notifying) page, it's written : 

> There is no need to write Client Characteristic Configuration descriptor at all. It is handled by the library/OS.

And in the [I got a BleError. Now what?](https://github.com/Polidea/react-native-ble-plx/wiki/=-FAQ:-I-got-a-BleError.-Now-what%3F), I found the opposite : 

> (iOS) Notifications emit error but it works using different tools
BleError: Characteristic X notify change failed for device ? and service Z
Most probably the device you are using is not conforming to the Core Bluetooth Specification and the Characteristic you try to monitor does not have a Client Characteristic Configuration descriptor.

So, why it is needed that the characteristic have one descriptor if, as said, it's handled by the library/OS ?

For your information, I'm using a BLE uBlox component, and if I try to monitor the characteristic with the u-blox BLE App, it's working. ([Android](https://play.google.com/store/apps/details?id=com.ublox.BLE)/[iOS](https://apps.apple.com/us/app/u-blox-ble/id575523395))
That means the peripheral is notifiable and properly working.

### Context

* Library version: 2.0.0
* Platform: Android/iOS.

```js
device.connect()
        .then(() => {
          console.log('Connected to device');
          console.log('Discovering services and characteristics ...');
          return super.discoverAllServicesAndCharacteristics();
        })
        .then(() => {
          console.log('Services and characteristics discovered');
          console.log('Looking for FIFO characteristic...');
          super
            .characteristicsForService(SERVICE_UUID)
            .then(characteristics => {
              const fifoCharacteristic = this.getFifoCharacteristic(characteristics);
              console.log('Start listening FIFO characteristic ...');
              fifoCharacteristic.monitor((error, char) => {
                console.log(error);
                console.log(char.value);
              });
            });
        })
        .catch(error => {
          reject(error);
        });
```

dariuszseweryn, 2020-05-08:
Please prove that your peripheral is conforming to BLE specification.

ad-boschung, 2020-05-11:
The u-blox device is conforming to BLE specification, and implements his own BLE service. https://www.u-blox.com/sites/default/files/LowEnergySerialPortService_ProtocolSpec_%28UBX-16011192%29.pdf
As I said, the u-blox BLE App works well with the device.

The following pictures show you the device and characteristic :  
Device : 
[image]

Characteristic : 
[image]

dariuszseweryn, 2020-05-11:
I need to see the whole list of attributes. If you log on Android you should be able to set the log level and get the logcat logs.

ad-boschung, 2020-05-12:
I have screens of the logs of the different steps. (Connection, Services Discovering, monitor)
Is it enough for you ? I don't know exactly which list of attributes you need to see ?  
Connection : 
[image]

Services discovering : 
[image]

Monitor error : 
[image]

ad-boschung, 2020-05-18:
@dariuszseweryn  Any news about this issue ?

gilador, 2020-05-19:
having the same issue

dariuszseweryn, 2020-05-21:
Could you please add also what JS print for the exception you get? See [this](https://github.com/Polidea/react-native-ble-plx/wiki/=-FAQ:-I-got-a-BleError.-Now-what%3F)

ad-boschung, 2020-05-22:
Here is the JS logs : 
```json
{
	"message": "Characteristic 2456e1b9-26e2-8f83-e744-f34f01e9d703 notify change failed for device E6516F9D-84C9-F0C6-9535-0E237D58C883 and service 2456e1b9-26e2-8f83-e744-f34f01e9d701",
	"errorCode": 403,
	"attErrorCode": 3,
	"iosErrorCode": null,
	"androidErrorCode": null,
	"reason": "Writing is not permitted.",
	"name": "BleError"
}
```

It's really strange, because the Characteristics attributes are set to writable true.

dariuszseweryn, 2020-05-22:
Your peripheral does not accept setting notifications (writing "set notification on" on CCC descriptor)

ad-boschung, 2020-05-22:
But, when I tried the first times it was working with this lib and suddenly stop working... And with the uBlox-ble app it is working on my peripheral and it is using the notifications too...

dariuszseweryn, 2020-05-22:
I have nothing to compare, I only see that currently the library behaves in a sane way. You could add some logs that show low level communication that happened previously or with uBlox-ble app.

ad-boschung, 2020-05-25:
It is really behaves in a sane way ?

dariuszseweryn, 2020-05-25:
I have nothing to compare to but the low level log indicates that the peripheral discarded a CCC descriptor write with status=3 which is "Writing is not permitted". That is all I know. I do not have a uBlox app to compare. You also could have not configured your peripheral correctly. The pdf you linked has a chapter about sequences which is not exactly what you have shown

ad-boschung, 2020-05-25:
It is really strange because as I said, it was working and suddenly not anymore...
We don't change the peripheral soft and the app soft, and especially not the CCC descriptor...

dariuszseweryn, 2020-05-25:
I would need something to compare to — if you could provide the same info for the older version of the software I could look for differences.

ad-boschung, 2020-05-26:
Okay, I understand that you need something to compare.  
But you said :

> your peripheral does not accept setting notifications (writing "set notification on" on CCC descriptor)

So, how is it possible that the ublox-ble app can write it, and not the lib ? It should be the same behavior ? What do you think ?

ad-boschung, 2020-05-26:
I found something interesting in the logs I provide you.  As you can see in the following logs above, it seams that you try to setCharacteristicNotification two times. One time with enable true, and one time with enable false. Is there a reason to do that two times... ?  

[image]

dariuszseweryn, 2020-05-26:
Two times because notifications are being disabled because of this error on writing descriptor. This is expected.

I do not know the ublox-ble app

ad-boschung, 2020-05-26:
I know that you don't know this app... But do you agree that it means that it should be possible to set notification if they do it !? I'm just to trying a solution to figure it out...

dariuszseweryn, 2020-05-26:
Yes, it definitely should be possible. How however — it is a completely different topic.

ad-boschung, 2020-05-26:
So I don't know why is it not working with this library. Maybe you can have a look at there code...  
They don't have React Native apps but native apps.  
https://github.com/u-blox/Android-u-blox-BLE
https://github.com/u-blox/iOS-u-blox-BLE

dariuszseweryn, 2020-05-26:
With all due respect — I do not have enough time to investigate every peripheral / app that is doing something. Help me help you. You have all resources to check what may be missing. I have asked for logs of successful communication so these can be quickly compared.

ad-boschung, 2020-05-26:
Don't worry, I completely understand.  
The problem is that I don't have any successful communication logs since now it doesn't work anymore. But if I can find the problem I'll let you know.

dariuszseweryn, 2020-05-26:
You could revert the code until it worked and then gather logs I suppose — using a repository this should be a simple task to do

ad-boschung, 2020-05-28:
I haven't changed one line of code since it worked. I tested some things with other BLE apps, then retested my app, and it hasn't worked since. But I haven't changed a single line... That's why I find this behavior so weird.

ad-boschung, 2020-05-28:
I can't revert... My current code is the solution which was working. I don't changed anything. I just tried OTHER APPS in between and then when I tried my app again, I had this issue. But it was the working state of my app. I don't know if you understand what I mean ?

dariuszseweryn, 2020-05-28:
Hmmm... I understood that you have updated this library from 1.1.1 (that was working) to 2.0.0 (which is not working).
If 2.0.0 was working but is not anymore — then I do not know what happened in between

ad-boschung, 2020-05-28:
Yes I talk about my code. Not the lib. I start my app using the 2.0.0.

dariuszseweryn, 2020-05-28:
It seems to be an issue with how the peripheral is used or a problem inside the peripheral's firmware. I think you should ask them. The library just reports what the system is saying (which is that the peripheral does not allow to change notification state)

ad-boschung, 2020-05-28:
Ok. I will try to investigate on that side... Thanks for your help and your time.  
I will let you know if I have some interesting informations.

ath001, 2020-05-30:
I have the same error. The BLE devices' notification characteristics works with other libraries. It seems the client Characteristic Configuration descriptor aren't being set like [Characteristic Notifying](https://github.com/Polidea/react-native-ble-plx/wiki/Characteristic-Notifying) says.

ad-boschung, 2020-06-02:
@ath001 With which library does work ? I could try it because I'm loosing several time to try to make this one working ... Everything is fine on my peripheral but still not working it is still not working ...

ad-boschung, 2020-06-02:
I tried some Java Android code to test my descriptors...

ad-boschung, 2020-06-17:
@dariuszseweryn any news ?

miltonrincon, 2020-06-19:
I am facing the same issue, Long story short , the notification characterisitic is disabled by default, so we have to enable the notification first, but i can not see any instruction to enable it using this.

dariuszseweryn, 2020-06-29:
@ad-boschung I had some time and I checked against a BLE peripheral implemented using an Android OS 7.0. This is what I see with logs enabled when I setup and cancel a notification on a particular characteristic:
```
2020-06-29 14:59:51.071 1345-1485/com.reactnativebleplxplayground D/BluetoothGatt: setCharacteristicNotification() - uuid: 00000002-0000-1000-8000-00805f9b34fb enable: true
2020-06-29 14:59:51.079 1345-1485/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(220331194)
2020-06-29 14:59:51.080 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(220331194)
2020-06-29 14:59:51.158 1345-1358/com.reactnativebleplxplayground D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
2020-06-29 14:59:51.160 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(220331194) in 80 ms
2020-06-29 14:59:53.029 1345-1485/com.reactnativebleplxplayground D/BluetoothGatt: setCharacteristicNotification() - uuid: 00000002-0000-1000-8000-00805f9b34fb enable: false
2020-06-29 14:59:53.029 1345-1484/com.reactnativebleplxplayground I/ReactNativeJS: 'monitor error', '{"message":"Operation was cancelled","errorCode":2,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":null,"name":"BleError","line":26115,"column":28,"sourceURL":"http://localhost:8081/index.bundle?platform=android&dev=true&minify=false"}'
2020-06-29 14:59:53.032 1345-1485/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(122882154)
2020-06-29 14:59:53.034 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(122882154)
2020-06-29 14:59:53.107 1345-1357/com.reactnativebleplxplayground D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
2020-06-29 14:59:53.110 1345-5620/com.reactnativebleplxplayground D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(122882154) in 76 ms
```
I cannot observe any issues. Maybe you could try targeting the latest `master` branch? (`"react-native-ble-plx": "git://github.com/Polidea/react-native-ble-plx.git#387e4f080f953b0476faa9c06cdd448aa7a1a1b4"`)
As for the shared code this is exactly what this library does. No magic there — I see no functional differences. From what I see everything works just fine and according to the spec. I do not have a uBlox device to verify. Feel free to boil it down to the root case and report back — I do not see any way to go forward on my end.

@miltonrincon Have you tried checking documentation? Most functions are for convenience available on all objects accessible (manager/device/service/characteristic)

@ath001 I have checked and I se that the descriptor is properly set as listed in the log above.

Pimbax, 2020-07-24:
Any news on this? I am facing the same issue. When I try to monitor the characteristic, notifications are being enabled and then disabled. I receive the following error: 
`I/ReactNativeJS: {"message":"Characteristic 0000ff35-0000-1000-8000-00805f9b34fb notify change failed for device ? and service 0000ff30-0000-1000-8000-00805f9b34fb","errorCode":403,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":"Cannot write client characteristic config descriptor (code 3) with characteristic UUID 0000ff35-0000-1000-8000-00805f9b34fb","name":"BleError"} `

Logcat:
[image]

I tried testing with nrfConnect and enabling the notifications works and I am receiving data.

orther, 2020-08-20:
I'm currently in the process of tracking down this same issue using u-blox Low Energy Serial Port Service. It worked one day and then stopped working at some point and have been try to troubleshoot it since then.

Does anyone have a solution or suggestion to get around this?

vid553, 2020-12-01:
I had the same problem and managed to solve it with using `bleManager.monitorCharacteristicForDevice(device.id, service, charName, (error, characteristic) => { ... }` instead of `device.monitorCharacteristicForService(service, charName, (error, characteristic) => { ... }`

closed 2021-06-16 by stale[bot]

mrjmrezende, 2023-03-21:
@ad-boschung @inamarinap  did you find some solution? I tried put the CCC with the writeDescriptorForService(), but I have the message "Cannot write to descriptor 00002902-0000-1000-8000-00805f9b34fb. It's not allowed by iOS and therefore forbidden on Android as well".
And in other services that I tried, I got this message: "BleError: Descriptor 00002902-0000-1000-8000-00805f9b34fb not found".

So, I'm facing the same issue until now, that is: Characteristic X notify change failed for device ? and service Y.

I'm tring read characteristic for a Omron Weight Scale and Body Composition.

cristianmoroaica, 2025-04-09:
I'm having the same issue. Any updates on this?

The code:
```
scanAndConnect = async () => {
        console.log("scanAndConnect: start");
        this.ble_manager.startDeviceScan(null, null, async(error, device) => {
            if (error || !device) {
                console.log("Error scanning for devices");
                return
            }
            if (device.name && device.name.includes('LM068')) {
                console.log("Found device: ", device)
                this.ble_manager.stopDeviceScan()

                console.log("Stopped device scan. Discovering services and characteristics for device: ", device.name)
                device.connect().then(() => {

                    console.log("Connected to device: ", device.name)
                    device.discoverAllServicesAndCharacteristics().then((res) => {
                        const sercvices_uuid = res.serviceUUIDs as string[];

                        console.log("Discovered all services for device: ", res);
                        device.characteristicsForService(sercvices_uuid[sercvices_uuid.length-1]).then((characteristics) => {

                            console.log("Characteristics for service: ", characteristics);
                            const characteristic = characteristics[0];
                            device.monitorCharacteristicForService(
                                characteristic.serviceUUID,
                                characteristic.uuid,
                                (error, data) => {
                                    if (error) {

                                        console.error("Error monitoring characteristic: ", JSON.stringify(error, null, 2));
                                        return;
                                    }
                                    if (data && data.value) {
                                        const buffer = Buffer.from(data.value, 'base64');
                                        const bytes = Array.from(buffer);
                                        console.log("Received data: ", bytes);
                                        // Handle the received data here
                                    } else {
                                        console.error("No data received");
                                    }
                                },
                                'monitorData', // optional identifier for the Subscription
                                'notification'
                            );
                        }).catch((err) => {
                            console.error("Error getting characteristics: ", err)
                        });
                    }).catch((err) => {
                            console.error("Error discovering device: ", err)
                        });
                }).catch((err) => {
                        console.error("Error connecting to device: ", err)
                });
            }
        })
    }
```

The logs:
```
 LOG  Multiple permissions result: {"android.permission.ACCESS_FINE_LOCATION": "granted", "android.permission.BLUETOOTH_CONNECT": "granted", "android.permission.BLUETOOTH_SCAN": "granted"}    
 LOG  scanAndConnect: start
 LOG  Found device:  {"id": "34:C9:F0:9A:2E:DD", "isConnectable": true, "localName": "LM068_UniversalLEDD", "manufacturerData": null, "mtu": 23, "name": "LM068_UniversalLEDD", "overflowService
UUIDs": null, "rawScanRecord": "AgEGFAlMTTA2OF9Vbml2ZXJzYWxMRUREAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=", "rssi": -34, "serviceData": null, "serviceUUIDs": null, "solicitedService
UUIDs": null, "txPowerLevel": null}
 LOG  Stopped device scan. Discovering services and characteristics for device:  LM068_UniversalLEDD
[... 13 lines cut ...]
  "name": "BleError"
}
```

React Native: 0.79.0
react-native-ble-plx: 3.5.0

Also getting the same error when trying to put the CCC as @ad-boschung 
> [@ad-boschung](https://github.com/ad-boschung) [@inamarinap](https://github.com/inamarinap) did you find some solution? I tried put the CCC with the writeDescriptorForService(), but I have the [...]

I cannot use monitorCharacteristicForDevice as it is not available anymore.
> I had the same problem and managed to solve it with using `bleManager.monitorCharacteristicForDevice(device.id, service, charName, (error, characteristic) => { ... }` instead of `device.monitorCharacteristicForService(service, charName, (error, characteristic) => { ... }`

BLE hardware module logs:
```
REP*:BLE_CONNECTED:=6aa8-d0-58d035:23
REP*:SER_CCFG=Enabled
BLE_ADAPTER=Ready For USE
REP*:SER_CCFG=Disabled
```

Same behavior as @inamarinap 
> Any news on this? I am facing the same issue. When I try to monitor the characteristic, notifications are being enabled and then disabled. I receive the following error: `I/ReactNativeJS: [...]

Works with nRF Connect for me as well. The notifications can be enabled through the app.