dotintent/react-native-ble-plx #667: startDeviceScan Unexpected Behavior.

0anw0, 2020-05-08:
## Prerequisites

- [-] I am running the latest version
- [-] I checked the documentation and found no answer
- [-] I checked to make sure that this issue has not already been filed

## Expected Behavior

A scan should be starting. 

## Current Behavior

What is the current behaviour?

startDeviceScan isn't even run, When I try to put a [console.log() command]. It doesn't recognize when interpreted.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Install react-native-ble-plx
2. link it to the project. and update SDK minimum version, Adding required permissions to AndroidManifest.xml
3. editing App.js file. 

### Context

startDeviceScan is not responding. The commands that are inside its scope aren't reached. Interesting thing is that when I disable the Bluetooth, the function runs automatically. And Produces a BLE error! 

* Library version: 2.0.0
* Platform: Android

React-Native-cli LOGS:- 
```
[Fri May 08 2020 02:32:19.430]  LOG      Checking Bluetooth
[Fri May 08 2020 02:32:19.473]  LOG      Trying to Scan
[Fri May 08 2020 02:32:19.544]  LOG      Failed to Scan
```
* Platform logs (logcat):
```
05-08 02:32:19.669 16943 17165 I ReactNativeJS: Checking Bluetooth
05-08 02:32:19.695 16943 17085 E OpenGLRenderer: allen debug liyu Key: 0
05-08 02:32:19.723  1062  1186 I SoundPool: Pid:1062 SoundChannel play
05-08 02:32:19.742 16943 17165 I ReactNativeJS: Trying to Scan
05-08 02:32:19.748 16943 17165 I ReactNativeJS: Failed to Scan
05-08 02:32:19.795   248  1962 I iMonitor: scan_blackbox enter
05-08 02:32:19.819 16943 17333 I RxBle#QueueOperation: Scan operation is requested to start.
05-08 02:32:19.839   326   417 W audio_hw_primary: adev_set_parameters, kvpairs : output_device=2
05-08 02:32:19.845   326   417 I AudioCustom_MMI: AudioMMIParse MMI test is close!
05-08 02:32:19.861   248   313 W JankService: virtual bool android::JankManualDumpLogAnalyser::processEntry(android::LogEntry *, int, bool), not beta version or commercial log off
05-08 02:32:19.872   326 13944 I APM_AudioPolicyManager: getOutputForDevice() device 2, stream 1, samplingRate 0, format 0, channelMask 3, flags 0
05-08 02:32:19.872   326 13944 I APM_AudioPolicyManager: getOutputForDevice() returns output 13
05-08 02:32:19.874   326   417 W audio_hw_primary: adev_set_parameters, kvpairs : output_device=2
05-08 02:32:19.874   326   417 I AudioCustom_MMI: AudioMMIParse MMI test is close!
05-08 02:32:19.875   326  1188 I APM_AudioPolicyManager: getOutputForDevice() device 2, stream 1, samplingRate 0, format 0, channelMask 3, flags 0
05-08 02:32:19.875   326  1188 I APM_AudioPolicyManager: getOutputForDevice() returns output 13
05-08 02:32:19.883 16943 17333 I BluetoothAdapter: getBluetoothLeScanner
05-08 02:32:19.900 17946 17958 W Bth     : BluetoothFreeze:tag: 181  uid: 10120  pid: 16943
05-08 02:32:19.909   326   529 E APM::AudioPolicyEngine: unknown stream type 10
05-08 02:32:19.909   326   529 E APM::AudioPolicyEngine: unknown stream type 10
05-08 02:32:19.909   326   529 E APM::AudioPolicyEngine: unknown stream type 10
05-08 02:32:19.909   326   529 E APM::AudioPolicyEngine: unknown stream type 10
05-08 02:32:19.909   326   529 E APM::AudioPolicyEngine: unknown stream type 10
05-08 02:32:19.909   326   529 E APM::AudioPolicyEngine: unknown stream type 10
05-08 02:32:19.909   326   529 I APM_AudioPolicyManager: getNewOutputDevice() selected device 2
05-08 02:32:19.909   326   529 I APM_AudioPolicyManager: setOutputDevice() device 0x0002 delayMs 0 force 1
05-08 02:32:19.909   326   529 I APM_AudioPolicyManager: setOutputDevice() changing to device 0x0002 , muteWaitMs 0
05-08 02:32:19.912   326   520 W audio_hw_primary: adev_set_parameters, kvpairs : srs_cfg:trumedia_enable=1
05-08 02:32:19.913   326   520 I AudioCustom_MMI: AudioMMIParse MMI test is close!
05-08 02:32:19.913   326   520 W audio_hw_primary: adev_set_parameters, kvpairs : HIFIPARA=STEREOWIDEN_Enable=true
05-08 02:32:19.914   326   520 I AudioCustom_MMI: AudioMMIParse MMI test is close!
[... 154 lines cut ...]
        console.log("scanned device:", scannedDevice);
      }
      this.setState({failed: false})
      console.log("Scan Finished")
    });
  }

  render() {
    return (
      <View style={styles.container}>
        <Text style={styles.title}>
          {this.state.attampt}
        </Text>
        <Button
          title='manual mount'
          onPress={()=>this.manualMount()}
        /> 
      </View>
    )
  }
}

```

Can Anyone help me fixing this? And make this function works well?

0anw0, 2020-05-08:
Solutions That didn't work with me:-
1- Restarting the device after installing the App. (didn't work).
2- Try another device (I used 3 different devices, different manufacture. but didn't work).
3- Remove ComponentDidMount and mount it manually. (didn't work).
4- Disable the Bluetooth. and Enable It again. (The Function runs and produces a BLE error after disabling the Bluetooth).
5- calling the function multiple times in a short time (ex: 10 sec.). Produces a BLE error. error code:600. (Although that all the app has access to the GPS ).

dariuszseweryn, 2020-05-08:
your platform logs were gathered without setting log level on the library. please set it and update the logs

0anw0, 2020-05-08:
```
this.manager.setLogLevel(LogLevel.Verbose);
```
this line should set the log level, shouldn't it?

dariuszseweryn, 2020-05-08:
Yup, my bad. What device are you running on?

0anw0, 2020-05-09:
The Devices are ( Lenovo MediaPad 7 [Android 7], HTC Desire 820 [Android 4.4] [ with SensorTag App ], Realme 5 [Android 9] ).  

The app has access to location Services { ACCESS_COARSE_LOCATION , ACCESS_FINE_LOCATION }. And I gave the app access to them manually after installing the application. 

I'm wondering if Gradle version, Or React Native version can solve my problem? My Gradle version is 6.0.1/ React Native Version is 0.62.2. If I downgraded any of both. Will this solve my problem?

dariuszseweryn, 2020-05-11:
> The app has access to location Services { ACCESS_COARSE_LOCATION, ACCESS_FINE_LOCATION }. And I gave the app access to them manually after installing the application.

These are _permissions_ not services. "GPS" must be turned on.

dariuszseweryn, 2020-05-11:
> And I faced the same error.

You did not mention any error before. What exact error is that?

0anw0, 2020-05-11:
The App does not preform any scans. The function itself does not run. console.log('' In scan'' ) which is inside the function does not appear in react native cli logs.

0anw0, 2020-05-14:
I didn't get any useful answers. I'm closing this issue.

closed 2020-05-14 by 0anw0