dotintent/react-native-ble-plx #668: Expose setNotifyValue to be able to stop characteristic notifications

stami, 2020-05-08:
Apple is suggesting that we set all characteristics' notifyValue false before disconnecting from the peripheral.
> In both cases, cancel any subscriptions you may have and then disconnect from the peripheral. You can cancel any subscription to a characteristic’s value by calling the setNotifyValue:forCharacteristic: method, setting the first parameter to NO. 
[Core Bluetooth Programming Guide](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/BestPracticesForInteractingWithARemotePeripheralDevice/BestPracticesForInteractingWithARemotePeripheralDevice.html)

It may be useful also to be able to turn characteristic notifications off but keep the connection to the device, so the `setNotifying(false)` should not be just a part of disconnecting.

## Current Behavior

There's no API (to my knowledge) for [setNotifyValue](https://developer.apple.com/documentation/corebluetooth/cbperipheral/1518949-setnotifyvalue) for turning notifications **off**. There's only turning it on with `monitorCharacteristicForDevice`.

## New behavior

Expose CoreBluetooth's or RxBluetoothKit's setNotifyValue to React Native interface.

### Acceptance criteria

BleManager has ability to turn characteristic notifications off.

### Breaking changes

No breaking changes, it can be implemented as additional API.

dariuszseweryn, 2020-05-08:
it is exposed to my knowledge. check subscription that is returned

stami, 2020-05-13:
I tried to follow the subscription removal, but I understood it's just stored as a UUID to a map, and then removed. Probably removing the JS callback for that event.

However I didn't find anything related to an action that would be run on removal. Neither is anything like `setNotifying(false)` printed in Xcode console output, while when starting notifications the delegate is printing some log about notifying being true.

dariuszseweryn, 2020-05-13:
Could you attach the logs you are referencing?

dariuszseweryn, 2020-06-29:
Monitoring a characteristic returns a `Subscription` which can has `.remove()` function which stops the notification and unregisters it on the server as well.

closed 2020-06-29 by dariuszseweryn