dotintent/react-native-ble-plx #669: Can anyone help me with background mode ios?

arberhh, 2020-05-08:
I managed to create a service on my code so it connects to ble devices successfully, also data gets send on background.

What i wanna achieve is that if connection is lost on background, the app should be able to rescan and reconnect once the device is in range again.
I followed the docs but still the moment the app goes in background no scan is started.

Here is the code that is used to scan in background:
`manager.startDeviceScan(
        ["0000fff0-0000-1000-8000-00805f9b34fb"],
        { allowDuplicates: true },
        (error, device) => {
          console.log("scannig");
          if (error) {
            console.log("error :", error);
          }
          if (device) {
            console.log("device :", device);
            if (device.name) {
              let deviceNameToLowerCase = device.name.toLowerCase();
              console.log("found device");
              
            }
          }
        }
      );`

This is my first question, so sorry if i'm being unclear.

arberhh, 2020-05-12:
@Cierpliwy could you please take a look at this.

dariuszseweryn, 2020-05-12:
Hello @arberhh 
Can you please be more specific about what your situation is, what you do expect, on what platform you test, on what phone, what is the scenario you test. There is a wiki page on ios background operation. for android there is the foreground service you need to implement.

arberhh, 2020-05-12:
@dariuszseweryn yes i managed to implement the foreground service for android.
So about my situation on ios devices i want to reconnect to device once the device is in range.
On android on disconnect event i called scan function and once the device is in range again, it reconnects.
For ios i updated the plist as the docs said, also enabled background mode.
I did new scanning operation providing the service uuid on disconnect event but still if disconnect happens and the app is in background no scan is started.
Thank you for responding

jukben, 2020-05-12:
I can share I have the same issue. I was even thinking if this is not how iOS works. However, I can't perform scan in the background: `startDeviceScan` simply doesn't fire if the app is not in foreground. 

Is it something well known?

priyanka-Sadh, 2020-05-22:
@arberhh I am facing the same issue about reconnect on background mode. Can someone please help me out.
Settings I have done :

in info.plist
`<key>UIBackgroundModes</key>
	<array>
		<string>bluetooth-central</string>
		<string>bluetooth-peripheral</string>
		<string>processing</string>
	</array>`

And in startDeviceScan giving array of UUID
`this.manager.startDeviceScan(['6e400001-b5a3-f393-e0a9-e50e24dcca9e'], { allowDuplicates: true }, (error, device) => {})`
 what extra I am missing to implement background scan on ios ?

arberhh, 2020-05-25:
@priyanka-Sadh sorry I didn't answer sooner, yes I used the same settings u used, but background scan didn't work for me, were u testing with a device or an emulator.

priyanka-Sadh, 2020-05-26:
@arberhh I am using on physical device.

dariuszseweryn, 2020-05-26:
Can you attach native logs? What context are you calling the manager from? Are you sure it is being called?

priyanka-Sadh, 2020-06-02:
@dariuszseweryn 
https://prnt.sc/ssdn0x 
here you go to see debug log(above link), once my app goes to background mode.
Suggest me What I need to do for background mode scanning.
And Is it possible with this library or not ? need to know this.

priyanka-Sadh, 2020-06-03:
@dariuszseweryn Please check above log. Sorry in delay

priyanka-Sadh, 2020-06-03:
@dariuszseweryn 
`  constructor(props) {
        super(props)
        this.manager = new BleManager({
            restoreStateIdentifier: 'BleInTheBackground',
              restoreStateFunction: restoredState => {
                if (restoredState == null) {
                  // BleManager was constructed for the first time.
                } else {
                    console.log("restoredState");
                    console.log(restoredState);
                  // BleManager was restored. Check `restoredState.connectedPeripherals` property.
                }
              }
        })
}`

console.log("restoredState"); is not displaying what does it mean ? I am writting `restoreStateIdentifier` at wrong place ?

dariuszseweryn, 2020-06-06:
It seems that the task is not there. If it has ended then probably you do not conform to what was described in [this wiki page](https://github.com/Polidea/react-native-ble-plx/wiki/Background-mode-(iOS)).

There is everything we know about iOS background behaviour. You have to deal with the implementation.

jukben, 2020-06-07:
Thanks for the link @dariuszseweryn, I probably found the culprit in my case

> You have to specify service UUID as the first argument of bleManager.startDeviceScan function to be able to discover the device in background mode. That means your peripheral should advertise specified service UUID in advertisement data. That's important.

I somehow missed this before.

closed 2020-06-15 by Cierpliwy