dotintent/react-native-ble-plx #673: Unable to detect connection lost in-between, It seems like method  onDeviceDisconnected not been call.

priyanka-Sadh, 2020-05-15:
scenario I am following.
1. Made a connection with device inside the mobile application.
2. After some time to detect disconnection, I just powered off of ble device.
3. Application should detect disconnection in that case too.
4. please check scanAndConnect method, inside that i am using disconnection listener, and guide me to implement this please. 
my code script which is now working. no error no nay message. It's been ignored i guess.
`
async scanAndConnect() {
        this.setState({text1:"Scanning..."})
        this.manager.startDeviceScan(null, null, (error, device) => {
            console.log("Scanning...");
            if (null) {
                console.log('null')
            }
            if (error) {
                this.alert("Error in scan=> "+error)
                this.setState({text1:""})
                this.manager.stopDeviceScan();
                return
            }
            if( /[_]/g.test( device.name ) ) 
            {
                let nameSplit = device.name.split('_');
                if(nameSplit[0] == 'mydevice'){
                    const serviceUUIDs= device.serviceUUIDs[0]
                    this.setState({text1:"Connecting to "+device.name, deviceName : device.name})
                    this.manager.stopDeviceScan();
                    //***********listener for disconnection
                   this.BleManager.onDisconnected(this.state.deviceid, (error, device) => {
                       console.log(error)
                       console.log('DisConnected ' + new Date().toISOString());
                  })
                    //***********listener for disconnection
                    this.manager.connectToDevice(device.id, {autoConnect:true}).then((device) => {
                        (async () => {
                            const services = await device.discoverAllServicesAndCharacteristics()
                            const characteristic = await this.getServicesAndCharacteristics(services)
                            this.setState({"deviceid":device.id, serviceUUID:serviceUUIDs, characteristicsUUID : characteristic.uuid,device:device })
                            this.setState({text1:"Conneted to "+device.name})
                        })();
                        this.setState({device:device})
                        return device.discoverAllServicesAndCharacteristics()
                    }).then((device) => {
                         return this.setupNotifications(device)
                    }).then(() => {
                        console.log("Listening...");
                    }, (error) => {
                        this.alert("Connection error"+JSON.stringify(error))
                    })
                }
            }
       });
    }

`

josef256, 2020-05-15:
im also having the same probleme. onDeviceDisconnected is ignored
ble plx version : 2.0.0

priyanka-Sadh, 2020-05-18:
I appreciate your respond.
What is the solution for it ? Did you find something @josef256 ?

josef256, 2020-05-18:
hello yes i found something,
i downgraded my version to 1.1.1 & it worked.(the 1.1.1 version is not compatible with react antive 0.60+, so you need to a little tweak to make it run)

more info in here #663

priyanka-Sadh, 2020-05-18:
Thanks @josef256 let me try for same.

priyanka-Sadh, 2020-05-18:
Are there any hope of same can support in 2.0.0
Library is being maintained I think.
Any thought about this. @josef256 and guys who maintaining this Library ?

priyanka-Sadh, 2020-05-18:
Hey @josef256 I tried V1.1.1 to setup in RN 0.62.2
I just need to remove `apply plugin: 'groovyx.android` and build installation worked well. 
So when I am trying Interrupting  the connection in between, device.onDisconnected() method is been called. But the issue coming inside the node_module ble-plx Library (check screen shot). Any suggestion on this please ?

[image]

josef256, 2020-05-19:
@priyanka-Sadh  post the full code with the onDisconnected call

also if you edited the file directly in the node_modules it will reset with every npm install (do the importance of a postInstall script)

priyanka-Sadh, 2020-05-19:
Hey @josef256 I was writting onDisconnected method call at wrong place. I fixed it, now onDisconnected method running fine on android.
But now I am trying same on ios and running the project says some error `'BleClient-Swift.h' file not found`, Any suggestion on this please.

josef256, 2020-05-19:
@priyanka-Sadh id didn't try it on ios **yet**.
but let me know if you fix it

PR #679 (2020-06-05): Send event when device is disconnected
This should fix the 3 linked issues, but I have no means to test it right now, hence the draft.

Fixes #663 
Fixes #673 
Fixes #675
closed 2020-06-05 by dariuszseweryn