dotintent/react-native-ble-plx #674: Bug Report Cannot read characteristics after update to 2.0.0

sabba88, 2020-05-15:
## Prerequisites

Please answer the following questions for yourself before submitting an issue.

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x ] I checked to make sure that this issue has not already been filed

## Expected Behavior
Reading data from device

## Current Behavior
After update from 1.1.0 to 2.0.0 I have problem to read from characteristics.
` console.log('characteristic', this.characteristicR);
  const read = await this.characteristicR.read();
`
This is the console output
[image]
The characteristics is properly setted as readable.
The problem is both in Android and Ios.

Do you have any suggestion?
Thanks in advance,
Samuele

sabba88, 2020-05-21:
Someone can help me?

dariuszseweryn, 2020-05-21:
It would be easier to help you if you would conform to the issue template and not delete most of the sections. They are there for a reason. Could you go back and provide requested logs?

sabba88, 2020-05-21:
Hi @dariuszseweryn,
I edit the original post. I hope it could be ok now.
Thanks in advance,
samuele

dariuszseweryn, 2020-05-21:
Native logs are needed. You have also [this wiki page](https://github.com/Polidea/react-native-ble-plx/wiki/=-FAQ:-I-got-a-BleError.-Now-what%3F). Please attach native logs and print the full error object

sabba88, 2020-05-21:
Hi @dariuszseweryn,
thanks for your reply. I update the first with more detailed information, using Verbose loglevel. I hope there will be enough.

Thanks in advance,
Samuele

dariuszseweryn, 2020-05-22:
Could you print object `this.characteristicR`?

sabba88, 2020-05-22:
It'already printed in the last screen ho js log. Just over the error. READ - Characteristic.

sabba88, 2020-05-26:
Hi,
when will be released this fix?
Thank you for your work,

Samuele

dariuszseweryn, 2020-05-26:
I hope for the end of this week

sabba88, 2020-06-03:
Hi,
any news about this fix?
Thanks in advance,

Samuele

dariuszseweryn, 2020-06-09:
You could try using `react-native-ble-plx: git://github.com/Polidea/react-native-ble-plx.git#387e4f080f953b0476faa9c06cdd448aa7a1a1b4` and tell me if it works for you.

I think that next week we will have a npm released version

sabba88, 2020-06-15:
Hi @dariuszseweryn,
it works with that fix. However, I see that the fix is only for Android, but the problem is also into Ios version.
Thanks,
Samuele

Cierpliwy, 2020-06-15:
Is it an issue on iOS? As you have two characteristics with the same UUID, make sure to call functions directly from `Characteristic` object (e.g. `characteristic.read()`). Otherwise, first matching characteristic will be selected for an action (writable in your case).

sabba88, 2020-06-15:
Hi @Cierpliwy ,
the issue is in both platform. With the previous fix of Darius the android environment works fine. WIth iOS still get the error.
This is where I setup the characteristics.
`
    const characteristics = await service.characteristics();
    this.characteristicW = characteristics[0];
    this.characteristicR = characteristics[1];
`

And this is how read the characteristic

`
        console.log('READ - characteristic', this.characteristicR);
        const read = await this.characteristicR.read();
        const receivedAddress = read.value.charAt(2) + read.value.charAt(3);
        console.log('Address Read Sent', sentAddress);
        console.log('Address Read Received', receivedAddress, read.value);
`

In the first post you can check all details.

Thanks,
Samuele

dariuszseweryn, 2020-06-29:
Could you print both characteristics? i.e. 
```
console.log(JSON.stringify(characteristicR));
console.log(JSON.stringify(characteristicW));
```

nriccar, 2020-07-22:
Any news on iOS?

sabba88, 2020-07-22:
Hi,
sorry for the delay.
I "restart" the app from 0 re-insalling every package with the last version and everything works fine.

Thanks,
Samuele

nriccar, 2020-07-22:
With react native 0.63.0?

Thanks for the info Samuele

sabba88, 2020-07-23:
No with 0.62.2

These are all my dependencies.

`
    "@react-native-community/async-storage": "^1.11.0",
    "@react-native-community/checkbox": "^0.4.2",
    "@react-native-community/masked-view": "^0.1.10",
    "@react-native-community/picker": "^1.6.3",
    "@react-navigation/bottom-tabs": "^5.5.2",
    "@react-navigation/native": "^5.5.1",
    "@react-navigation/stack": "^5.5.1",
    "prop-types": "^15.7.2",
    "react": "16.11.0",
    "react-native": "0.62.2",
    "react-native-animatable": "^1.3.3",
    "react-native-base64": "^0.0.2",
    "react-native-ble-plx": "^2.0.0",
    "react-native-connectivity-status": "^1.5.1",
    "react-native-dialog-input": "^1.0.7",
    "react-native-fs": "^2.16.6",
    "react-native-gesture-handler": "^1.6.1",
    "react-native-localization": "^2.1.6",
    "react-native-orientation": "^3.1.3",
    "react-native-reanimated": "^1.9.0",
    "react-native-safe-area-context": "^0.7.3",
    "react-native-screens": "^2.8.0",
    "react-native-share": "^3.3.3",
    "react-native-svg": "^9.11.1",
    "react-native-svg-charts": "^5.3.0",
    "react-native-swipe-list-view": "^2.0.0",
    "react-native-swiper": "^1.5.14",
    "shortid": `"^2.2.14"``

nriccar, 2020-07-23:
For me is not working, I have an iPhone Xs and iOS 13.6. 

Same same, new project, with your same versions packages. 

Any guess?

closed 2020-10-03 by stale[bot]