dotintent/react-native-ble-plx #677: Question

JayaniH, 2020-05-18:
When I scan for devices, only my laptop is detected as a device, and phones(that support Bluetooth LE) are not detected.
Also, only the device id is shown for my laptop, and the device name and other attributes are shown as null.

```
import {BleManager} from 'react-native-ble-plx';

const DeviceManager = new BleManager();

export const scan = function scan() {
    const subscription = DeviceManager.onStateChange((state) => {
        if (state === 'PoweredOn') {
          DeviceManager.startDeviceScan(null, null, (error, device) => {
            if (error) {
              console.log("error",error);
            }
            if (device !== null) {
                console.log("device found ----> [id,name]", device.id, device.name);
            }
          });
          subscription.remove();
        }
    }, true);
}
```

This is my code. Why aren't all the devices detected?

josef256, 2020-05-19:
you need to simulate a peripheral with your phone : i suggest using the app **BLE  Peripheral simulator** on android

dariuszseweryn, 2020-05-21:
☝️ If a peripheral does not advertise it cannot be scanned

closed 2020-05-21 by dariuszseweryn