dotintent/react-native-ble-plx #682: Communication between phones

mrousavy, 2020-05-25:
## Question

First off - I know a question like this has been asked before ([here](https://github.com/Polidea/react-native-ble-plx/issues/644)), but it has been answered as not supported.

However, on [this](https://forums.developer.apple.com/thread/111829) post in the apple developer forums, a user said:

> As of macOS 10.9 and iOS 6, Mac and iOS devices can also function as Bluetooth low energy peripherals, serving data to other devices, including other Mac and iOS devices.

As far as I know, Android supports this on newer devices aswell.

Now my question is, is it possible to build a discovery service using this library?

I am really lost right now, my use case is: I want to build something like the iOS AirDrop **View**. So Users open a popup, and can then see other nearby devices by their custom broadcasted username. Is this possible using `react-native-ble-plx` and if not, does anyone know how I can achieve this?
 Thanks a lot!

dariuszseweryn, 2020-06-05:
> Is this possible using react-native-ble-plx and if not, does anyone know how I can achieve this?

As you have already seen in the above mentioned issue — this library supports only central role, and to have a BLE connection a central and a peripheral are needed. As for the second part of your question I cannot help you.

closed 2020-06-05 by dariuszseweryn

whatnottodo, 2022-03-05:
@mrousavy did you ever find a solution?