dotintent/react-native-ble-plx #683: Question: `connectedDevices` - inconsistent behavior between iOS and Android

askielboe, 2020-05-27:
## Question
 
I'm trying to get devices connected through the device Bluetooth settings from within my react-native app. I've managed to solve it on iOS using `connectedDevices`, but on Android I'm seeing a different behavior.
 
**iOS**: On iOS I simply get the list of devices using `connectedDevices`, then `connectToDevice`, `discoverAllServicesAndCharacteristicsForDevice` and I can get and subscribe to services. Easy.
 
Under the hood it looks like `RxBluetoothKit` [`CentralManager`](https://github.com/Polidea/RxBluetoothKit/blob/8ead7ca5ab242c4f18c1840839fb54a504abfcb7/Source/CentralManager.swift) calls the Apple Core Bluetooth method [`retrieveConnectedPeripherals`](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518924-retrieveconnectedperipherals). That's all good.
 
**Android**: On Android I don't get any devices on `connectedDevices`. Digging through the `RxAndroidBle` library it looks like the implementation is quite different from `RxBluetoothKit`. Here it looks like `connectedDevices` is a private map in [BleModule.java](https://github.com/Polidea/MultiPlatformBleAdapter/blob/d2279e0621b285f2d16382af668ae3981abf0639/android/library/src/main/java/com/polidea/multiplatformbleadapter/BleModule.java), and the only place where anything is added to the map is on `safeConnectToDevice`.
 
From the Android SDK it looks like the way to get devices paired to the local Bluetooth adapter is through [`getBondedDevices`](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getBondedDevices()), which I can't find any reference to in `RxBluetoothKit`.

Am I correct in my assessment that the behavior of `connectedDevices` is different between Android and iOS, and is there a recommended way to proceed in order to get bonded devices on Android?

Cheers!

bitcrumb, 2020-06-15:
I investigated this a couple of weeks back as well. And I had the exact same findings. These where my notes back then:

[image]

Cierpliwy, 2020-06-19:
As name suggests "getBondedDevices"  return "the set of BluetoothDevice objects that are bonded (paired) to the local adapter.". Connected device may or may not be a bonded device. There is a semantic difference and I wouldn't want to mislead users.

> iOS: On iOS I simply get the list of devices using connectedDevices, then connectToDevice, discoverAllServicesAndCharacteristicsForDevice and I can get and subscribe to services. Easy.

If you are happy to connect to an already known device, there is a function [`bleManager.devices(...)`](https://polidea.github.io/react-native-ble-plx/#blemanagerdevices) to do that.

matgardon, 2020-08-11:
Hi, as it stands, the lack of a proper support for `getBondedPeripherals` means I am switching to another react-native library (`react-native-ble-manager`) which supports it. This is a 'solution' while awaiting support, but I guess it is not an option for everybody (and not one you desire on your side).
It doesn't too complicated to implement in this library, is there any reason (security or otherwise) it's not ?

The use-case is pretty legit: your user wants to configure one of your bluetooth device, so he must turn it on and scan it first to connect, but sometimes tech-savvy end-users will have already paired their device in the OS Bluetooth section before going to your app, how do your retrieve devices in this state while also scanning for possibly new visible products ( both of which you should be able to filter for known devices you can configure based on a specific serviceUuid) ?

closed 2020-10-18 by stale[bot]

RRMoelker, 2020-10-29:
We are running into the same issue of Android not working like iOS.

Or more specific. The Android function does not work as described:

"Returns a list of the peripherals (containing any of the specified services) currently connected to the system which have discovered services. Returned devices may not be connected to your application. Make sure to check if that's the case with function isDeviceConnected." - [connectedDevices doc](https://polidea.github.io/react-native-ble-plx/#blemanagerconnecteddevices)

The function does not return the devices connected to the system. But those connected to through the plx library. I think the aforementioned [getBondedDevices](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getBondedDevices()) fits the description a lot better.